ros-navigation/navigation2 #951: Suggestion: List common tunable parameters in the yaml file

MengNan-Li, 2019-07-19:
_**Such as:**_
     observation_sources: front_laser
     front_laser:
       topic: /scan
       max_obstacle_height: 2.0
       data_type: "LaserScan"
       clearing: True
       marking: True
       **inf_is_valid: True
       min_obstacle_height: 0.0
       max_obstacle_height: 2.0
       obstacle_range: 3.0
       raytrace_range: 3.5**

Listing commonly adjustable parameters will make it easier for us to adjust the running effect of the robot. This is a small suggestion. ☺

SteveMacenski, 2019-07-19:
Funny you bring it up, we were just talking about that today on our weekly call :-)

mjeronimo, 2019-07-19:
@MengNan-Li  @SteveMacenski  Perhaps we can have two parameters files: the first one would have every parameter in the system. The second would be commonly used parameters. If you agree, the task would be to identify those parameters that should be the in the commonly used set. 

If we identify these commonly used parameters in the code using the ParameterDescriptor's additional information field, we can use the parameter dumping utility to automatically dump either all parameters or the subset (in either YAML or MD formats). What do you think?

SteveMacenski, 2019-07-19:
I think that works. We could have 1 wiki page that has "all" and another with "all important", I'm not sure I'd go out of my way to implement the paring down of parameters through the utility (unless its easy) but we can hand select them for now.

mjeronimo, 2019-07-20:
Sure, I agree: hand-selected would be the first, quick option. It wouldn't be too much trouble after that to add this capability to make it automatic.

orduno, 2019-10-22:
We could use [this](https://arxiv.org/abs/1706.09068) as a starting point.

SteveMacenski, 2019-10-22:
I think the best starting point would be to finish the ticket to have a table of all the parameters everywhere to chose from, then look at that guide for descriptions of our selected candidates

SteveMacenski, 2019-11-06:
Closing from #1305 imminent merging

closed 2019-11-06 by SteveMacenski