ros-navigation/navigation2 #952: How to impove the running speed of robot?

MengNan-Li, 2019-07-19:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - binaries
- Version or commit hash:
  - master: aaa8c2e
- DDS implementation:
  - OpenSplice
- Client library (if applicable):

#### Steps to reproduce issue

```
1. Start robot driver
2. ros2 launch nav2_bringup nav2_bringup_launch.py ...
```

#### Expected behavior
The moving speed of robot adjusts with parameter adjustment.
#### Actual behavior
The robot moves at a slower speed and the line speed is about 0.4m/s.
#### Additional information
I adjust the max_vel_ params of dwb_controller, but the running speed of robot is still slow.
dwb_controller:
  ros__parameters:
    use_sim_time: True
    debug_trajectory_details: True
    min_vel_x: 0.0
    min_vel_y: 0.0
    max_vel_x: 0.9
    max_vel_y: 0.0
    max_vel_theta: 1.0
    min_speed_xy: 0.0
    max_speed_xy: 0.9
    min_speed_theta: 0.0

----
## Feature request

#### Feature description

#### Implementation considerations

SteveMacenski, 2019-07-19:
Did you check the settings on your velocity smoother or anything downstream of the stack to make sure its the stack’s fault? This sounds like maybe a limit was put in downstream on a smoother or on the hardware

crdelsey, 2019-07-19:
If there isn't something outside the stack limiting the speed, it would be helpful to see the output of the `evaluation` topic

MengNan-Li, 2019-07-22:
I want to know what the name of the controller_frequency parameter of DWA in ROS1 is in ROS2. After changing the code below, we can adjust the movement speed of the robot by parameters.
```
nav2_dwb_controller/dwb_plugins/src/standard_traj_generator.cpp
void StandardTrajectoryGenerator::startNewIteration(
  const nav_2d_msgs::msg::Twist2D & current_velocity)
{
  // velocity_iterator_->startNewIteration(current_velocity, sim_time_);
  velocity_iterator_->startNewIteration(current_velocity, 0.1);
}
```

crdelsey, 2019-07-22:
> I want to know what the name of the controller_frequency parameter

It was unimplemented. I've just added that feature back in PR #970. The name will be the same as ROS 1 - `controller_frequency`

----

Regarding the code change above - you can change the sim_time_ value by setting the `sim_time` parameter instead of changing the code. It is already in the default nav2_params.yaml file and is 1.7 by default. 

I'm not sure why that matters to the running speed though. It should only change how far ahead we project the velocity. 0.1 seems like a very low value though. I'd expect that to cause problems with avoiding obstacles.

MengNan-Li, 2019-07-25:
@crdelsey I just changed the one line code. Other sim_time_ values are still set in the default nav2_params.yaml file.

SteveMacenski, 2019-09-04:
@crdelsey @MengNan-Li can this be closed with #970 ?

MengNan-Li, 2019-09-05:
OK

closed 2019-09-05 by MengNan-Li