ros-navigation/navigation2 #953: Recovery Behaviors Roadmap for the Eloquent Release

mhpanah, 2019-07-19:
**Objective**: Scope-based failure Handling

Utilizing Behavior Trees with Recovery Node allows one to handle failures at multiple scopes. With this capability, any action in a large system can be constructed with specific set of recovery actions suitable for that action. Thus, failures can be handled locally within the scope. 
With that in mind, the tree below represents one example on how to implement scope-base recovery actions suitable with the current local and global planner.  This tree is my proposed high level recovery behavior representation for the eloquent release and a detailed representation for each node will be added.  In summary, we are planning to have specific set of recovery behaviors locally within each planner and also at higher level before aborting the mission.   We have already implemented some of the needed BT action nodes for the dashing and the following are to be implemented for the eloquent:

- [ ] Identify possible communication failures such as topics or service calls due to timeout or other reasons for the global planner 

- [ ] Identify possible communication failures such as topics or service calls due to timeout or other reasons for the local planner

- [ ] Recovery actions for the communication failures, such as utilizing the lifecycle manager to restart/reset the identified module that caused the communication failure 

- [ ] Implement conservative reset BT Action node for the global planner

- [ ] Implement aggressive reset BT Action node for the global planner

- [ ] Implement conservative reset BT Action node for the local planner

- [ ] Monitor localization uncertainty

[image]

SteveMacenski, 2019-07-19:
I dont think the costmap conservative/aggressive costmap resets are a concept we need to bring into navigation2. If I'm going to reset part of a costmap, I'm going to probably reset all of it -- because how can you trust anything now. But if @DLu or others have reasons why they might disagree with that use-case I'm open to keeping it. I've made that argument in my STVL package to some folks and haven't yet* found a person to give a convincing reason to keep it.

* but again, super open to change my mind on it if someone can provide a plausible situation

If there are communication issues resulting in recovery modes being triggered, that's a huuuuge problem to a point it makes me ask the question "Why are we even using ROS2". I know there are problems tight now we need to have some answer to because things aren't stable at the moment, but I would really like to not build infrastructure around that. Now if a node crashes and that's why its not responding, that's different. If messages are consistently being lost on a reliable QoS, that's serious cause for alarm for an autonomous agent moving around a world with humans

Thinking about some better recoveries would be awesome too. RIght now its just backup, spin, and clear costmaps. What else might we like to do?

mhpanah, 2019-07-19:
@SteveMacenski The communication issues can happen due to either ROS 2, Node crashes, Sensor Failures, or etc. By checking the communication failure, we can identify the root cause and invoke a suitable recovery action.

Regarding ROS 2 communication, the main question to ask is whether ROS2 communications will be reliable by the Eloquent release? If the answer is no, then I think for the Elequonet release we still need to check for it until it gets more reliable. 

If any node crashes, the lifecycle manager will get invoked to re-start/reinitialize internal states of that node and bring it into a stable active state. 

If a sensor fails and stops sending data, one recovery action would be to stop the robot, and restart the sensor driver and check to see if the issue is resolved or not.

These are just a few ideas on what the communication check and recovery should do.  A more detailed plan is to be discussed.

DLu, 2019-07-31:
You can construct scenarios where conservative resets are worthwhile, but they mostly occur when you have a laggy sensor like the PR2's tilting laser scanner which may take a few seconds to re-sense a nearby local obstacle.

SteveMacenski, 2019-10-01:
Adding comment per @mhpanah request:

We want to be able to do behaviors and check if it fixed the problem between the behaviors (and really, we'd like to know the *moment* its fixed even during a recovery to start getting back along our way). This would imply that rather than doing the current recovery behavior sequence star where we do each sequentially, we need a way to do each, test if whatever failed can now compute a path / control effort, and if not, continue to the next recovery.

This discussion started around my addition of the wait behavior, which in my experience just waiting a few seconds after an issue, the issue will often fix itself. If someone's harassing the robot, pause, they back off, go along its way. There's a stuck voxel in the voxel layer, gets cleared out, along its way, etc. Having the robot pause for a few moments is better than spinning for every single issue and less noticeable to surrounding people watching that there may be something wrong

SteveMacenski, 2019-10-17:
@mhpanah any movement in this?

closed 2019-12-23 by mhpanah