ros-navigation/navigation2 #954: Create system test with a holonomic robot

crdelsey, 2019-07-19:
## Feature request

#### Feature description
The nav stack has support for holonomic robots but it is untested. There needs to be an automated test that ensures it still works properly.

#### Implementation considerations
This could be a variant of the nav2_system_test package with a different robot model used.

SteveMacenski, 2019-08-08:
@crdelsey functionally holonomic robots are treated the same as differential drive in the stack: NavFn will plan irrespective to kinematics, and DWB just needs samples set for Y as well as X.

I don't think this is something that is needed for any performance or test. Regressions in DWB that disable, for instance, the functioning of the Y motion won't actually make it fail to navigate because a holonomic robot can act like a differential drive robot

crdelsey, 2019-08-09:
Maybe I should retitle this to just "Test with a holonomic robot". For all we know right now, we could seg fault or just not take advantage of the Y velocity like you mention. My first instinct is to automate the test if possible, however, I'll take a manual test if that's all we've got. I just want to be able to say "Yes it works" and have some confidence that I'm actually right.

SteveMacenski, 2019-08-09:
Ah I see. Do we know anyone with a holonomic robot we can poke? Or I suppose just a reconfigured description/gazebo package would be sufficient. I can't think of many other than Pepper and a couple of proprietary warehouse bots.

ninamwa, 2019-10-15:
Hi. Will the navigation2 stack work with a holonomic robot? Where should this parameter be set? I could only find examples for this with the navigation1 stack.

crdelsey, 2019-10-15:
@ninamwa We haven't tested with a holonomic robot yet, however, I suspect it will work since the code has simply been ported from ROS 1.

I believe the robot should work without any parameter changes. It would just move like a differential drive robot in that case.

To take full advantage of the holonomic robot though, there are two sets of parameters you need to change:

```
amcl:
  ros__parameters:
    robot_model_type: "omnidirectional"
```

And, assuming you are using the Dashing release, these are the parameters you need to change for the controller. Obviously, you need to set the values to whatever is appropriate for your robot. The values below are about right for a holonomic robot that is comparable to a Turtlebot 3 in speed.

```
dwb_controller:
  ros__parameters:
    min_vel_y: -0.26
    max_vel_y: 0.26
    max_speed_xy: 0.26
    min_y_velocity_threshold: 0.001
    acc_lim_y: 2.5
    acc_lim_theta: 3.2
    decel_lim_x: -2.5
    decel_lim_y: -2.5
    vy_samples: 20
```

If you end up trying this, please reply here with your results

ninamwa, 2019-12-07:
@crdelsey Hi. We tried this at our KMR iiwa robot yesterday, and it was a success. I only did the changes you suggested!

SteveMacenski, 2019-12-07:
Awesome! Super glad you had a good experience, means we did a couple things right. 

Can you add a short gif of your robot moving around? It would mean alot to us to have a short clip to share of your robot omni robot moving around.

SteveMacenski, 2020-01-14:
User above tested on holonomic working

closed 2020-01-14 by SteveMacenski

Osal22, 2024-09-17:
After setting recommended parameters I do not see any difference in velocity commands. Robot still behaves like differential drive anything further I am missing.

manavthakkar, 2024-09-20:
I tried with the above mentioned parameters and just like the experience of @Osal22 , the robot behaves like a diff drive