ros-navigation/navigation2 #955: Add a system test that includes the voxel layer

crdelsey, 2019-07-19:
## Feature request

#### Feature description
The voxel layer is part of the nav stack but is untested. A system test is needed that includes the voxel layer.

SteveMacenski, 2019-07-19:
FYI: adding in a 3D sensor into gazebo is *very heavy* I would recommend only working from bag files for this. 

You'll need to run this on a very expensive tower to get the 3D sensors stuff in gazebo running at any reasonable rate

crdelsey, 2019-07-19:
We've already got some tests that are manually run (those tests that verify our dependencies are clean before releasing). This and others could be in that category. In that case, it probably doesn't matter how long it takes. That said, the faster we can make the test the better so long as it covers what we need it to cover

SteveMacenski, 2019-07-19:
Got it, maybe we just setup a dummy 3D sensor that's just super low resolution or low FOV so that we minimize its impact in CI

yathartha3, 2019-09-20:
@orduno Here is the issue for Voxel Layer.

SteveMacenski, 2019-11-06:
@crdelsey do you consider hand testing & validating the voxel layer through #1291 in the spirit of completing this ticket?

crdelsey, 2019-11-11:
Yes. Let's close this as resolved by #1291

closed 2019-11-11 by crdelsey