ros-navigation/navigation2 #965: Navigation2 (master) doesn't build after #946

mlherd, 2019-07-21:
Navigation2 (master) doesn't build after these (#946) changes.

This is the error message I get during build.

```
/home/melih/navigation2_ws/src/navigation2/nav2_amcl/src/amcl_node.cpp:44:10: fatal error: tf2_ros/create_timer_ros.h: No such file or directory
 #include "tf2_ros/create_timer_ros.h"
```

**Reason**: Ros2 Geometry Dashing branch doesn't have the create_timer_ros.cpp file, but their Ros2 branch includes the file.

- https://github.com/ros2/geometry2/tree/dashing/tf2_ros/src
- https://github.com/ros2/geometry2/tree/ros2/tf2_ros/src

If a user follows the official ROS 2 build from source steps they won't get the create_timer_ros.cpp file.

```
mkdir -p ~/ros2_ws/src
cd ~/ros2_ws
wget https://raw.githubusercontent.com/ros2/ros2/release-latest/ros2.repos
vcs import src < ros2.repos
```

**Solution**: 
Build [tf2_ros ros2](https://github.com/ros2/geometry2/tree/ros2/tf2_ros/src) branch from source (recommended) or comment out the changes (not recommended)

SteveMacenski, 2019-08-01:
@mlherd that header really doesnt do that much, can we just overlay it in our repo temporarily to get it building?

I cant find a configuration of branches and commits which gets navigation2 to build, can you share what you have working? I've tried ros2 latest, and rolling back a number of commits but never quite get it working

SteveMacenski, 2019-08-02:
Update: I found if I nuked the TF related Debians I had installed, then I could get it to build, I think TF is linking sometimes against the wrong version of the library even when being built in the same workspace. 

Soln for temp use:
- nuke tf debians
- build geometry2 on ros2 branch in ros2_nav_dependencies

bradsquicciarini, 2019-08-07:
To provide a somewhat simpler workaround to get it built, you can replace your `tf2` and `tf2_ros` system headers and libraries.

**Replace libraries**
```
NAV_DEPEND_WS=<nav_ws>
sudo cp $NAV_DEPEND_WS/ build/tf2/libtf2.so /opt/ros/dashing/lib/libtf2.so`
`/opt/ros/dashing/lib/libtf2_ros.so`

## 
```
sudo rm 
```

bradsquicciarini, 2019-08-07:
To provide a somewhat simpler workaround to get it built, you can replace your `tf2` and `tf2_ros` system headers and libraries.

**Replace libraries**
```
NAV_DEPEND_WS=<path to  ros2_nav_dependencies_ws>
sudo cp $NAV_DEPEND_WS/build/tf2/libtf2.so /opt/ros/dashing/lib/libtf2.so
sudo cp $NAV_DEPEND_WS/build/tf2_ros/libtf2_ros.so /opt/ros/dashing/lib/libtf2_ros.so
```
**Replace headers**
```
NAV_DEPEND_WS=<path to  ros2_nav_dependencies_ws>
sudo cp -r $NAV_DEPEND_WS/install/tf2/include/tf2 /opt/ros/dashing/include/
sudo cp -r $NAV_DEPEND_WS/install/tf2_ros/include/tf2_ros /opt/ros/dashing/include/
```

SteveMacenski, 2019-08-08:
We appear to be building again with recent PRs

shivaang12, 2019-09-23:
@mlherd 

I am having the same issue when building from scratch. I tried to build tf2_ros branch ros2 from source in outerlay_ws but during the build it uses the old package in underlay workspace. How would I proceed further?

Best,
Shivang

mlherd, 2019-09-25:
@shivaang12 It has been a while not building the master branch. I will test it and let you know if I get the same problem.

mlherd, 2019-09-25:
@shivaang12 Today, I tested building Nav2 Master and Ros2 Master. All the packages were successfully built. Make sure to follow the exact order below.

1- Build ROS 2 Master - make sure you clone ROS 2 master
2- Source ROS 2 Master setup.bash
3- Build Navigation 2 Dependencies 
4- Source Navigation 2 Dependencies
5- Build Navigation 2 Master -  make sure you clone Navigation 2 master

shivaang12, 2019-09-26:
@mlherd 

Thanks! It worked after I removed the ROS2 dashing which I had install using apt, and build it from scratch. Sorry for the trouble.

Best,
Shivang