ros-navigation/navigation2 #967: AMCL fails in OSRF CI with "Promise already satisfied" exception

crdelsey, 2019-07-22:
## Bug report

**Required Info:**

- Operating System:
  - 
- Installation type:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue
View CI output here
http://build.ros2.org/view/Eci/job/Eci__nightly_navigation2_ubuntu_bionic_amd64/

#### Expected behavior

#### Actual behavior

#### Additional information
[amcl-6] [WARN] [amcl]: Waiting for map....
10:03:04 10: [amcl-6] terminate called after throwing an instance of 'std::future_error'
10:03:04 10: [amcl-6]   what():  std::future_error: Promise already satisfied
10:03:04 10: [ERROR] [amcl-6]: process has died [pid 19138, exit code -6, cmd '/tmp/ws_overlay/install_isolated/nav2_amcl/lib/nav2_amcl/amcl __node:=amcl __params:=/tmp/ws_overlay/install_isolated/nav2_bringup/share/nav2_bringup/launch/nav2_params.yaml'].

orduno, 2019-07-22:
Are there futures in AMCL? I don't recall using actions.

This might be an [upstream issue](https://github.com/ros2/rclcpp/issues/783).

I was getting the same issue when updating the Global Planner Tester #927. On my case, getting the result from the client instead of the goal handle seems to solve the issue.

``` cpp
auto future_goal_handle = client->async_send_goal(goal);

// check if the goal was sent and accepted

auto goal_handle = future_goal_handle.get()

auto future_result = planner_client->async_get_result(goal_handle);
```

crdelsey, 2019-07-22:
I see the exception occur in world_model as well, so it does seem like it could be an upstream issue. We'll need to keep an eye on that one.

SteveMacenski, 2019-08-08:
https://github.com/ros2/geometry2/pull/143

Seems resolved, we can reopen if we see again

closed 2019-08-08 by SteveMacenski