ros-navigation/navigation2 #971: Support multiple robots navigating simultaneously

orduno, 2019-07-22:
## Feature request

#### Feature description

We'd like to be able to support multiple robots navigating simultaneously (both in simulation and real hardware) -- no coordination at this moment, more on that coming soon :)

For this we need to:
- [ ] If simulation, have Gazebo launch multiple robots (on a single instance)
- [ ] Define a partitioning method (grouping services, actions, etc. from a single robot)
- [ ] Facilitate launching multiple robots.
- [ ] Visualize multiple robots in RVIZ. Extend plugins to provide goals to multiple robots.

#### Implementation considerations

For partitioning, we have a few options:
- Logical
  - Namespaces
  - DDS Keyed Data (not exposed by ROS2)
  - DDS Domain and Participants (not sure if exposed by ROS2)
- Physical

Most users in ROS1 used separate namespaces for different robots (together with a multi-master pkg). We can replicate that as a starting point, however, there are several reasons we might want an alternative approach, i.e. DDS discovery, Robot encapsulation, etc.

SteveMacenski, 2019-07-23:
I’m curious why this is given high priority? Running multiple navstacks on multiple robots as is is simple- its mostly a simulation need. Is there someone trying to do this right now we’re blocking?

crdelsey, 2019-07-23:
We have some bugs relating to being able to partition robots into separate namespaces #746.

We'd also like to be able to run multiple robots in gazebo to get better test coverage. We'd get to run the stack more per test run and also the other robots can act as dynamic obstacles to each other.

SteveMacenski, 2019-07-24:
Ok, that makes sense, but that also seems like a deterministic nightmare to maintain, but it's good capability to have though in in the N+1 timeframe.

orduno, 2019-10-01:
Tasks completed so closing the ticket.

closed 2019-10-01 by orduno