ros-navigation/navigation2 #973: Generated message files are include in Code Coverage

ruffsl, 2019-07-22:
## Bug report

As per the title, generated message files seem to be include in Code Coverage analysis.

**Required Info:**

- Operating System: Ubuntu 18.04
  - 
- Installation type: source
  - 
- Version or commit hash: d076e5d

#### Steps to reproduce issue
Example of such uploaded analysis can be seen from the CI job that synced to Codecov.io:

https://circleci.com/gh/ros-planning/navigation2/1799

#### Expected behavior

I suspect anything in the workspace build folder should be excluded from the code coverage calculation.

#### Actual behavior

[image]

#### Additional information

See here for current example:
* https://codecov.io/gh/ros-planning/navigation2/tree/master/build
* https://codecov.io/gh/ros-planning/navigation2/compare/d076e5d23e03dadfff91db4f199f71698892da9c...bd4dd9a5be01aeb5b236bad2b2d27a1b2ce65447/tree

crdelsey, 2019-07-24:
I've noticed that as well. I've removed the build directory from the coverage report, and it is gone if I look at the data locally, but somehow codecov.io still picks it up. Their report doesn't match the one I get with `genhtml`.

ruffsl, 2019-08-10:
I've cleaned up the upload pipeline for my other pr, and now I'm no longer see this issue:

[image]

https://codecov.io/gh/ruffsl/navigation2/tree/26cc2c9de8cb78652ff437d6b3dcbdd12bcc08ca

ruffsl, 2019-10-01:
Hmm, now that the debug build is up again, this issue seems to have come back.

https://codecov.io/gh/ros-planning/navigation2/tree/0df1713bcb4bbe43471c9ed45cae84065bfb8875

crdelsey, 2019-10-01:
I see what's happening. I'm not sure of the fix yet.

Lets look at `build/nav2_msgs/rosidl_generator_cpp/nav2_msgs/action/back_up__struct.hpp`, for example.

In the extract step, the log shows
```
Extracting /opt/overlay_ws/install/nav2_msgs/include/nav2_msgs/action/back_up__struct.hpp
```
*It thinks the file is in the `install` directory, not `build`.*

So when we get the step where we remove `build/*`, well, it's not there and doesn't get removed.

It makes sense that the coverage tool thinks the file is in `install` because it is referenced from there in other packages.

I'm not sure how it ends up showing up under `build` in the final report though.

SteveMacenski, 2019-11-06:
@crdelsey would it be possible to just remove `nav2_msgs` from coverage computation? Its not like we actually need test coverage over the message markdown files

crdelsey, 2019-11-11:
> would it be possible to just remove nav2_msgs from coverage computation?

Let's find out #1340.

I was starting to look at this during the tiny ticket sprint, but the build was too broken at the time to get useful code coverage results.

PR #1340 (2019-11-15): Attempt to filter out generated message files from code coverage.

---

## Basic Info

| Info | Please fill out this column |
| ------ | ----------- |
| Ticket(s) this addresses   | fixes #973 |
| Primary OS tested on | NA |
| Robotic platform tested on | CI |

---

## Description of contribution in a few bullet points

* Add filtering rules to remove any files in the _msg packags, so they don't affect coverage results.
* Also, reduce the number of tests that Travis runs to improve the odds of completing in time.
closed 2019-11-15 by crdelsey