ros-navigation/navigation2 #974: world_model: Error occurred while doing error handling.

ZakariaChekakta, 2019-07-24:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Source  
- Version or commit hash:
  - Dashing-devel hash 756265f 
- DDS implementation:
  - rmw_opensplice_cpp

#### Steps to reproduce issue
Follow steps in https://github.com/ros-planning/navigation2/tree/dashing-devel/nav2_bringup

#### Expected behavior
[world_model] [INFO] [world_model]: Activating
#### Actual behavior

[world_model-4] [WARN] []: Error occurred while doing error handling.
[world_model-4] [FATAL] [world_model]: Lifecycle node entered error state
[lifecycle_manager-1] [ERROR] [lifecycle_manager]: Failed to change state for node: world_model
[lifecycle_manager-1] [ERROR] [lifecycle_manager]: Failed to bring up node: world_model, aborting bringup

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

crdelsey, 2019-07-24:
This seems likely to be temporary DDS error. How often do you see this?

ZakariaChekakta, 2019-07-25:
@crdelsey , I always see it, even when I changed the DDS implementation to rmw_connext_cpp, and I have tried also the default one FastRTPS

crdelsey, 2019-07-29:
@ZakariaChekakta I'm am seeing this problem now as well. I am investigating.

hyunseok-yang, 2019-08-27:
I have a same issue with rmw_opensplice

hyunseok-yang, 2019-09-19:
Now it's working well with nav2_0.2.4 on OpenSplice

closed 2019-09-19 by SteveMacenski