ros-navigation/navigation2 #975: No response on running map_server

shikharvaish28, 2019-07-24:
I have built and sourced the ```navigation2```. After sourcing using ```source build/navigation2_ws/install/setup.bash```, I run 
```bash
./build/navigation2_ws/build/nav2_map_server/map_server /home/hipert/Downloads/random/navigation2/map.yaml
```
and the terminal gets stuck on this 
```bash
[INFO] [map_server]: Creating
```

The content of my ```map.yaml``` file is 
```yaml
image: map.pgm
resolution: 0.05
origin: [-9.38389, -11.5719, 0.0000]
negate: 0
occupied_thresh: 0.65
free_thresh: 0.196
```

I have tried with the ```map.pgm``` as well but nothing seems to work.
 Also, on running the ```rviz``` I cannot see the topic of map add.

 Can anyone help me with it? What is the expected nature of the map server?

ZakariaChekakta, 2019-07-24:
#864

crdelsey, 2019-07-24:
@mjeronimo What do you think of adding some extra messages to the lifecycle nodes to inform users that a lifecycle transition is expected? eg. At the end of the constructor, we could print something like
"Node constructed. Awaiting transition to lifecycle configured state. If running this without lifecycle manager, please run 'ros2 run nav2_util lifecycle_bringup map_server'"

SteveMacenski, 2019-07-24:
@crdelsey I thought as part of some of the early discussions on lifecycle that the default was to the active state for all nodes so this wouldn't be an issue for users? I think that lifecycle is something that should be abstracted from essentially all users unless they're looking to "pull back the curtain" or a pro-user.

crdelsey, 2019-07-25:
@SteveMacenski I agree it would be best if users didn't need to know about lifecycle unless they wanted to. I think that is mostly the case if they run navigation using the launch files provided. But if they run a node directly as OP did, that's not the case.

The launch files take care of lifecycle by setting a param on the `lifecycle_manager`, telling it to automatically start all nodes. Without `lifecycle_manager`, they don't autostart.

I had the beginnings of an idea to fix this. I just drew this up to help explain.

[image]

As you know, all the nav2 nodes are separated into a library and a executable that instantiates the node from the library. I was thinking, all the executables could also instantiate a lifecycle manager node that autostarts them. Then, when someone does a ros2 run, they launch the executable, which autostarts.

If we don't want that behavior, we can use the composed node functionality in launch, which, I think, will just pull in the library and instantiate the node. I think it provides its own "main" effectively.

SteveMacenski, 2019-10-10:
Should this ticket be renamed/closed?

crdelsey, 2019-10-14:
Opened new issue #1240.

closed 2019-10-14 by crdelsey