ros-navigation/navigation2 #976: [Question] Is AMCL working on the latest commit?

vinnamkim, 2019-07-24:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - source
- Version or commit hash:
  - [latest commit](https://github.com/ros-planning/navigation2/commit/2c17640b568085cdd20e9420bd8f38cc05508ec7)
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp(latest master branch)

#### Steps to reproduce issue
1. Launch TurtleBot3 simulation
```
(Terminal 1) ros2 launch nav2_bringup nav2_simulation_launch.py
(Terminal 2) ros2 topic echo /amcl_pose
```
2. Click Startup button of Navigation2 gadget in Rviz
3. Give initial pose of the robot on the map

#### Expected behavior
`/amcl_pose` should be being published continously.
#### Actual behavior
`/amcl_pose` is published just once and no response.

crdelsey, 2019-07-24:
The master branch seems very broken right now. I believe the problem is in the upstream ROS 2 master, but haven't confirmed the exact cause.

I believe the best workaround right now is to use the latest ROS 2 dashing release instead of their master branch.

crdelsey, 2019-07-24:
Related to #961

vinnamkim, 2019-07-24:
@crdelsey Is there a stable branch of Navigation2 which is compatible with the latest ROS 2 dashing "release"?

crdelsey, 2019-07-24:
We are planning to update dashing-devel at the end of this month. I will make a test branch today and I will post it here if you want to try it.

I believe our master branch is compatible with the ROS 2 dashing release if you `git revert c15c93fcec944a`

mkhansen-intel, 2019-07-24:
@crdelsey - shouldn't we be telling people to use the 0.2.2 tag? That should be verified as working with ROS2 dashing-devel, right?

crdelsey, 2019-07-24:
Yes. 0.2.2 is the latest stable release. I assumed @vinnamkim meant 0.2.2 was not working for him and he wanted the latest and greatest code.

@vinnamkim What did you mean by 
> I tried dashing-devel branch but it seems not helpful.

If you do want to try the latest changes, an early version of 0.2.3 is available on the `dashing-sync` branch. This is a temporary branch and is only lightly tested at the moment. 0.2.2 (the latest on `dashing-devel`) is the recommended version.

mkhansen-intel, 2019-07-24:
@vinnamkim - if you're using 0.2.2, then AMCL should be working. What ROS2 version are you using? You should be using the ROS2 dashing release tag also.

vinnamkim, 2019-07-25:
@crdelsey @mkhansen-intel 

> I tried dashing-devel branch but it seems not helpful.

I made mistake on building the dashing-devel branch with the ROS2 dashing release. So, I wrongly thought it's not build-able. Sorry for confusing.

crdelsey, 2019-08-09:
@vinnamkim Everything seems to be working now on both the master branch and the dashing-devel branch. I'm closing this now.

closed 2019-08-09 by crdelsey