ros-navigation/navigation2 #978: Parameterizing all the hard coded values such as timeout values in the stack

mlherd, 2019-07-24:
Parameterizing timeout values in the code has been discussed in #937 and during the weekly meetings. I am opening this issue up for further discussion.

Some questions for the discussion.

- Should we parameterize all the hard coded magic numbers?
- What is the best way to parametrize these values, so we don't end up having a lot of parameters in one file? Maybe categorizing them as required parameters, optional parameters or advanced parameters etc.
- Where should we set these parameters nav2 params file, launch etc?

SteveMacenski, 2019-07-24:
I believe we should definitely parameterize every number. There shouldn’t be hardcoded values unless that number has a verifiable and correct value for all cases. An example is that where are frequently magic numbers in motion planning derived from a more complex expression but its a number for which no one would change because its qualitatively determined. 

Before categorization, the comment about maybe a parameter != ros parameter. Is there a class of parameters we enable but don’t expose to a ROS API? I haven’t thought about that one before. 

In categorization, I think first a good step would be to create a generally complete list of parameters that people believe are on the edge of “should this be a parameter or not” to see if this is even an issue needing further addressing. We discussed DDS-timeouts and TF-timeouts. What else is out there? If its just those 2, adding 2 “quality” parameters to each node is reasonable to me. Nearly every TF consuming node in ROS1 will expose the TF timeout as a param so people are used to that but with a reasonable default most people probably don’t need to move too much or at all. The additional DDS timeout is a “cost of doing business” to me in ROS2 people need to get used to and again if we make the defaults rational then most people just won’t touch it. 

Parameters I think should still be in yaml files. The single file we have is excessive but I don’t think at this stage its worth trying to focus on minutia like that. Global Parameters  might go in the launch file directly to be written once and distributed to all nodes but I can’t think of many of those except sim time, and since the new launch system is python, we can add a argparse flag for it.

SteveMacenski, 2019-10-11:
Update: PR coming that fixes all magic numbers except timeouts. Timeouts I think we want all nodes to have a `tf_timeout` parameter similar to `use_sim_time`, which can be set globally at launch or by configuration file

SteveMacenski, 2019-11-06:
#1070 being packed into this ticket, they're asking for the same thing

Marwan99, 2020-05-08:
@SteveMacenski suggested I attempt working on this ticket. I currently looking into parameterising the tf timeouts then will move to services and actions timeouts.

Below is a list  of files where a tf buffer is accessed to retrieve a transform. The numbers represent the line number in the respective file. If all is good, i will proceed by parameterising all the ones currently using hard coded values. Please let me know if I missed anything or  if a current buffer lookup not using a timeout should be changed to use a timeout .
 
* [amcl_node.cpp](https://github.com/ros-planning/navigation2/blob/master/nav2_amcl/src/amcl_node.cpp)

    Not using a timeout:
    - 570 (should this be using a timeout??)
    - 441, 752, 807, 808, 1006

    Using hard coded timeout: 246

    *Note: AMCL currently has a parameter named transform_tolerance used for publishing the map to odom tf in the future*  

* [static_layer.cpp](https://github.com/ros-planning/navigation2/blob/master/nav2_costmap_2d/plugins/static_layer.cpp) 

    Not using a timeout: 362    

* [costmap_2d_ros.cpp](https://github.com/ros-planning/navigation2/blob/master/nav2_costmap_2d/src/costmap_2d_ros.cpp) 

    Using the transform_tolerance parameter as timeout value: 535

    Not using timeout: 202

* [observation_buffer.cpp](https://github.com/ros-planning/navigation2/blob/master/nav2_costmap_2d/src/observation_buffer.cpp)

    Not using timeout: 98,  102,  138,  149

    Using the transform_tolerance parameter as timeout value: 75     

* [nav2_costmap_2d/src/collision_checker.cpp](https://github.com/ros-planning/navigation2/blob/master/nav2_costmap_2d/src/collision_checker.cpp)

    using hard coded timeout in nav2_util::getCurrentPose declaration: 176

* [nav2_dwb_controller/dwb_core/src/dwb_local_planner.cpp](https://github.com/ros-planning/navigation2/blob/master/nav2_dwb_controller/dwb_core/src/dwb_local_planner.cpp)
    
    Using the transform_tolerance parameter as timeout value: 278,  286,  352,  530,  602 

* [nav2_behavior_tree/plugins/condition/transform_available_condition.cpp](https://github.com/ros-planning/navigation2/blob/master/nav2_behavior_tree/plugins/condition/transform_available_condition.cpp)

    Not using timeout: 68, 73

* [nav2_behavior_tree/plugins/condition/goal_reached_condition.cpp](https://github.com/ros-planning/navigation2/blob/master/nav2_behavior_tree/plugins/condition/goal_reached_condition.cpp)

    using hard coded timeout in nav2_util::getCurrentPose declaration: 73

* [nav2_recoveries/plugins/back_up.cpp](https://github.com/ros-planning/navigation2/blob/master/nav2_recoveries/plugins/back_up.cpp)
    
    using hard coded timeout in nav2_util::getCurrentPose declaration: 55, 66

* [nav2_recoveries/plugins/spin.cpp](https://github.com/ros-planning/navigation2/blob/master/nav2_recoveries/plugins/spin.cpp)
    
    using hard coded timeout in nav2_util::getCurrentPose declaration: 72, 90

SteveMacenski, 2020-05-08:
AMCL
- 570 is looking at a specific timestamp, no timeout required
- 441 is using a specific timestamp,  no timeout required
- 752 should include a timeout
- 807/808 is using a specific timestamp,  no timeout required
- 1006 is using a specific timestamp,  no timeout required
- 246 should include a timeout

Static
- 362 should use a timeout

Costmap 2d
- 535 is correct (need to standardize param name)
- 202 is looking for the most recent just to know if its available, thats OK as is

Observation
- 75 is correct (need to standardize param name)
- 98/102 should use a timeout 
- 138/149 is using a specific timestamp,  no timeout required

Collision checker
- 176 uses the getCurrentPose which has a transform tolerance baked in 

DWB
- is correct (need to standardize param name)

Transform available
- I don't see a TF call at 68
- 73 I'm debating. I think the intent is just to tell if there is a transform at all, so I think this is fine, but also it could be worthwhile for this to block for a very short period of time on startup to see if one is now available. What do you think?

Goal reached
- 73 uses the getCurrentPose which has a transform tolerance baked in

Backup
- uses the getCurrentPose which has a transform tolerance baked in

Spin
- uses the getCurrentPose which has a transform tolerance baked in

---

From this audit, it looks like `transform_tolerance` is a pretty standard name for what we're looking at. We should use that name then. 

Any usage of getting the map->base_link or map->odom transform, you can update to use the `getCurrentPose` method. For the calls of this, do you think we should also parameterize beyond that? 

Finally, I'd also do a little benchmarking before and after updating these timeouts. I am a little worried about the static layer and observation layer timeout. I tried to track down in TF where the tolerance is used in look up to see if in the case of currently available if adding that tolerance would be more costly but I couldn't find it. I found where `canTransform` looks that up and there is a little extra cost, but not where the transform itself happens.

SteveMacenski, 2020-05-15:
PR imminent to land.

closed 2020-05-15 by SteveMacenski

pschalkwijk, 2020-06-19:
In amcl_node, line 572: 
`tx_odom = tf_buffer_->lookupTransform(
      base_frame_id_, tf2_ros::fromMsg(msg.header.stamp),
      base_frame_id_, tf2_time, odom_frame_id_,);`

should the transform_tolerance_ be used in this lookupTransform as well?

`tx_odom = tf_buffer_->lookupTransform(
      base_frame_id_, tf2_ros::fromMsg(msg.header.stamp),
      base_frame_id_, tf2_time, odom_frame_id_,transform_tolerance_);`