ros-navigation/navigation2 #980: Navigation 2 Stack Documentation Main Structure

mlherd, 2019-07-25:
The documentation is an important part of this stack. In my opinion it should;

- explain the main parts of stack in details
- be easy to search
- be easy to read
- be easy to maintain
- include many tutorials and examples
- serve all user types

@SteveMacenski posted his feedback for the tutorial that I created for TB3 [here](#896). I used his suggestions, our team's input, and the Navigation 1 stack as an example to create the structure below. This is just a proposal for the main documentation structure and feel free to share your feedback. 

In addition to all Github Pages is a candidate for documentation for now, but again we are open to any suggestions.

 - Overview
      - What is Navigation 2?
      - What is new? Nav2 vs Nav1.
      - Release information
 - HW/SW Requirements
 - Install and Build Guides
     - For users
		 - Install and Build Instructions
				     - Install Binary
                    - Build from source
          - Tutorials
                    - Simple Robot in Gazebo
                    - Turtlebot 3
                    - Turtlebot 2 etc.
                    - Tuning up the Navigation Stack for your Robot
           - Parameters
     - For Developers         
		 - Installation and Build Instructions
                     - Build from source (Master)
                     - Debugging and Testing
                     - Navigation 2 Stack Architecture
			- World Model
			- DWB
			- NAVFN                   
            - BT Navigator  etc.
  - Contribute
	  -  Help Wanted 
	  - Report a bug
 - Troubleshooting Guide
- Applications, Publications and Videos

SteveMacenski, 2019-07-26:
If you're looking at this, I would also recommend making a nice clean graphic that can be used for an understanding [this move base figure](https://user-images.githubusercontent.com/14944147/61925010-1ef91c80-af1f-11e9-89cc-bf089b67166b.png) is the defacto explanation of navigation1. Having something to point to would be great.

mlherd, 2019-07-26:
I created a [UML diagram](https://raw.githubusercontent.com/mlherd/ros2/master/img/nav2_uml_diagram.jpg) that shows how things are connected in the stack, but I guess it would be nice if I create another figure which is more high level and similar to the move base. It can be helpful when comparing Nav1 and Nav2 as well. I will be working on that.

mlherd, 2019-07-29:
The links to the some of the experiments that I have done using different documentation options.

- [Readme.md](https://github.com/mlherd/navigation2/tree/new-tutorials/doc/tb3_tutorial)
- [Github Wiki](https://github.com/mlherd/ros2/wiki)*
- [Github Pages](https://mlherd.github.io/nav2_ghp_test/)*

*testing purposes only, so the content may not be correct

SteveMacenski, 2019-07-29:
I'm a fan of option 2 (GitHub wiki) of that bunch.

A nice figure would be great as a explanation, though I dont think you need add rviz/gazebo because those aren't really the "real" tools you use for making your autonomous robots, those are just for testing around. But it is illustrative of a intro-demo so that probably has value.

closed 2019-12-06 by mkhansen-intel