ros-navigation/navigation2 #985: Reliable and robust networking

orduno, 2019-07-29:
## Feature request

#### Feature description

Expanding what we have in #592 on the topic of networking reliability,  let's discuss what we can improve in this area. Here's some to start with:

- [ ] Check for timeouts at every service/action call and fail gracefully
- [ ] Verify we are receiving messages at a reasonable frequency

Please feel free to edit the ticket and add your ideas.

mhpanah, 2019-07-29:
@orduno This is related to the first two items of #953.

SteveMacenski, 2020-05-15:
This now covers doing the testing and service/action timeout - TF has been taken care of via #978

naiveHobo, 2020-05-17:
Service timeouts are already implemented in the ```check_future``` method which is called in ```tick()```:
https://github.com/ros-planning/navigation2/blob/fc762f659d98ce4ca2ebc5d87e6fc556742ffc9a/nav2_behavior_tree/include/nav2_behavior_tree/bt_service_node.hpp#L98-L114

SteveMacenski, 2020-05-18:
OK, I know a few are going to use this, but we need a survey of all of them to make sure that all of them do / will have a timeout on the `spin_until_future_complete()`s -- or are using the asynchronous callback methods which I think I only did in the BT action node. You'll find service callers in waypoint navigator and recovery server too.

SteveMacenski, 2020-06-19:
@naiveHobo you get a chance to finish this and there was no action?

SteveMacenski, 2020-07-07:
Pinging again @naiveHobo

SteveMacenski, 2020-09-25:
Things requiring timeouts (in spin until future complete)
- nav2_behavior_tree/include/nav2_behavior_tree/bt_action_node.hpp
- nav2_lifecycle_manager/src/lifecycle_manager_client.cpp
- nav2_bringup/nav2_gazebo_spawner/nav2_gazebo_spawner/nav2_gazebo_spawner.py
- nav2_rviz_plugins/src/nav2_panel.cpp

After which this ticket is complete

closed 2020-09-28 by SteveMacenski