ros-navigation/navigation2 #992: Temporary: Place world model costmap in the navFn process

SteveMacenski, 2019-07-30:
At the moment we only support costmaps and NavFn planners in ROS2, while that's going to change, for the immediate to intermediate term, it seems like we'll be with that for awhile. There's no other users of the global costmap & is a direct analog to the local costmap with DWB architecture.

I propose this to minimize networking traffic and lower latency. Long term the planner and world model (whether navFn, a plugin, whatever and costmap, grid maps, whatever) should be in the same process since the global costmap can be quite large. 

If folks agree, I'll do it.

SteveMacenski, 2019-07-30:
Additionally it will allow them to share a TF buffer which will get us down to only 3 buffers which is pretty good IMO

mkhansen-intel, 2019-07-30:
I was just talking to some of the team about this recently. I agree that in it's current state, it's not really a world model useful for anything but NavFn, and we haven't had time to work on getting it to a well designed world_model, so I'm in agreement with you at least. Should we discuss in the working group before giving a green light?

SteveMacenski, 2019-07-30:
Yeah sounds good

orduno, 2019-07-31:
>  Long term the planner and world model (whether navFn, a plugin, whatever and costmap, grid maps, whatever) should be in the same process since the global costmap can be quite large.

I guess eventually planner, controller, and world model will be separate nodes but composed into a single process. I'm ok with making this temporal change.

SteveMacenski, 2019-07-31:
I dont necessarily think that planner and controller need to be in the same process, if they're not sharing a single world model. I haven't heard someone very seriously discussing breaking down the local/global costmaps into 1 costmap. I'm still a fan of that method but I haven't rectified in my mind yet how that would ideally go since it would require:
- Multiresolution storage (local costmaps are usually higher resolution than global)
- Differing update rates or buffers for local "close" and far data efficiently
- keeping the odometric and global pose frames from aliasing 

But I would agree that long term the [thing using a world model to complete a task] and the world model itself should be in the same process so that there's no overhead in getting info from very large maps. 

Lets bring this up in the weekly tomorrow but I've started a really basic transition to make it a slow conservative move (ei dump in same process with 2 nodes to minimize serialization/DDS, then more deeply integrate thinking about how to best manage the lifecycle elements).

SteveMacenski, 2019-08-29:
Done here: https://github.com/SteveMacenski/navigation2/tree/model_navfn

Only blocker is that the planner tests assume that you can just expose a server for the planner to use to grab a costmap and that makes this far more difficult to quickly get the test running, it would be faster to start over and it's not in my scope right now to patch tests.

closed 2019-09-10 by SteveMacenski