ros-navigation/navigation2 #997: Reduce dependency workspaces down to core needs

SteveMacenski, 2019-07-31:
Right now we're building an (insane) number of things, in potentially up to 5 workspaces to get things to work. We should start working to simplify now that many things are being released through the build farm now. I've started auditing and it seems that *nearly* everything is available from the `ros2_ws` in apt with the exception of (seemingly) move_base_msgs (which is a simple port I'm going to take care of right now). In the `ros2_nav2_dependencies` same digs. This also means **no more sourcing 4 files**

Thoughts?

SteveMacenski, 2019-07-31:
My port of move_base_msgs https://github.com/ros-planning/navigation_msgs/pull/10/files

I think that should* be everything for building without any additional workspaces

crdelsey, 2019-08-01:
>  with the exception of (seemingly) move_base_msgs

We don't have a dependency on `move_base_msgs` to the best of my knowledge. Where does that come in?

> no more sourcing 4 files

During the initial build you have to source each workspace in turn. But then, running it, I just source the last workspace, and it brings in the others. For most subsequent builds I just source that last workspace as well. This works fine unless
1. One workspace has packages that override another.
2. You make a big enough change that packages are added or removed.

> We should start working to simplify now that many things are being released through the build farm now

If you are working on the dashing-devel branch, it can be simple already.
```
First build only:
> sudo apt install ros-dashing-desktop ros-dashing-gazebo-ros-pkgs
> source /opt/ros/dashing/setup.bash
> cd to nav_ws
> rosdep install --from-paths nav_ws/src --ignore-src
> colcon build
```

```
Subsequent builds and running:
> cd to nav_ws
> source ./install/setup.bash
> ros2 run, or colcon build
```

Ok. It's still not that simple, but I think the only way to make it simpler is to install the binary release
```
> sudo apt install ros-dashing-navigation2
```

If you want to build nav2 master, against ros2 master, I believe all the workspaces are still necessary in certain circumstances.

SteveMacenski, 2019-08-01:
`move_base_msgs` is built in `ros2_ws` in the `ros-planning` directory. I dont think we need it, but that's the only package I could find in the `ros2_ws` that wasn't just available in apt for install. 

The point being that a typical developer shouldn't need to have to build 300 packages from source just to develop in navigation2, which is what the initialization and build scripts assume. The target distro is Dashing so we can use the dashing releases in apt to greatly simplify our builds. Sourcing from /opt/ros is also much faster for various reasons. 

... does that work? That's what i was looking for as the default: install dependencies from apt (rosdistro) and just need the metapackage in 1 workspace to develop, no other source workspaces. My impression was there we're using enough bleed from other packages that we can't meaningfully develop without all this extra garbage around when we're targeting a stable API.

SteveMacenski, 2019-08-01:
So more testing...

it appears* that you don't need the ros2_ws at all when we install from apt -- can we remove that from the init / build scripts? Now that we're using a stable API there shouldn't be a huge number of reasons for navigation2 to need to build from fresh source. 

For new release candidate (beta for E, etc) its trivial to change the apt url to `shadow` to get bleed builds from apt on the ros build farm

closed 2019-08-01 by SteveMacenski