ros-navigation/navigation2 #998: Document & update tools for building and sychronizing workspaces

SteveMacenski, 2019-08-01:
It appears the build scripts are out of date, we should update them with the current best workflow, and deal with sychronization between workspaces + scripts to automate pulling latest changes

crdelsey, 2019-08-02:
Agree that this needs to go in the docs. Let me write the procedure up here quickly for reference:

## Building the dashing-devel branch on Ubuntu

### First time

1. Install ROS 2 from Debian packages (https://index.ros.org/doc/ros2/Installation/Dashing/Linux-Install-Debians/)
2. `mkdir -p ~/nav2_ws/src`
3. `cd ~/nav2_ws/src`
3. `git clone` this repo and `git checkout dashing-devel`
4. `source /opt/ros/dashing/setup.bash`
5. `cd ~/nav2_ws`
6. `rosdep install -y -q --from-paths src --ignore-src`
7. `colcon build --symlink-install --merge-install`

### Subsequent builds

1. `cd ~/nav2_ws`
2. `source ./install/setup.bash`
3. `colcon build --symlink-install --merge-install`

## Building the master branch

### First time

#### 1. Setup Nav2 WS

1. `mkdir -p ~/nav2_ws/src`
3. `cd ~/nav2_ws/src`
3. `git clone` this repo and `git checkout master`

#### 2. ROS2 WS

1. Setup and build ROS 2 from source according to the official instructions https://index.ros.org/doc/ros2/Installation/Dashing/Linux-Development-Setup/ but with a couple of minor changes.

    a. Instead of downloading the `release-latest` version of the `ros2.repos` file, get the `master` version. eg. `wget https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos`

    b. This is optional, but I'd build a RELEASE version of the code with the `--merge-install` option instead of using the standard build command. eg. `colcon build --symlink-install --merge-install --packages-skip ros1_bridge --cmake-args -DCMAKE_BUILD_TYPE=Release`

#### 3. Navstack dependencies

1. `mkdir -p ~/ros2_nav_dependencies_ws/src`
2. `cd ~/ros2_nav_dependencies_ws`
3. `vcs import src < ~/nav2_ws/src/navigation2/tools/ros2_dependencies.repos`
4. `source ~/ros2_ws/install/setup.bash`
3. `rosdep install -y -q --from-paths src --ignore-src` (If your dependencies are already installed, you can skip this step)
5. `colcon build --symlink-install --merge-install --cmake-args -DCMAKE_BUILD_TYPE=Release`

#### 4. Build Nav2 WS
1. `cd ~/nav2_ws`
1. `source ~/ros2_nav_dependencies_ws/install/setup.bash`
2. . `rosdep install -y -q --from-paths src --ignore-src` (If your dependencies are already installed, you can skip this step)
2. `colcon build --symlink-install --merge-install --cmake-args -DCMAKE_BUILD_TYPE=Release`

### Subsequent builds

1. `cd ~/nav2_ws`
2. `source ./install/setup.bash`
3. `colcon build --symlink-install --merge-install`

SteveMacenski, 2019-08-02:
The ROS2 workspace (as I'm finding) is somewhat regularly broken, what's the recommended way to handle that?

What's the reason to source your install space between builds (Subsequent builds bullet 2)? Nothing in building should depend on the install space.

Is there anything we can do to make master run on dashing ROS debians? This TF/ament build issue has completely stalled me out for days and I don't see problems like this going away anytime soon.

crdelsey, 2019-08-02:
There are a few things you can do to speed things up, however, they all involve skipping steps. You need to understand the steps you are skipping and what has changed. In many cases these steps below won't work.

## Updating the ros2_ws or ros2_nav_dependencies_ws
1. You can just run `vcs pull` from the root of the workspace to fetch the latest code.
2. Then source the setup.bash in that workspace and invoke the colcon build step.

This works ok if the upstream branches haven't changed and repos haven't been added or deleted. I normally don't do this with the ros2_ws but I do with the ros2_nav_dependencies_ws.

For the ROS2 workspace, I normally delete the src, install, and build directories. Then I just do the following steps

1. `mkdir src`
2. `wget https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos`
3.  `vcs import src < ros2.repos`
4.  `colcon build --symlink-install --merge-install --packages-skip ros1_bridge --cmake-args -DCMAKE_BUILD_TYPE=Release`

## If working in one package
You can skip colcon if your updates don't impact multiple packages. Instead
1. cd to the build directory of that package eg. `cd ~/nav2_ws/build/nav2_amcl`
2. Invoke make directly. eg `make`. If you do `make help` it will show you all the make targets, so you could build individual targets if you want even.

SteveMacenski, 2019-08-02:
I'm not sure why you're sourcing the setup file of the workspace you're building before you build. I don't think that accomplishes anything? That was true with `catkin`, does `colcon` do something when you source the install space before building that same workspace?

crdelsey, 2019-08-07:
> What's the reason to source your install space between builds (Subsequent builds bullet 2)? Nothing in building should depend on the install space.

@SteveMacenski Sorry I'm slow here; I've been swamped lately. I source the `nav2_ws/install/setup.bash`  in subsequent builds merely for convenience. You are right. I could `source ros2_nav_dependencies_ws/install/setup.bash` instead, but I'm usually in the nav2_ws directory already, so the current workspace is more convenient. Also, the next step after building would be to source the nav2_ws so I can run it. I save that second `source`ing this way.

SteveMacenski, 2019-08-07:
I had a chat at Open Robotics today and we talked about having custom repo files for tracking some static state of master instead of being broken all the time or on the debian releases. The work flow would work something like this:

- Repos file targeting current state
- We all nuke our ros2_ws and dependencies and re-create using this at the same time
- We work on this 
- Every N weeks or on some regular interval, we retarget the current state and update the repos file, and recompile
- Then now every N weeks we deal with ABI compatibility / library change issues and we can do it all at once instead of wasting a bunch of time with breaking edge. 

OR said they can setup a custom config for us to use that repos file in CI to have 1 build farm of master (which will be failing but OK when ABI changes) and another of the current targeted versions (which should be successful)

thoughts?

SteveMacenski, 2019-10-17:
I think recent documentation PRs fix the spirit of my issue

closed 2019-10-17 by SteveMacenski