ros-navigation/navigation2 #1000: Takes a while for RVIZ to become responsive after launch

orduno, 2019-08-02:
## Bug report

**Required Info:**

- Operating System: Ubuntu 18.04
- Installation type: Source
- Version or commit hash: 2ede683356dc51cbd1732275aaacbf27ce3d0bd3
- DDS implementation: Fast-RTPS
- Client library (if applicable): rclcpp

#### Steps to reproduce issue
Launch the nav stack in simulation: `ros2 launch nav2_bringup nav2_simulation_launch.py simulator:=gzserver`

#### Expected behavior
As soon as Rviz is available, it is responsive to user inputs.

#### Actual behavior

It takes a few seconds after launch for Rviz to become responsive:

[image]

#### Additional information

If the robot model is disabled, then it is responsive:

[image]

Perhaps an option is to disable it by default.

mlherd, 2019-08-03:
I noticed this problem today as well. It seems like for some reason parsing URDF files takes longer than usual.

SteveMacenski, 2019-08-03:
We should look at the size of the model. If the dae or stl has too many triangles it can really take a long time to load- most people export their CAD stl’s too high.  

If we think its too large and that’s the issue, I can lower the resolution in my Autodesk license and upload a downsampled version to this repo.

SteveMacenski, 2019-08-05:
Yes, definitely the CAD models. the Waffle plates are 16MB, each tire is a MB, astra is 8MB, these are way too high resolution for visualization

mlherd, 2019-08-06:
I wonder if this causes rviz to have low fps as well.

SteveMacenski, 2019-08-06:
Good question, I haven't noticed but potentially the case. There was a robotics project I did a few years ago where I accidently exported a *1GB* DAE file and I wondered for days why it took RVIZ _minutes_ to load. I later was sending the file to a friend to test for me and realized on gmail attachment how large it was

crdelsey, 2019-08-07:
@orduno Is this fixed now that #1001 is in?

orduno, 2019-08-19:
Yes, will close the issue.

closed 2019-08-19 by orduno