ros-navigation/navigation2 #1002: Navigation2 Goal Tool hangs rviz for a few seconds

SteveMacenski, 2019-08-02:
The underlying call should spin a thread to complete the task, it shouldn't freeze the front end

mlherd, 2019-08-06:
It freezes for 3 seconds. This should only happen if something goes wrong in the navigation stack. I added the timeouts in #937. This was a temporary solution. Previously, Rviz's UI was getting frozen forever or stopped working. I think @bpwilcox is going to have a PR soon that will change/eliminate the rviz plugin, so this will no longer be a problem.

SteveMacenski, 2019-08-06:
Ok, well I put in a PR until its gone to spin up a thread to do the operation. To my knowledge that's good form for things in front-end that take longer than human perceivable time

mlherd, 2019-08-24:
I think PR #944 fixes this issue. I haven't tested it yet.

SteveMacenski, 2019-08-24:
:+1: it was on my list for this week to verify but if you've done it, awesome!

(I've been a little busy in some other parts of the ecosystem, I should be back in nav-ing in about a week)

closed 2019-08-24 by SteveMacenski