ros-navigation/navigation2 #1006: Opening gazebo, rviz, and navstack process in different orders will affect connections

SteveMacenski, 2019-08-05:
I can only get navstack to work reliably when gazebo is already open, and rviz after navstack is up.

Its probably DDS/RMW related that's out of our control but leaving the ticket to later come back to

mlherd, 2019-08-06:
They should be run in this order:

1- Gazebo
2- TB3 Bringup
3- Navigation Bringup
4- Rviz

It will probably not work if you run them in a different order.

SteveMacenski, 2019-08-06:
@mlherd the point I was making is that that's _bad_ and should not be the case. This is an issue needing to be resolved but its probably more deeply rooted than this repo but its a bug to start the tracking down process

mlherd, 2019-08-06:
I agree. I just wanted to add the right order here, so if someone reads this issue, they will at least know the right order.

SteveMacenski, 2019-08-24:
I did some more digging on this front, it really is going to be something so far below this project that its not really worth addressing here. A multitude of tickets across the ecosystem express a similar issue so its not worth keeping in our queue.

closed 2019-08-24 by SteveMacenski