ros-navigation/navigation2 #1008: Thoughts on removing simple navigation topic interface [Needs input]

SteveMacenski, 2019-08-06:
In Move Base, we could use the action server client or the `/move_base_simple/goal` topic to send a robot to go to a location.

There's now a navigation2 Goal tool which makes the action server client and works. The cases to send a goal without feedback could use the action client and then just ignore any returns. 

Any objection to just simply removing the use of topic-based navigation in favor of actions? The downside I see is that it is harder to just send a quick go-to-pose command through a terminal but the `ros2 action send_goal [name] [type] [yaml]` should be able to handle that now. I don't see a substantial downside

bpwilcox, 2019-08-07:
You've mentioned a downside of removing topic, even if not too substantial, can you articulate what the downside is of keeping the topic interface? Just want to compare pros/cons :)

SteveMacenski, 2019-08-07:
It's more things to maintain, which I try to avoid when possible... and right now its broken. Rather than fixing it, since its completely superseded we can remove it and simplify the API and removing a potential race condition if implemented/modified improperly.

crdelsey, 2019-08-08:
Having a simple topic interface to send the goal seemed like a nice-to-have feature. I'm not sure how widely it's used. Maybe we can deprecate it in some way, such that the community can complain if they think we should keep it?

Maybe we can print a deprecation warning to the console if we receive a message on the topic interface. Then, if no one complains, we can remove it for eloquent or f-release.

SteveMacenski, 2019-08-08:
Sounds good, except its still then broken and we need to fix it :(

SteveMacenski, 2019-08-08:
@crdelsey I'll do a PR right now to add that warning, I'm not going to have the cycles to debug the topic interface being broken at this very moment though

SteveMacenski, 2019-08-19:
done

closed 2019-08-19 by SteveMacenski