ros-navigation/navigation2 #1013: Register nodes as components

bpwilcox, 2019-08-07:
## Feature request
Use the `rclcpp_component` library to register the navstack nodes as components to be composed (or standalone) 

#### Feature description
We've discussed the potential for the stack to be configured into different processes by utilizing components in ROS2, yet none of our nodes have been registered as components. We should make this change to enable this use case. 

See https://index.ros.org/doc/ros2/Tutorials/Composition/

#### Implementation considerations

- Components need to have NodeOptions as the constructor argument, not sure what impact that has on the signature of the nodes in the stack.

shivaang12, 2020-01-15:
Do you any high level planning of how you want to approach this? I think we might need to split the nodes in order to make them compatible for running standalone as well as through component.

SteveMacenski, 2020-01-15:
They should all work to my knowledge just registering in the CMakelists and with the cpp macros. They all derive from the node type and take in `NodeOptions`.

SteveMacenski, 2021-09-15:
PR open and to be merged.

closed 2021-09-15 by SteveMacenski