ros-navigation/navigation2 #1014: [Process] Review Criteria and Responsible Parties

SteveMacenski, 2019-08-07:
We've decided a bit ago to have 2 approvals per PR to be merged, I think that's still a good process but I wanted to get it refined a _little_ more if possible. 

What I'm proposing is that for given packages there's a list of >2 people who are the "responsible parties" for that package to review and approve. Others are welcome to get involved, but that puts some structure about who we should nominally expect input from. Perhaps internally at Intel you all know who to chat with, but from the outside perspective I'm not necessarily sure who I should be requesting for given PRs. A little formality to it will lower the inbox spam of requesting PRs from folks that aren't the right people to poke.  

In the Wiki page (or somewhere else, doesn't matter) it would be nice to have a page with package names, and the names of nominal reviewers for it. 

For instance, I would be good for `costmap_2d` `navfn planner` `amcl` but probably not very helpful in `rviz plugins` and `dynamic params`.

bpwilcox, 2019-08-07:
I would agree with your sentiment, though a suggestion could be that you can always request that the maintainers of a certain package (provided in package.xml) review changes relevant to said package. Perhaps maintainers should be updated as well.

crdelsey, 2019-08-08:
We had quick discussion on this a couple of weeks ago, and it was kind of left at "I'm supposed to assign a couple of reviewers to each PR as it comes in". However, this isn't working out very well; I'm just too swamped to do this in a timely way.

At the beginning of the year, @mkhansen-intel and @mjeronimo were planning to step back from the day-to-day minutiae of the project to focus on bigger issues. That hasn't worked out so much in practice, but, I believe that is still the intent. They were to be included on reviews if we specifically wanted their input, but otherwise were to be left off. @mkhansen-intel, @mjeronimo is that still the case?

Unless they say otherwise, that leaves these people as the core set of reviewers:
 - @crdelsey 
 - @bpwilcox 
 - @orduno 
 - @mhpanah 
 - @SteveMacenski

SteveMacenski, 2019-08-08:
Got it, I always feel a little awkward scrolling through a list of you guys and just randomly selecting 2-3 without really knowing who might _actually_ care about that subsystem. Some loosely followed guidelines would just make me feel less awkward about just randomly requesting from someone active with a pulse.

Or if people just ignore the notifications and don't care when the PR isn't relevant to them, then I can just stop feeling awkward.

crdelsey, 2019-08-08:
What I originally wanted was for the PR submitter to just leave the reviewers blank, and then people would pick them up on a regular basis and review. Some of us do that, but not everyone.

I think that is still a reasonable thing to do unless there is someone in particular you think should review it. Every once in a while, I'll go through any that don't have enough reviewers and assign someone.

I haven't assigned any to you, Steve, since you review them without any prompting. I'll leave it that way unless you prefer something different.

SteveMacenski, 2019-08-08:
Honestly even if you assigned me, I don't even know how I'd find that out. But you're welcome to do so if there's something you want me to look at and I'm not responding. Basically if the PR is up for 2 days and I haven't responded, I'm probably not planning on it without a nudge. 

I think that's a reasonable thing to try: just not assigning anyone and letting people come to it, as long as people are consistently checking and picking up PRs. That feels like "organized chaos" though

mjeronimo, 2019-08-08:
@crdelsey Yes, the intention was for me to focus on determinism, safety, etc. However, I can still be available to review in various areas including lifecycle nodes, behavior trees, parameters, launch, rviz plugins, and overall architectural changes.

crdelsey, 2019-08-09:
> That feels like "organized chaos" though

I thought organized chaos was supposed to be a good thing.

Anyhow, as I look back at this issue, I see Brian and Carlos are both responding positively. That makes 3 of the 5 core reviewers in agreement with the idea of having per package reviewers. Let me sync with the team here again. I thought we rejected this idea, but maybe my memory is faulty.

crdelsey, 2019-08-12:
After talking to folks here, seems like people don't want to be pigeon-holed and would prefer to review stuff throughout the stack. So, I believe the process ought to be, leave the reviewers list blank unless there is someone specific you want to review that PR.

People should pick up the PR and review it on their own.

If it is languishing, feel free to add reviewers or email (or matrix, call, etc) me or other team members

SteveMacenski, 2019-08-12:
Got it, sounds good, closing for now

closed 2019-08-12 by SteveMacenski