ros-navigation/navigation2 #1015: High CPU usage even at idle

ruffsl, 2019-08-07:
## Bug report

The security working group has been creating a SROS2 demo targeting the turtlebot 3 with a running navigation stack. However our meager mortal laptops @mikaelarguedas and I have (~2017 Dell XPS i7 32GB) are struggling to run the tb3 demo while maintaining a usable desktop, as about about 40% of our CPU is under load even when the robot is not planning or moving:

[image]

I fear many participants in the workshop with lesser machines will struggle to run the demo with added security cipher overhead or unavoidable virtual machines on mac/windows as workshops happen to do when linux is involved. Is there anything we can do to lighten this up, like rate limit spinning in `dwb_controler`, `amcl`, `map_model`, or patches we can pull into our .repo file targeting dashing?

https://github.com/ros-swg/turtlebot3_demo

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - source
- Version or commit hash:
  - Patched from https://github.com/ros-swg/turtlebot3_demo/blob/master/docker/dependencies.repos
  - starting around 9cbda503593d91081ad255fcf2d06e09c9d9f760
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
See current tutorial README.md as of writing:
https://github.com/ros-swg/turtlebot3_demo/tree/ccc63b8462a714f6f870db97ae8797d167522463

#### Expected behavior
Ideal load for the navigation stack when not moving would be low like in ROS1, mostly dominated by gzviewer and rviz if not running in headless mode.

#### Actual behavior
The processes for `dwb_controler`, `amcl`, `map_model` take the lion's share of CPU, even with the robot is localized and not moving. Note that this is even without security or gzclient enabled.

[image]

#### Additional information

The above is just from launching these two launch files

``` shell
ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py
ros2 launch turtlebot3_navigation2 navigation2.launch.py use_sim_time:=true
```

Where gzclient can be disabled in `navigation2.launch.py` via:

``` diff
-            cmd=['gazebo', '--verbose', world, '-s', 'libgazebo_ros_init.so'],
+            cmd=['gzserver', '--verbose', world, '-s', 'libgazebo_ros_init.so'],
```

SteveMacenski, 2019-08-07:
@ruffsl are you using any custom config files or just the defaults?

If I'm reading this right: you're seeing 12% CPU usage for the world model (global costmap), DWB controller (which also is running a costmap), and AMCL (woah, that's shouldn't be even remotely that high). Frankly 12% CPU seems about right for the costmap items since that's spinning regardless if anyone's using it. AMCL shouldn't be 12% ever.

Any other errors/warnings like inability to transform laser scans? Particularly coming from AMCL?

mikaelarguedas, 2019-08-07:
> Also is this on dashing or bleed?

We're currently targeting dashing (with some tiny config changes to avoid errors in RViz https://github.com/ros-planning/navigation2/compare/dashing-devel...mikaelarguedas:mikael/dashing-sync).

> Any other errors/warnings like inability to transform laser scans? Particularly coming from AMCL?

I am seeing a bunch of dropped message and tf errors still:
```
[amcl-3] [INFO] [amcl_rclcpp_node]: [signalFailure] Drop message: frame 'base_scan' at time 67.401 for reason(0)
[world_model-4] [INFO] [global_costmap.global_costmap_rclcpp_node]: [signalFailure] Drop message: frame 'base_scan' at time 71.601 for reason(0)
[world_model-4] [INFO] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: . canTransform returned after 1.001 timeout was 1.
```
Is it a common set of error messages or are we doing something bad somewhere?

Currently looking at https://github.com/ros2/geometry2/pull/146 and https://github.com/ros2/rviz/pull/375 to see if these improve things.

> Is this running with the encryption going or just normal?

I believe these results are just normal, with encryption CPU loads gets significantly higher on my machine

> are you using any custom config files or just the defaults?

We're using these parameters: https://github.com/ROBOTIS-GIT/turtlebot3/blob/ros2/turtlebot3_navigation2/param/burger_params.yaml

The diff from the ones from navigation2@master:
```diff
$ diff burger_params.yaml nav2_params.yaml 
60c60
<     max_vel_x: 0.22
---
>     max_vel_x: 0.26
64c64
<     max_speed_xy: 0.22
---
>     max_speed_xy: 0.26
86,87c86,87
<     PathAlign.scale: 32.0
<     GoalAlign.scale: 24.0
---
>     PathAlign.scale: 0.0
>     GoalAlign.scale: 0.0
103c103
<       robot_radius: 0.105
---
>       robot_radius: 0.22
125c125
<       robot_radius: 0.105
---
>       robot_radius: 0.22
145c145
<     yaml_filename: "map.yaml"
---
>     yaml_filename: "turtlebot3_world.yaml"
```

ruffsl, 2019-08-07:
> @ruffsl are you using any custom config files or just the defaults?

You can see our repos file we are using in the `Version or commit hash` bullet above, as we have a few commits optop of 9cbda503593d91081ad255fcf2d06e09c9d9f760 for changing use_sim_time, unreliable transport, and tweaking the rviz config view. Currently: https://github.com/mikaelarguedas/navigation2/commits/mikael/dashing-sync

> Frankly 12% CPU seems about right for the costmap items since that's spinning regardless if anyone's using it.

I'd still argue even that's aggressive, could that be toned down as well by shrinking the local cost map area?

> Any other errors/warnings like inability to transform laser scans? Particularly coming from AMCL?

We see lot of info messages from `world_model`:

```
[world_model-4] [INFO] [global_costmap.global_costmap_rclcpp_node]: [signalFailure] Drop message: frame 'base_scan' at time 179.203 for reason(0)
[world_model-4] [INFO] [global_costmap.global_costmap_rclcpp_node]: [signalFailure] Drop message: frame 'base_scan' at time 179.402 for reason(0)
[world_model-4] [INFO] [global_costmap.global_costmap_rclcpp_node]: [signalFailure] Drop message: frame 'base_scan' at time 179.603 for reason(0)
```
Here is an capture from `navigation2.launch.py`, while successfully localizing and navigating:

<details>
```
[INFO] [launch]: All log files can be found below /home/ruffsl/.ros/log/2019-08-07-14-33-32-485199-dox-5447
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [lifecycle_manager-1]: process started with pid [5466]
[INFO] [map_server-2]: process started with pid [5467]
[INFO] [amcl-3]: process started with pid [5468]
[INFO] [world_model-4]: process started with pid [5469]
[INFO] [dwb_controller-5]: process started with pid [5470]
[INFO] [navfn_planner-6]: process started with pid [5477]
[INFO] [recoveries_node-7]: process started with pid [5484]
[INFO] [bt_navigator-8]: process started with pid [5499]
[INFO] [rviz2-9]: process started with pid [5511]
[lifecycle_manager-1] [INFO] [lifecycle_manager]: Creating
[map_server-2] [INFO] [map_server]: Creating
[world_model-4] [INFO] [world_model]: Creating
[lifecycle_manager-1] [INFO] [lifecycle_manager]: [34m[1mStarting the system bringup...[0m[0m
[lifecycle_manager-1] [INFO] [lifecycle_manager]: [34m[1mCreating and initializing lifecycle service clients[0m[0m
[amcl-3] [INFO] [amcl]: Creating
[lifecycle_manager-1] [INFO] [lifecycle_manager]: [34m[1mConfiguring and activating map_server[0m[0m
[dwb_controller-5] [INFO] [dwb_controller]: Creating
[dwb_controller-5] [INFO] [local_costmap.local_costmap]: Creating
[world_model-4] [INFO] [global_costmap.global_costmap]: Creating
[recoveries_node-7] [INFO] [recoveries]: Configuring Spin
[recoveries_node-7] [WARN] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[recoveries_node-7] [INFO] [recoveries]: Configuring BackUp
[navfn_planner-6] [INFO] [navfn_planner]: Creating
[bt_navigator-8] [INFO] [bt_navigator]: Creating
[map_server-2] [INFO] [map_server]: Configuring
[map_server-2] [INFO] [map_server]: OccGridLoader: Creating
[map_server-2] [INFO] [map_server]: OccGridLoader: Configuring
[map_server-2] [WARN] [map_server]: Mode parameter not set, using default value (trinary): bad conversion
[map_server-2] [INFO] [map_server]: Activating
[map_server-2] [INFO] [map_server]: OccGridLoader: Activating
[lifecycle_manager-1] [INFO] [lifecycle_manager]: [34m[1mConfiguring and activating amcl[0m[0m
[amcl-3] [INFO] [amcl]: Configuring
[amcl-3] [INFO] [amcl]: initTransforms
[amcl-3] [WARN] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[amcl-3] [INFO] [amcl]: initPubSub
[amcl-3] [INFO] [amcl]: Subscribed to map topic.
[amcl-3] [INFO] [amcl]: Activating
[lifecycle_manager-1] [INFO] [lifecycle_manager]: [34m[1mConfiguring and activating world_model[0m[0m
[world_model-4] [INFO] [world_model]: Configuring
[world_model-4] [INFO] [global_costmap.global_costmap]: Configuring
[world_model-4] [INFO] [global_costmap.global_costmap]: Using plugin "static_layer"
[world_model-4] [INFO] [global_costmap.global_costmap]: Using plugin "obstacle_layer"
[world_model-4] [INFO] [global_costmap.global_costmap]: Subscribed to Topics: scan
[world_model-4] [INFO] [global_costmap.global_costmap]: Using plugin "inflation_layer"
[world_model-4] [INFO] [global_costmap.global_costmap]: StaticLayer: Resizing costmap to 384 X 384 at 0.050000 m/pix
[world_model-4] [INFO] [world_model]: Activating
[world_model-4] [INFO] [global_costmap.global_costmap]: Activating
[... 118 lines cut ...]
[world_model-4] [INFO] [global_costmap.global_costmap_rclcpp_node]: [signalFailure] Drop message: frame 'base_scan' at time 51.202 for reason(0)
[world_model-4] [INFO] [global_costmap.global_costmap_rclcpp_node]: [signalFailure] Drop message: frame 'base_scan' at time 51.401 for reason(0)
[... cut ...]
[world_model-4] [INFO] [global_costmap.global_costmap_rclcpp_node]: [signalFailure] Drop message: frame 'base_scan' at time 54.201 for reason(0)
[world_model-4] [INFO] [global_costmap.global_costmap_rclcpp_node]: [signalFailure] Drop message: frame 'base_scan' at time 54.402 for reason(0)
[world_model-4] [INFO] [global_costmap.global_costmap_rclcpp_node]: [signalFailure] Drop message: frame 'base_scan' at time 54.602 for reason(0)
[dwb_controller-5] [INFO] [dwb_controller]: Reached the goal!
[bt_navigator-8] [INFO] [bt_navigator]: Navigation succeeded
```

</details>

SteveMacenski, 2019-08-07:
To be frank, I've been mostly focused on capability and not yet performance, but it looks like I may refocus some of that effort to performance. If you're running in simulation you can drop the controller rate to 5hz and the global costmap update rate to 1hz. Set those `*Align` configs to 0. That should help a bit.

The CPU 12% for the costmaps seems about right* given all the problems (TF dropping messages, processing a costmap by itself was never a super light weight thing, is there a depth camera involved there or just laser?), but I'm not saying that there isn't something wrong, just from hand wavey background knowledge of getting really deep into costmap 2D in ROS1 navigation 12% is like the lowest I've ever had **especially** when publishing that costmap at rate, which right now the Nav2 version doesnt have subscriber detection, but I havent run with only a laser.

orduno, 2019-08-08:
@mjeronimo do these numbers match what you've found so far from your performance analysis?

orduno, 2019-08-08:
In addition to what has been proposed, consider reducing the global costmap resolution and the local costmap window size. This will certainly have some impact on navigation quality, but it's worth experimenting. We want to try that at some point.

Also, for simulation you can use `nav2_simulation_launch` with the argument `simulator:=gzserver`

```
ros2 launch nav2_bringup nav2_simulation_launch.py simulator:=gzserver
```

SteveMacenski, 2019-08-08:
I think the curent global costmap resolution is 5cm, that's a pretty good number, maintaining it shouldn't be that hard if the ranges are setup for the laser to be small. The RPlidar is pretty short range and low resolution, I don't know that lowering the global costmap resolution will have much impact given the small amount of data being processed.

I'm more suspicious that potentially the global costmap is being updated at 5hz in the default configs (will look today), that would add some extra "umph" when our replanning rate I think its 1hz default.

I'm generally suspect of AMCL if those numbers can be reproduced. These items are on my queue to think about tomorrow or next week depending on time availability.

SteveMacenski, 2019-08-08:
I'm going to assign myself to start on this, but help from other folks is appreciated.

crdelsey, 2019-08-08:
My first guess as to the problem would be ros2/geometry2#119

All the nodes mentioned have TF listeners, so the above would be a problem for them.

mjeronimo, 2019-08-08:
@orduno In my recent performance analysis, on the NUC I've been using I see about 3% CPU to spin a node. After running a navigation, the nodes should fall back to this level. One can compare a sample ROS2 application, such as a simple pub/sub example, for reference. 

Since the three modules mentioned are the ones that have the TF listener, as Carl mentions, I would suspect that the additional CPU utilization is coming from there. There have been recent changes TF listener and the latest ROS2 code should have addressed this.

SteveMacenski, 2019-08-08:
> 3% a node

jeeze. AMCL used to take ~3% running the particle filter

SteveMacenski, 2019-08-09:
Another thing I just recalled is that we dont use the costmap updates topics yet since RVIZ doesn't support them as of today. Sending the full costmap every time isn't a cheap operation.

SteveMacenski, 2019-08-09:
I'm seeing crazy costmap update rates while debugging another issue 

```
[dwb_controller-5] [INFO] [local_costmap.local_costmap]: Map update time: 0.000919024
[dwb_controller-5] [INFO] [local_costmap.local_costmap]: Cycle: 0.000919024

```

ruffsl, 2019-09-06:
I've updated our demo for https://github.com/ros-swg/turtlebot3_demo , so you can check if you're able to reproduce the same performance usage changes. As a side note, perhaps maintainers here could help us triage a QoS/tf issue I think we might be having with navigation + sros2 . https://github.com/ros-swg/turtlebot3_demo/issues/12

crdelsey, 2019-10-07:
@ruffsl Is this still an issue?

ruffsl, 2019-10-07:
I'll check back again after the improvements to fast rtps QoS are sync'ed into dashing:
https://discourse.ros.org/t/preparing-for-dashing-sync-and-patch-release-2019-10-10/10898

ruffsl, 2019-10-18:
I tested this again using the latest recent dashing sync, and everything seems to be much better. The lion's share of the CPU usage remaining seems to be just from the gazebo and rviz GUIs:

[image]

@crdelsey , feel free to close this, however we are still having issues enabling security for navigation2:
See latest comment here: https://github.com/ros-swg/turtlebot3_demo/issues/12#issuecomment-543981496

SteveMacenski, 2019-10-18:
@ruffsl dear god, what computer are you running on? Gazebo takes **3%** CPU?

closed 2019-10-18 by SteveMacenski

ruffsl, 2019-10-18:
> what computer are you running on? Gazebo takes 3% CPU?

It's about a 3yo CPU in a workstation. I was thinking perhaps the GPU ray tracing is working in my favor, but it doesn't seem like the turtlebot3 is using `gpu_ray` , so perhaps its my CPU core count.

https://github.com/ROBOTIS-GIT/turtlebot3_simulations/blob/224f347204a3af3315313a620dcab4808d21deed/turtlebot3_gazebo/models/turtlebot3_burger/model.sdf#L94
http://gazebosim.org/tutorials?tut=ros_gzplugins

```
Linux ubuntu 4.20.4-042004-generic #201901222207 SMP Tue Jan 22 22:09:25 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
```

<details>

[image]

SteveMacenski, 2019-10-18:
I need to get off this peewee laptop game then. My computer nearly catches fire when I run gazebo with a depth sensor for longer than 20 minutes