ros-navigation/navigation2 #1021: Move base simple goal interface isn't processing goals

SteveMacenski, 2019-08-08:
Also... `move_base` isnt a "thing" anymore.

Naming suggestions? We'd have to change in RVIZ2's code as well but that's a trivial PR

crdelsey, 2019-08-08:
`nav_goal`?

SteveMacenski, 2019-08-08:
`nav2_goal` `navigation2/goal` `bt_navigator/goal` 

Any of the above, I really don't have an opinion. Anyone have something they would strongly like to assert?

mjeronimo, 2019-08-09:
In general, I think the topic names should be kept abstract. I'd opt for goal_pose. Names like this can always be remapped to be more specific, if desired. I don't think namespaces, like navigation2 need to be in the name itself, but can be handled elsewhere. Perhaps @orduno has some input since he's looking at multi-robot scenarios and separating all of the topic and other names.

SteveMacenski, 2019-08-09:
Well multirobot in rviz ~isn't~ I don't think is really supported (.... I think? I've never thought to try in rviz). I don't think adding `nav` is too specific considering the button is called `2D Nav Goal`, I think if we changed to `goal_pose` then the RVIZ PR would probably change the button name to `Goal Pose` as well to make abstract.

Any issues with that plan?

mjeronimo, 2019-08-09:
@SteveMacenski  No issue with the rviz side of things. I was referring to the nav2 stack itself and the names we should use for the various topics. In deciding this several factors should be considered including multi-robot, the launch system, etc. @orduno is currently looking at this and I suggested that he make some recommendations to the team that we can review.

closed 2019-08-19 by SteveMacenski