ros-navigation/navigation2 #1026: Audit topics to ensure connections are correct

SteveMacenski, 2019-08-09:
I just found a relatively concerning problem where the collision checker in the spin recovery was looking at `global_costmap/costmap_raw` but the global costmap is actually publishing at `costmap_raw` so it was never getting any information!

We should look and make sure there's no other namespace / typo issues

SteveMacenski, 2019-08-09:
I'm seeing 2 publishers on the namespace-less `costmap` topic, which i assume are the 2 costmaps, that's probably not good, there's got to be some folks not getting the information

SteveMacenski, 2019-08-09:
There are agents subed to _raw, costmap, and updates, we need to rectify them a bit

crdelsey, 2019-08-10:
Did you see this behavior before or after #999 went in? I think there is a bug in #999. I think it needs the same fix that was added in #1020

SteveMacenski, 2019-08-11:
Agreed, I notice it fixed after I merged in #1020 but I was waiting on a confirmed reason why before closing out, but since you bring it up as well that's sufficient :+1:

closed 2019-08-11 by SteveMacenski