ros-navigation/navigation2 #1028: Use costmap 2D topic and updates, not raw when possible

SteveMacenski, 2019-08-09:
Right now there are a few places using the `costmap_raw` topic. That's really heavy to both send and receive, we should be using the updates topic in conjection with the main costmap topic for better performance.

Once that is done, and if done well in the CostmapSubscriber utility, then we can remove the _raw topic since we, right now, don't get `getNumSubscribers()` to not publish it if no one's listening in ROS2.

bpwilcox, 2019-08-13:
So just to clarify this issue: the costmap_raw topic publishes a `nav2_msgs/Costmap` msg which is uint8 data directly from the costmap values. At the time, there is no updates msg, so are you recommending that we create  a `nav2_msgs/CostmapUpdate` msg to align with the `map_msgs/msg/OccupancyGridUpdate` or something different?

SteveMacenski, 2019-08-13:
More or less, yes. The main reason is to not publish the full NxN costmap at 5hz, serializing and passing that massive thing around for any even modest sized space (a floor of an office building) has non-trivial impacts on performance

bpwilcox, 2019-08-13:
@SteveMacenski Right, I agree. It's unclear from your ticket however if you were recommending to remove the `costmap_raw` topic entirely, which currently publishes the "raw" or unconverted costmap values, since the `costmap` topic using  `nav_msgs/OccupancyGrid` converts those values into int8 data -1 to 100.

SteveMacenski, 2019-08-14:
I was recommending to remove it if we didn't find a strong reason to include it. Usually when I used the `always_send_full_costmap` option in the past it was more or less for "lazy" reasons of not wanting to build my own object to take the costmaps and updates and rectify them together, just taking the raw costmap topic let me avoid it for simple R&D.

If we include a subscriber that does that work for you (`CostmapSubscriber`) then hey, that need for quick R&D is gone because it comes batteries included. 

Always sending the raw costmap over the wire at full rate even if no one is listening is super wasteful and potentially one of the largest things we send around in the navigation stack and it doesn't have great utility. I'd say keep the option to do that if we could `getNumlisteners()` again, but it doesnt seem like that's going to be an option anytime soon. 

For instance, in costmap_2d in ROS1, I have benchmarked the global costmap CPU load jumping to 10% of a core because of publishing the massive costmaps at full rate.

mkhansen-intel, 2019-08-14:
@SteveMacenski - there is a function to find out how many subscribers there are to a topic: `rclcpp::Node::count_subscribers()`

https://github.com/ros2/rclcpp/blob/25f196989c02c092d6073c8383f0fbdfca6a4dcd/rclcpp/include/rclcpp/rclcpp.hpp#L96

I think that's what you want here.

SteveMacenski, 2019-08-14:
Oh sweet, its around now

shivaang12, 2019-10-25:
@SteveMacenski Are we still considering this change/update or not?

SteveMacenski, 2019-10-25:
I'd like to get back to what the original costmap publisher was doing: https://github.com/ros-planning/navigation/blob/melodic-devel/costmap_2d/src/costmap_2d_publisher.cpp

We check for a new subscription to publish a full costmap and otherwise just publish updates. As Matt points out above we can tell how many subscribers are around now so we can do this.

I'm not 100% sure what the raw  topics were added for, maybe a debug artifact in the earlier days? Or maybe something someone wants. Either way, similar addition of only publishing when someone is listening. The costmaps are one of the heaviest things (besides pointclouds) to handle. Minimizing the traffic from them being unnecessarily published and reducing the size of them (with updates) is a big improvement.

It also looks like rviz2 was updated to use the `_updates` https://github.com/ros2/rviz/blob/ros2/rviz_default_plugins/src/rviz_default_plugins/displays/map/map_display.cpp#L187 as well so we should be good to go

shivaang12, 2019-10-25:
@SteveMacenski Sounds cool! I am game to do it if it okay?

SteveMacenski, 2019-10-25:
Sure!

bpwilcox, 2019-10-25:
> I'm not 100% sure what the raw topics were added for, maybe a debug artifact in the earlier days?
>
I thought I had clarified before, but the raw topics were added to publish the `uint8` raw costmap data before conversion (and data loss) to the occupancy grid map topic that is used primarily for visualization. The raw costmap topic is being used by the recovery server to locally construct a costmap into its `CostmapSubscriber`. 

All that said, I think having the publishers only publish when something subscribes to it is fair.

SteveMacenski, 2019-11-06:
I traced back through, right now the only one that uses this is the collision checker -- why can't that use the costmap itself? 

> The raw costmap topic is being used by the recovery server to locally construct a costmap into its `CostmapSubscriber`

Sure, I understand why it might exist to support the costmap subscriber, but why does the costmap subscriber require it? The only client of that is the collision checker, and that is more than fine with the reduced data. 

Moreover, looking at the collision checker, its checking if in collision by 

```
  if (cost == LETHAL_OBSTACLE) {
```

and those `LETHAL_OBSTACLE` values are defined by the `uint8` `254` not the raw values, so right now using the raw data seems to me to be invalid.

bpwilcox, 2019-11-06:
> @bpwilcox I traced back through, right now the only one that uses this is the collision checker -- why can't that use the costmap itself?
> 
The collision checker is being used in the recovery server where it does not itself contain a costmap2DROS object, thus it needs to create a costmap2D in order to inspect the cost values and access other related API, thus it creates the CostmapSubscriber.

> Sure, I understand why it might exist to support the costmap subscriber, but why does the costmap subscriber require it? The only client of that is the collision checker, and that is more than fine with the reduced data.
> 
I can agree with this point, since we are only checking for the lethal objects here, we could use the reduced data. This could mean reconstructing the costmap2D (since we are using its API) after converting the occupancy grid data from the topic, or rewriting the collision checker to only use the topic (probably not my preferred option).

> Moreover, looking at the collision checker, its checking if in collision by
> 
> ```
>   if (cost == LETHAL_OBSTACLE) {
> ```
> 
> and those `LETHAL_OBSTACLE` values are defined by the `uint8` `254` not the raw values, so right now using the raw data seems to me to be invalid. Unless I'm misreading that, this is a safety issue.

The costmap raw data is an unsigned char, so I believe this is correct. The test should verify that. 

Moreover, I think there is value in having the ability to re-construct the costmap2D on a remote node via the subscriber without data loss. Given that we can now limit publishing the topic if there are no subscribers, it is up to the user's system design to do so.

SteveMacenski, 2019-11-07:
To your first comment, that still doesn’t explain why you need the raw data, just that you need a costmap representation over a topic, which exists. 

Second, I’d prefer to change it to the actual costmap topic. The whole point of the costmap is to do collision checking. If the costmap itself is insufficient and you need these raw values, then no where in the stack should not be using the raw data. Having these 2 pipes open isn’t necessary. It can be up to the user to use them, but only if they understand the difference and the value proposition of both. There’s now extra topics up to mix up & since the collision checker is using it, that means there’s now even more network traffic. 

This doesn’t seem to me to be a good trade off in complexity/traffic compared to the idea that it might be useful.

bpwilcox, 2019-11-12:
I am okay with changing the collision checker to use the `/costmap` occupancy grid topic since it is primarily only doing binary collision checks. I think the best option would be to have a flag for the `CostmapSubscriber` to grab either the `OccupancyGrid.msg` topic (by default) or the raw `Costmap.msg` topic, where internally it will convert the topics to a `Costmap2D` object so that applications can access the `Costmap2D` api.

SteveMacenski, 2019-11-12:
Sounds reasonable. Thought from implementation details that sounds hard to make clean but you have a better understanding of that object than I. 

I might rename `Costmap` to `RawCostmap` to make it clear

bpwilcox, 2019-11-12:
> Sounds reasonable. Thought from implementation details that sounds hard to make clean but you have a better understanding of that object than I.
> 
I think it's cleaner to do conversions at the `CostmapSubscriber` level, its job is to abstract the msg and create a costmap2D object for use in remote nodes. Then none of the `CollisionChecker` code needs to change.

> I might rename `Costmap` to `RawCostmap` to make it clear
>
You mean the `nav2_msg/Costmap.msg`? I'm not opposed to the rename.

SteveMacenski, 2019-11-12:
> You mean the nav2_msg/Costmap.msg? I'm not opposed to the rename.

Yeah.

>  I might consider adding an overload on the constructor of the CollisionChecker to accept a pointer to the costmap2D directly

Makes sense, though right now I think the only user is the recovery server and if you have direct access to a costmap, this wrapper isn't really necessary

bpwilcox, 2019-11-12:
> Makes sense, though right now I think the only user is the recovery server and if you have direct access to a costmap, this wrapper isn't really necessary
>
It removes duplicated code for collision checking, so DWB could could use the `CollisionChecker` too

SteveMacenski, 2019-11-12:
Then would you be removing the footprint and the costmap subscriber? That pretty much changes the structure of the code quite a bit more than just a constructor swap. 

I agree in concept there's code that could be removed. I think the `marginal_utility / marginal_effort` to too low to expend the effort personally, but power to you :small_airplane:

SteveMacenski, 2020-05-18:
Per #1352, I think this is not worth it anymore. There were open questions without good answers to fix this issue. Due to ROS2 not allowing multiple nodes to share parameters, it wouldn't be tractable to re-add multiple copies of identical params for many processes to make it functional. Using the raw transport will be done until the next environment redesign.

closed 2020-05-18 by SteveMacenski