ros-navigation/navigation2 #1030: test_collision_checker is failing

crdelsey, 2019-08-12:
## Bug report

**Required Info:**

- Operating System:
  - 
- Installation type:
  - 
- Version or commit hash:
  - Latest master as of 12-Aug-2019 (97db4b2d133)
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue
Re-enable `test_collision_checker` in nav2_costmap_2d`. It fails with the following output
```
    Start 8: test_collision_checker

8: Test command: /usr/bin/python3 "-u" "/home/crdelsey/src/ros_dev_workspace/ros2_ws/install/share/ament_cmake_test/cmake/run_test.py" "/home/crdelsey/src/ros_dev_workspace/navigation2_ws/build/nav2_costmap_2d/test_results/nav2_costmap_2d/test_collision_checker.xml" "--package-name" "nav2_costmap_2d" "--generate-result-on-success" "--env" "TEST_MAP=/home/crdelsey/src/ros_dev_workspace/navigation2_ws/navigation2/nav2_costmap_2d/test/map/TenByTen.yaml" "TEST_LAUNCH_DIR=/home/crdelsey/src/ros_dev_workspace/navigation2_ws/navigation2/nav2_costmap_2d/test/test_launch_files" "TEST_EXECUTABLE=/home/crdelsey/src/ros_dev_workspace/navigation2_ws/build/nav2_costmap_2d/test/integration/test_collision_checker_exec" "--command" "/home/crdelsey/src/ros_dev_workspace/navigation2_ws/navigation2/nav2_costmap_2d/test/integration/costmap_tests_launch.py"
8: Test timeout computed to be: 60
8: -- run_test.py: extra environment variables:
8:  - TEST_LAUNCH_DIR=/home/crdelsey/src/ros_dev_workspace/navigation2_ws/navigation2/nav2_costmap_2d/test/test_launch_files
8:  - TEST_EXECUTABLE=/home/crdelsey/src/ros_dev_workspace/navigation2_ws/build/nav2_costmap_2d/test/integration/test_collision_checker_exec
8:  - TEST_MAP=/home/crdelsey/src/ros_dev_workspace/navigation2_ws/navigation2/nav2_costmap_2d/test/map/TenByTen.yaml
8: -- run_test.py: invoking following command in '/home/crdelsey/src/ros_dev_workspace/navigation2_ws/build/nav2_costmap_2d/test/integration':
8:  - /home/crdelsey/src/ros_dev_workspace/navigation2_ws/navigation2/nav2_costmap_2d/test/integration/costmap_tests_launch.py
8: [INFO] [launch]: All log files can be found below /home/crdelsey/.ros/log/2019-08-12-10-42-39-744419-crdelsey-DESK3-21883
8: [INFO] [launch]: Default logging verbosity is set to INFO
8: [INFO] [map_server-1]: process started with pid [21886]
8: [INFO] [lifecycle_manager-2]: process started with pid [21887]
8: [INFO] [costmap_tests-3]: process started with pid [21888]
8: [costmap_tests-3] Running main() from /home/crdelsey/src/ros_dev_workspace/ros2_ws/install/src/gtest_vendor/src/gtest_main.cc
8: [costmap_tests-3] [==========] Running 3 tests from 1 test case.
8: [costmap_tests-3] [----------] Global test environment set-up.
8: [costmap_tests-3] [----------] 3 tests from TestNode
8: [costmap_tests-3] [ RUN      ] TestNode.uknownSpace
8: [map_server-1] [WARN] [map_server]: Mode parameter not set, using default value (trinary): bad conversion
8: [costmap_tests-3] [ERROR] [test_collision_checker]: Pose Goes Off Grid.
8: [costmap_tests-3] [ERROR] [test_collision_checker]: Footprint Goes Off Grid.
8: [costmap_tests-3] [ERROR] [test_collision_checker]: Footprint Goes Off Grid.
8: [costmap_tests-3] [INFO] [test_collision_checker]: Configuring
8: [costmap_tests-3] [INFO] [test_collision_checker]: StaticLayer: Resizing costmap to 10 X 10 at 1.000000 m/pix
8: [costmap_tests-3] [INFO] [test_collision_checker]: Activating
8: [costmap_tests-3] [INFO] [test_collision_checker]: Deactivating
8: [costmap_tests-3] [INFO] [test_collision_checker]: Cleaning Up
8: [costmap_tests-3] [       OK ] TestNode.uknownSpace (200 ms)
8: [costmap_tests-3] [ RUN      ] TestNode.FreeSpace
8: [costmap_tests-3] [ERROR] [test_collision_checker]: Footprint Hits Obstacle.
8: [costmap_tests-3] [INFO] [test_collision_checker]: Configuring
8: [costmap_tests-3] [INFO] [test_collision_checker]: StaticLayer: Resizing costmap to 10 X 10 at 1.000000 m/pix
8: [costmap_tests-3] [INFO] [test_collision_checker]: Activating
8: [costmap_tests-3] /home/crdelsey/src/ros_dev_workspace/navigation2_ws/navigation2/nav2_costmap_2d/test/integration/test_collision_checker.cpp:323: Failure
8: [costmap_tests-3] Expected equality of these values:
8: [costmap_tests-3]   collision_checker_->testPose(2.5, 7, 0)
8: [costmap_tests-3]     Which is: false
8: [costmap_tests-3]   true
8: [costmap_tests-3] [INFO] [test_collision_checker]: Deactivating
8: [costmap_tests-3] [INFO] [test_collision_checker]: Cleaning Up
8: [costmap_tests-3] [  FAILED  ] TestNode.FreeSpace (175 ms)
8: [costmap_tests-3] [ RUN      ] TestNode.CollisionSpace
8: [costmap_tests-3] [ERROR] [test_collision_checker]: Footprint Hits Obstacle.
8: [costmap_tests-3] [ERROR] [test_collision_checker]: Footprint Goes Off Grid.
8: [costmap_tests-3] [INFO] [test_collision_checker]: Configuring
8: [costmap_tests-3] [INFO] [test_collision_checker]: StaticLayer: Resizing costmap to 10 X 10 at 1.000000 m/pix
8: [costmap_tests-3] [INFO] [test_collision_checker]: Activating
8: [costmap_tests-3] [INFO] [test_collision_checker]: Deactivating
8: [costmap_tests-3] [INFO] [test_collision_checker]: Cleaning Up
8: [costmap_tests-3] [       OK ] TestNode.CollisionSpace (171 ms)
8: [costmap_tests-3] [----------] 3 tests from TestNode (546 ms total)
8: [costmap_tests-3] 
8: [costmap_tests-3] [----------] Global test environment tear-down
8: [costmap_tests-3] [==========] 3 tests from 1 test case ran. (546 ms total)
8: [costmap_tests-3] [  PASSED  ] 2 tests.
8: [costmap_tests-3] [  FAILED  ] 1 test, listed below:
8: [costmap_tests-3] [  FAILED  ] TestNode.FreeSpace
8: [costmap_tests-3] 
8: [costmap_tests-3]  1 FAILED TEST
8: [ERROR] [costmap_tests-3]: process has died [pid 21888, exit code 1, cmd '/home/crdelsey/src/ros_dev_workspace/navigation2_ws/build/nav2_costmap_2d/test/integration/test_collision_checker_exec'].
8: [INFO] [lifecycle_manager-2]: sending signal 'SIGINT' to process[lifecycle_manager-2]
8: [INFO] [map_server-1]: sending signal 'SIGINT' to process[map_server-1]
8: [map_server-1] [INFO] [map_server]: Creating
8: [map_server-1] [INFO] [map_server]: Configuring
8: [map_server-1] [INFO] [map_server]: OccGridLoader: Creating
8: [map_server-1] [INFO] [map_server]: OccGridLoader: Configuring
8: [map_server-1] [INFO] [map_server]: Activating
8: [map_server-1] [INFO] [map_server]: OccGridLoader: Activating
8: [map_server-1] [INFO] [rclcpp]: signal_handler(signal_value=2)
8: [INFO] [map_server-1]: process has finished cleanly [pid 21886]
8: [lifecycle_manager-2] [INFO] [lifecycle_manager]: Creating
8: [lifecycle_manager-2] [INFO] [lifecycle_manager]: Starting the system bringup...
8: [lifecycle_manager-2] [INFO] [lifecycle_manager]: Creating and initializing lifecycle service clients
8: [lifecycle_manager-2] [INFO] [lifecycle_manager]: Configuring and activating map_server
8: [lifecycle_manager-2] [INFO] [lifecycle_manager]: The system is active
8: [lifecycle_manager-2] [INFO] [rclcpp]: signal_handler(signal_value=2)
8: [lifecycle_manager-2] [INFO] [lifecycle_manager]: Destroying
8: [INFO] [lifecycle_manager-2]: process has finished cleanly [pid 21887]
8: -- run_test.py: return code 1
8: -- run_test.py: generate result file '/home/crdelsey/src/ros_dev_workspace/navigation2_ws/build/nav2_costmap_2d/test_results/nav2_costmap_2d/test_collision_checker.xml' with failed test
8: -- run_test.py: verify result file '/home/crdelsey/src/ros_dev_workspace/navigation2_ws/build/nav2_costmap_2d/test_results/nav2_costmap_2d/test_collision_checker.xml'
1/1 Test #8: test_collision_checker ...........***Failed    1.05 sec
```

SteveMacenski, 2019-08-12:
I think its fixed, most recent PRs are passing https://github.com/ros-planning/navigation2/pull/1031

crdelsey, 2019-08-29:
It's passing CI because I disabled this test. @bpwilcox Can you take a look at this and see if the failure is legit or a false positive.

yathartha3, 2019-09-21:
Its been passing on mine, and haven't seen it fail.

bpwilcox, 2019-10-01:
Resolved

closed 2019-10-01 by bpwilcox