ros-navigation/navigation2 #1032: [behavior tree] remove restriction for 2 nodes in recovery node

SteveMacenski, 2019-08-12:
Right now there's a restriction in `RecoveryNode` for only 2 children: main navigation and recovery sequences. 

This is a reasonable first step, but we may want to have a few sequences that we iterate through on certain types of failures. For instance if we have an experimental controller which has some known-failure modes, when we fail we want to go to the next sequence which may contain the action to another controller server which is stable in the failure modes of the prior one. 

Analogous statements for the planner could be made. If we have some planner that fails to produce a path, maybe I want to try another set of planners before enacting behavioral recoveries users may be able to identify

mhpanah, 2019-08-15:
@SteveMacenski The restrictions on the RecoveryNode is intentional and it must contain 2 children #832.  I was planning to give a detailed talk on realizing multi-scope recovery behaviors with the recovery node at the ROSCon but that didn't go through.  If time allows maybe @mjeronimo can go over the recovery behaviors when he is presenting BT at ROSCon. I can go over #953 at our next WG.

The scenario that you are proposing could simply be implemented with a fallback node. For example: if plannerA is running and the intention is to run plannerB iff plannerA fails before invoking recovery actions, the way to construct this tree would be:

```xml
<RecoveryNode>
  <Fallback>
    <plannerA>
    <plannerB>
  </Fallback>
  <...Invoke Recovery Actions when planner A and B fails...>
</RecoveryNode>
```

SteveMacenski, 2019-08-15:
After looking over the code and (at least my understanding of this, correct me as needed) options available, sure, things could be done a certain way. But I would argue devils advocate: "but why add artificial barriers?" if someone wants to do it that way, and it seems totally valid, why add the hard limit that isn't limited by technology or some kind of buggy behavior?

Also the Fallback node as far as I can see is not documented and I don't see any source for it in the project. There's a "it exists" comment in the readme but no explanation of how it operates, but that's besides the point for this ticket.

mhpanah, 2019-08-15:
These are the rules on the control flow type nodes based on literature. BT consists of control flow nodes, such as fallback, sequence, parallel, decorator, etc. and execution nodes such as condition and action nodes which are the leaf nodes of the tree. As another example, the decorator node also has a limit on the number of its children which is one.

Let's talk in more detail in WG.  Regarding the documentation, I agree we should update our documentation with a brief description of all the nodes that we are using.

crdelsey, 2019-08-15:
> if someone wants to do it that way, and it seems totally valid, why add the hard limit that isn't limited by technology or some kind of buggy behavior? 

I think the idea is that the "Recovery" node is a relatively standard concept in behavior trees; it just happens to be lacking from our particular behavior tree library. Changing its behavior would result deviate from the standard.

The idea of having a sequence of recovery actions can be achieved by combining existing simpler bt nodes with the recovery node. This allows you to keep the recovery node itself simpler, and allows the user to decide for themselves what kind of recovery behavior they want.

That said, if a sequence of recovery actions is a very common thing, we could:
1. Create another node type that combines the behavior of a recovery node and a fallback or whatever into one, or
2. Add sequences into to the recovery node and change the "standard" of what a recovery node is.

I kind of like the idea of keeping the recovery node simple. IMHO, the most likely recovery behavior we'll want is some sort of selector or switch statement thing that does different recoveries depending on the failure return code. In that case, adding in a sequence to the Recovery node concept would be wasted.

SteveMacenski, 2019-08-15:
> I think the idea is that the "Recovery" node is a relatively standard concept in behavior trees

OK enough said, I didn't know that concept was not of our own creation.

closed 2019-08-15 by SteveMacenski

crdelsey, 2019-08-15:
@mhpanah Can you add a link to some of reference material you found on Recovery nodes?