ros-navigation/navigation2 #1033: [bt navigator] Either use rate controller to throttle both path + controller or neither

SteveMacenski, 2019-08-12:
Right now we're using a `RateController` decorator to throttle the path planner but the controller's speed is set in the while loop of the main action in the controller server. We should make these defined in the same place whether in the BT or in the configuration file. I'd say in the configuration file since it helps make a given `xml` magic-number-less to find more general utility. 

I'd vote to remove the concept of `RateController` and have the speed of replanning be internal to the server sp. server. 

Correct me if I'm wrong here, I can't find any reference to the `FallBack` xml tag so if there's something more going on here let me know

crdelsey, 2019-08-15:
I'd say that represents a fundamental shift in the design concept of the stack. The basic concept was that:
1. bt_navigator was told to go to a location
2. bt_navigator told the global planner to compute a path
3. bt_navigator told the controller to follow that path
4. When the controller finished following the path, bt_navigator completed the action

The rate controller around 2, was thought of as an internal detail of bt_navigator to improve results, without changing the other components.

If we pulled the Rate loop out of DWB and into bt_navigator, we'd end up with something more like move_base from ROS 1, where the local controller is no longer a controller, but just a planner. It would just be a service to compute the cmd_vel.

If, instead, we pushed the rate controller down into the global planner, ... well that is a new concept. It sort of makes the parallel re-planning a fundamental concept of the stack rather than an optional extra.

If we really want to make it more consistent, conceptually, I'd prefer to make it more ROS1 like and add the RateController to the local planner side of the tree as well. There are potential performance implications there, though; the timing on requesting a new local plan and getting the result is a bit more critical.

SteveMacenski, 2019-08-15:
I wouldn't assert one way over another, I'm just looking for consistency. Right now both the planner and controllers are `action` but are treated very differently. Moving forward we need to have runtime configurable controllers and planners more than we do right now. There's discussion on the general controller action that loads a `nav2_core2 or similar` controller plugin and operates. I'm all for that, but then we should have a similar concept with the planners so there's a natural analog.

SteveMacenski, 2019-08-15:
I imagine we'll want a similar thing with a planner API in a `nav2_core1 or similar` that is loaded into a larger action which takes care of calling and giving it the objects it needs (tf buffer, costmaps, maybe some publishers or utilities), but shouldn't then that action control the rate of execution, and when the BT goes around to the planner node it takes the plan and hands off to the controller? 

I like the concept of keeping them separated. Having them directly talk is a little too limiting to me, ideally I'd like to keep them talking over the behavior tree, because then I can conceivably put something between them to do some QA on the path before just shooting over to the controller.

crdelsey, 2019-08-15:
> I wouldn't assert one way over another, I'm just looking for consistency

Likewise, but I see the current locations of the "Rate" controllers consistent with the existing architectural concept. I think, if we push the rate controller down to a "global planner" component, we've changed the architectural concept somewhat.

Ros 1 had the concept that `move base` was the controller and `DWA` and `NavFn` were just algorithm plugins.

If we push the rate controller down into the "global planner", what does that make the relationship between bt_navigator and the global planner. It's no longer clear to me where one's responsibility ends and the other picks up.

Maybe if we elaborated on how this would work, the concept would fall out it. 

What do you envision as the interface to the global planner? 

Is it still an action interface?  If so, how would you define that the action is done? Does the global planner need a goal checker as well? 

Presumably, it is constantly providing paths to the goal? Do they come back using the action feedback mechanism to BT navigator?

crdelsey, 2019-08-16:
Having had a night to sleep on it, I can maybe articulate this better.

Consistency is a good thing when the underlying concepts are consistent. To me, the fact that DWB and NavFn both have a rate controlled loop is an accidental similarity rather than a fundamental similarity in the concepts they represent.

DWB without the rate loop is a simple cmd_vel optimizer. With the rate loop it becomes a path following controller.

Navfn without the rate loop is a simple path planner. I can't think of a concept it represents when it has the rate loop. The fact that we don't have a good end condition for the action (it just needs to be cancelled when the goal is reached) suggests to me that this architecture isn't quite right.

So, while we can force them to be consistent at the moment, I feel that could lead to architectural problems in the future.

If we can come up with a reasonable concept of what a combined rate controller/global planner represents, then I'm in.

SteveMacenski, 2019-08-17:
I'm not quite following your logic. Architecturally, there's no real difference I can discern between having the "call planner again" command or "check for new update from planner" in the orchestor. But then including the rate in the planner makes the concept of a "planning action" meaningful, since right now its really not an action, its a service to compute something once and hold still. 

It seems TMI for the BT navigator to even worry about calling the planner at some rate, it should be worried about higher level things. To me, these concepts of planning and control are _totally consistent_, in fact, DWB and the "controllers" we talk about in ROS navigation are really trajectory generators not even controllers (queue up larger argument about "path planning" vs "trajectory generation" but that's not worth getting into right now).

How things are organized right now isn't really rational, having the rate of recomputing in the local planner but the rate of recomputing of the global planner somewhere else. Moreover having some magic number in the XML for the BT of the replanning rate is less portable than a parameter to be read in by a robot system or a remap. 

Another argument I would make is that sometimes I might want multiple planners running, so if one fails I have another to fall back on (or in cases I have right now, in certain identifiable situations I want to use one, then another to start then follow an ongoing path). If I have control in my "master planner" that wraps the other planners together into a scoring system of the execution rate, i can make sure that the planners that are heavier (optimization based) can be run at a reduced speed  than my search based planners without having to hack the BT.  

At the end of the day we're just moving a while loop from the BT to the planning action, that makes no impact on architecture as I see it. I'd argue it reinforces the use of independently spinning action servers as nearly all of the rest of the changes to the stack are modeled after.

SteveMacenski, 2019-11-13:
I think this is the direction we have momentum in, its not worth changing at this point

closed 2019-11-13 by SteveMacenski