ros-navigation/navigation2 #1034: no transformation between the base_footprint frame  and the map frame

ZakariaChekakta, 2019-08-14:
## Bug report

**Required Info:**

- Operating System:
  -    Ubuntu 18.04

- Installation type:
  -   Source

- Version or commit hash:
  - Dashing version

- DDS implementation:
  - Default Fast-RTPS

#### Steps to reproduce issue

```
ros2 launch nav2_bringup nav2_simulation_launch.py
```

#### Expected behavior
display the robot model in RVIZ and it should be no issues in the transformation 
#### Actual behavior
[image]
[image]

SteveMacenski, 2019-08-14:
In order for the robot to navigate and have that costmap be updated, there has to be a valid map->base link transform. Do you see this consistently when launching say, 3x in a row?

If you subscribe to the tf topic do you see odom->base footprint and map->odom transforms over the pipe? If so then its rviz’s connection to the lot.

ZakariaChekakta, 2019-08-15:
@SteveMacenski , I have just run echo to the /tf topic to see if there is a transformation between map and base_footprint, and suddenly, in rviz  the robot model appears and the transformation too. I don't understand why it happened like that but thank you 
[image]

closed 2019-08-15 by ZakariaChekakta

crdelsey, 2019-08-15:
I suspect there are bugs in FastRTPS, the way ROS 2 is using it, or the way multicast is operating on your network. Sometimes it seems that publishers and subscribers don't see each other. Then when you do a ros2 topic echo, that triggers rediscovery and things start to work again.

I'm guessing though. I haven't had a chance to really debug these types of issues yet.