ros-navigation/navigation2 #1035: License in package.xml

shr-project, 2019-08-14:
Currently
https://github.com/SteveMacenski/navigation2-release/blob/release/dashing/nav2_amcl/package.xml#L21
says only LGPL, while:
https://github.com/SteveMacenski/navigation2-release/blob/release/dashing/nav2_amcl/src/main.cpp#L3
https://github.com/SteveMacenski/navigation2-release/blob/release/dashing/nav2_amcl/test/unit/test_localization_amcl.cpp#L3
says Apache-2
https://github.com/SteveMacenski/navigation2-release/blob/release/dashing/nav2_amcl/src/amcl_node.cpp#L5
https://github.com/SteveMacenski/navigation2-release/blob/release/dashing/nav2_amcl/include/nav2_amcl/amcl_node.hpp#L5
says LGPL-2.1+

superflore maps this to LGPL-2 in:
https://github.com/ros-infrastructure/superflore/blob/master/superflore/utils.py#L179
and LGPL-2 isn't valid SPDX Identifier in:
https://spdx.org/licenses/
so it would be better to declare the license correctly at the source - in package.xml.

I've already reported this in:
https://github.com/SteveMacenski/navigation2-release/issues/1
but was redirected to here.

SteveMacenski, 2019-08-14:
Thanks

@mkhansen-intel this looks like another place some default Intel headers have overwritten existing stricter copyrights to the original authors. My understanding of GPL-based licenses is that all works modified on top of them are also GPL-variant, I think that means all the AMCL code you guys ported needs to be LGPL.

Should someone at Intel take a few minutes and find all the places where Intel copyrights were added and make sure there's not more things in violation of licenses? There's a bunch of places where BSD has just been crushed after the port to Apache 2.0, but the original work still deserves its license

rotu, 2019-08-14:
@SteveMacenski, LGPL is much less "infectious" than the GPL. It can be linked with otherwise licensed code.

Also, we should standardize copyright identifiers. I like using the SPDX full name or identifier, but there are also names given in ament_copyright https://github.com/ament/ament_lint/blob/3714573db303b4d3d815f433b3ace1718df2ed2b/ament_copyright/ament_copyright/licenses.py#L40-L44

SteveMacenski, 2019-08-14:
well the problem is that alot of this stuff was not written by us originally (or even in the nav stack originally, AMCL was ripped out of Stage way back when). We have to respect their licenses regardless of what we would want.

I think AMCL is the only iffy one, everything else was BSD

crdelsey, 2019-08-15:
I think some of AMCL was originally LGPL. We added some new files under Apache 2.0, but should probably change the license to LGPL to avoid the mixed license. I expect, when you link them together directly, like we do, the resulting license would have the minimum set of overlapping permissions.

As I understand the license, that would be effectively the same as LGPL

SteveMacenski, 2019-08-15:
@crdelsey my understanding of LGPL is that you **cant** make code additions without being also LGPL. Alot of the stuff that's been done is refactoring, not feature additions, and should retain the original license

crdelsey, 2019-08-15:
> Alot of the stuff that's been done is refactoring, not feature additions, and should retain the original license

100% agree. I'm giving people the benefit of the doubt and assuming anything with a different license is new content. Where that assumption is incorrect, we need to fix that.

> my understanding of LGPL is that you cant make code additions without being also LGPL

IANAL, but looking at GPL where the rules are clearer in my mind, there is a concept of GPL compatible licenses. You can link a BSD 2-Clause component with a GPL component. The resulting whole would be licensed as GPL, since it is the more restrictive license.

I believe that is the concept in play here in AMCL. We have LGPL components (certain files) linked with Apache 2 components (other files). The resulting whole is LGPL as it is the most restrictive.

SteveMacenski, 2019-08-15:
I think the best bet here both (in my mind) architecturally & to keep in compliance is to move all the AMCL stuff back into AMCL, then reassign all the licenses to the original then... back away slowly... 

[image]

crdelsey, 2019-08-16:
Yeah. That would probably be the right thing to do. It seems we've polluted nav2_utils with LGPL, effectively turning that library into LGPL as well.

SteveMacenski, 2019-08-16:
It might be faster to find the git hash before AMCL was broken apart and just take the directory from then and nuke AMCL as we have it today, I dont think we made any big patches.

SteveMacenski, 2019-08-24:
I don't recall who ported / util-ized AMCL, is it someone still active that can return it to an isolated package?

PR #1074 (2019-08-29): Recombine AMCL library components with AMCL to have consistent license

---

## Basic Info

| Info | Please fill out this column |
| ------ | ----------- |
| Ticket(s) this addresses   | fixes #1035 |
| Primary OS tested on | NA |
| Robotic platform tested on | NA |

---

## Description of contribution in a few bullet points

* moves the map, sensors, pf, motion model and angleutils from nav2_util back to nav2_amcl, where it originally came from. This brings all LGPL components back into a single package.
* Re-licenses Intel contributions in nav2_amcl as LGPL to resolve conflicts and have a co [... cut]
closed 2019-08-29 by crdelsey