ros-navigation/navigation2 #1042: Can't launch map_server

routiful, 2019-08-19:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - binaries
- Version or commit hash:
  - ROS 2 Dashing Diademata - Patch Release 2
- DDS implementation:
  -  Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

```
$ ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py
$ros2 launch turtlebot3_navigation2 navigation2.launch.py use_sim_time:=True
```

#### Expected behavior

#### Actual behavior
```
robotis@robotis:~$ ros2 launch turtlebot3_navigation2 navigation2.launch.py 
[INFO] [launch]: All log files can be found below /home/robotis/.ros/log/2019-08-19-19-32-14-542748-robotis-3793
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [lifecycle_manager-1]: process started with pid [3805]
[INFO] [map_server-2]: process started with pid [3806]
[INFO] [amcl-3]: process started with pid [3807]
[INFO] [world_model-4]: process started with pid [3808]
[INFO] [dwb_controller-5]: process started with pid [3809]
[INFO] [navfn_planner-6]: process started with pid [3812]
[INFO] [recoveries_node-7]: process started with pid [3821]
[INFO] [bt_navigator-8]: process started with pid [3830]
[INFO] [rviz2-9]: process started with pid [3831]
[lifecycle_manager-1] [INFO] [lifecycle_manager]: Creating
[map_server-2] [INFO] [map_server]: Creating
[lifecycle_manager-1] [INFO] [lifecycle_manager]: Starting the system bringup...
[lifecycle_manager-1] [INFO] [lifecycle_manager]: Creating and initializing lifecycle service clients
[world_model-4] [INFO] [world_model]: Creating
[amcl-3] [INFO] [amcl]: Creating
[lifecycle_manager-1] [INFO] [lifecycle_manager]: Configuring and activating map_server
[bt_navigator-8] [INFO] [bt_navigator]: Creating
[rviz2-9] qt.qpa.screen: QXcbConnection: Could not connect to display 
[recoveries_node-7] [INFO] [recoveries]: Configuring Spin
[rviz2-9] Could not connect to any X display.
[dwb_controller-5] [INFO] [dwb_controller]: Creating
[navfn_planner-6] [INFO] [navfn_planner]: Creating
[ERROR] [rviz2-9]: process has died [pid 3831, exit code 1, cmd '/opt/ros/dashing/lib/rviz2/rviz2 -d /opt/ros/dashing/share/nav2_bringup/launch/nav2_default_view.rviz __node:=rviz2 __params:=/tmp/launch_params_k2zrtg5k'].
[world_model-4] [INFO] [global_costmap.global_costmap]: Creating
[recoveries_node-7] [WARN] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[recoveries_node-7] [INFO] [recoveries]: Configuring BackUp
[dwb_controller-5] [INFO] [local_costmap.local_costmap]: Creating
[map_server-2] [INFO] [map_server]: Configuring
[map_server-2] [INFO] [map_server]: OccGridLoader: Creating
[map_server-2] [INFO] [map_server]: OccGridLoader: Configuring
[map_server-2] [WARN] [map_server]: Mode parameter not set, using default value (trinary): bad conversion
[map_server-2] [WARN] []: Error occurred while doing error handling.
[map_server-2] [FATAL] [map_server]: Lifecycle node entered error state
[lifecycle_manager-1] [ERROR] [lifecycle_manager]: Failed to change state for node: map_server
[lifecycle_manager-1] [ERROR] [lifecycle_manager]: Failed to bring up node: map_server, aborting bringup
```
#### Additional information
I tested navigation2 packages in Gazebo simulation. But I can't launch map_server.
This error was occured when I tested it using turtlebot3 robot.

crdelsey, 2019-08-19:
Unfortunately, it is difficult to see the error messages from map server without ros2/rclcpp#777 which won't be available until Patch Release 3.

If I run the latest `navigation2.launch.py` from your repo (SHA ROBOTIS-GIT/turtlebot3@a7c13beb30 ) against the Dashing Patch Release 2, everything starts with no problems.

Usually, map server problems are caused by a problem with the path to the map. Could you check the map path to see if it looks correct? Run your launch file like you described above. Then, in another terminal, run:
```
ps ax | grep nav2_map_server
```
In the nav2_map_server command line, you should see parameter like `__params:=/tmp/tmpq1klk8yc`

Take a look in that file and make sure the `yaml_filename` parameter under `map_server` looks correct. On my machine it shows up as 
```
yaml_filename: /home/crdelsey/src/ros_dev_workspace/turtlebot3_ws/install/share/turtlebot3_navigation2/map/map.yaml
```

If that is correct, is the path on your machine a longer string than above? We've seen a problem where the maximum size of a string in the YAML file is less than the length of a long path.

orduno, 2019-08-19:
@routiful Does it fail if launching directly with `nav2_bringup_launch.py`?

routiful, 2019-08-20:
@crdelsey @orduno 
I've tested with launch  `nav2_bringup_launch.py` 
```
robotis@robotis:~$ ros2 launch nav2_bringup nav2_bringup_launch.py use_sim_time:=True map:='/opt/ros/dashing/share/nav2_bringup/launch/turtlebot3_world.yaml' params:='/opt/ros/dashing/share/nav2_bringup/launch/nav2_params.yaml'
```

terminal output
```
[lifecycle_manager-1] [INFO] [lifecycle_manager]: Creating
[lifecycle_manager-1] [INFO] [lifecycle_manager]: Starting the system bringup...
[lifecycle_manager-1] [INFO] [lifecycle_manager]: Creating and initializing lifecycle service clients
[lifecycle_manager-1] [INFO] [lifecycle_manager]: Configuring and activating map_server
[map_server-2] [INFO] [map_server]: Creating
[amcl-3] [INFO] [amcl]: Creating
[world_model-4] [INFO] [world_model]: Creating
[dwb_controller-5] [INFO] [dwb_controller]: Creating
[bt_navigator-8] [INFO] [bt_navigator]: Creating
[recoveries_node-7] [INFO] [recoveries]: Configuring Spin
[world_model-4] [INFO] [global_costmap.global_costmap]: Creating
[recoveries_node-7] [WARN] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[recoveries_node-7] [INFO] [recoveries]: Configuring BackUp
[navfn_planner-6] [INFO] [navfn_planner]: Creating
[map_server-2] [INFO] [map_server]: Configuring
[map_server-2] [INFO] [map_server]: OccGridLoader: Creating
[map_server-2] [INFO] [map_server]: OccGridLoader: Configuring
[map_server-2] [WARN] [map_server]: Mode parameter not set, using default value (trinary): bad conversion
[map_server-2] [INFO] [map_server]: Activating
[map_server-2] [INFO] [map_server]: OccGridLoader: Activating
[dwb_controller-5] [INFO] [local_costmap.local_costmap]: Creating
[lifecycle_manager-1] [INFO] [lifecycle_manager]: Configuring and activating amcl
[amcl-3] [INFO] [amcl]: Configuring
[amcl-3] [INFO] [amcl]: initTransforms
[amcl-3] [WARN] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[amcl-3] [INFO] [amcl]: initPubSub
[amcl-3] [INFO] [amcl]: Subscribed to map topic.
[amcl-3] [INFO] [amcl]: Activating
[lifecycle_manager-1] [INFO] [lifecycle_manager]: Configuring and activating world_model
[world_model-4] [INFO] [world_model]: Configuring
[world_model-4] [INFO] [global_costmap.global_costmap]: Configuring
[world_model-4] [INFO] [global_costmap.global_costmap]: Using plugin "static_layer"
[world_model-4] [INFO] [global_costmap.global_costmap]: Using plugin "obstacle_layer"
[world_model-4] [INFO] [global_costmap.global_costmap]: Subscribed to Topics: scan
[world_model-4] [INFO] [global_costmap.global_costmap]: Using plugin "inflation_layer"
[world_model-4] [INFO] [global_costmap.global_costmap]: StaticLayer: Resizing costmap to 384 X 384 at 0.050000 m/pix
[world_model-4] [INFO] [world_model]: Activating
[world_model-4] [INFO] [global_costmap.global_costmap]: Activating
[world_model-4] [INFO] [global_costmap.global_costmap]: Checking transform
[world_model-4] [INFO] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: . canTransform returned after 1.01 timeout was 1.
[amcl-3] [WARN] [amcl]: Waiting for map....
[world_model-4] [INFO] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: . canTransform returned after 1.007 timeout was 1.
[world_model-4] [INFO] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: . canTransform returned after 1.085 timeout was 1.
[amcl-3] [WARN] [amcl]: Waiting for map....
[world_model-4] [INFO] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: . canTransform returned after 1.094 timeout was 1.
[world_model-4] [INFO] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: . canTransform returned after 1.084 timeout was 1.
[amcl-3] [WARN] [amcl]: Waiting for map....
[world_model-4] [INFO] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: . canTransform returned after 1.095 timeout was 1.
[world_model-4] [INFO] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: . canTransform returned after 1.092 timeout was 1.
[amcl-3] [WARN] [amcl]: Waiting for map....
[world_model-4] [INFO] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: . canTransform returned after 1.091 timeout was 1.
[world_model-4] [INFO] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: . canTransform returned after 1.088 timeout was 1.
[amcl-3] [INFO] [amcl_rclcpp_node]: [signalFailure] Drop message: frame 'base_scan' at time 9.402 for reason(0)
[... cut ...]
[world_model-4] [INFO] [global_costmap.global_costmap_rclcpp_node]: [signalFailure] Drop message: frame 'base_scan' at time 11.404 for reason(0)
[world_model-4] [INFO] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: . canTransform returned after 1.083 timeout was 1.
```

Check file in /tmp/....
```
map_server:
ros__parameters: {use_sim_time: true, yaml_filename: /opt/ros/dashing/share/    nav2_bringup/launch/turtlebot3_world.yaml}
```

Do I miss something??

routiful, 2019-08-20:
There isn't to broadcast map_tf

```
RESULTS: for all Frames

Frames:
Frame: /base_link, published by <no authority available>, Average Delay: 1.56627e+09, Max Delay: 1.56627e+09
Frame: /base_scan, published by <no authority available>, Average Delay: 1.56627e+09, Max Delay: 1.56627e+09
Frame: /caster_back_link, published by <no authority available>, Average Delay: 1.56627e+09, Max Delay: 1.56627e+09
Frame: /imu_link, published by <no authority available>, Average Delay: 1.56627e+09, Max Delay: 1.56627e+09
Frame: /wheel_left_link, published by <no authority available>, Average Delay: 1.56627e+09, Max Delay: 1.56627e+09
Frame: /wheel_right_link, published by <no authority available>, Average Delay: 1.56627e+09, Max Delay: 1.56627e+09
Frame: base_footprint, published by <no authority available>, Average Delay: 1.56627e+09, Max Delay: 1.56627e+09

All Broadcasters:
Node: <no authority available> 59.5564 Hz, Average Delay: 1.56627e+09 Max Delay: 1.56627e+09
```

SteveMacenski, 2019-08-20:
How the stack currently works is that you have to set an initial pose before the map->odom transform is published

routiful, 2019-08-20:
@SteveMacenski 
Thank you for your comments. I've found [tb3_tutorial](https://github.com/mlherd/navigation2/tree/new-tutorials/doc/tb3_tutorial) and finally succeded to launch navigation2. 
While error messages was shown in terminal, I couldn't thought that click `2D initial pose`.
Thanks all :)

closed 2019-08-20 by routiful

SteveMacenski, 2019-08-20:
@orduno Perhaps a reason to start with an identity transformation, I have a strong feeling we’ll be fielding this question a bunch otherwise

robotpilot, 2019-08-20:
@SteveMacenski @orduno 
I agree with Steven. I've been using ROS and navigation for years, but I didn't think the program started like this. It's a good idea to write a clear description of this, but I think it's better to change the way to start the program. And I think that many people use navigation without using GUI like RViz.

mkhansen-intel, 2019-08-21:
PR #1018 will add the ability to specify a starting pose at launch, which can be used instead of setting the initial pose.