ros-navigation/navigation2 #1043: nav2_rviz_plugins (pose_tool) doesn't build with LLVM

oneattosecond, 2019-08-19:
Latest dashing-devel branch of nav2 doesn't build with LLVM... I get pages and pages of errors about missing members in the global namespace. Have googled and tried every random suggestion on the interwebs, but haven't found anything to get unblocked yet.

- Operating System:
  - MacOS 10.14.x, latest Xcode
- Installation type:
  - building from source
- Version or commit hash:
  - dashing-devel branch, which hasn't been touched in a few weeks
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
Try to compile with colcon, see pages of errors as per below

Starting >>> nav2_rviz_plugins
--- stderr: nav2_rviz_plugins                               
In file included from $NAV2/build/nav2_rviz_plugins/nav2_rviz_plugins_autogen/mocs_compilation.cpp:2:
In file included from $NAV2/build/nav2_rviz_plugins/nav2_rviz_plugins_autogen/JQVGUX53DN/moc_goal_tool.cpp  In file included from $NAV2/src/navigation2/navigation2/nav2_rviz_plugins/src/goal_tool.cpp:15:
In file included from $NAV2/src/navigation2/navigation2/nav2_rviz_plugins/include/nav2_rviz_plugins/goal_tool.hpp:22:
In file included from $ROS2/install/rviz_default_plugins/include/rviz_default_plugins/tools/pose/pose_tool.hpp:43:
In file included from $ROS2/install/rviz_ogre_vendor/opt/rviz_ogre_vendor/include/OGRE/OgreVector3.h:31:
In file included from $ROS2/install/rviz_ogre_vendor/opt/rviz_ogre_vendor/include/OGRE/OgrePrerequisites.h:376:
In file included from $ROS2/install/rviz_ogre_vendor/opt/rviz_ogre_vendor/include/OGRE/OgreStdHeaders.h:28:
$XCODEPATH/usr/include/c++/v1/cmath:313:9: error: no member named 'signbit' in the global namespace
using ::signbit;
      ~~^
:10:
In file included from $NAV2/build/nav2_rviz_plugins/nav2_rviz_plugins_autogen/JQVGUX53DN/../../../../src/navigation2/navigation2/nav2_rviz_plugins/include/nav2_rviz_plugins/goal_tool.hpp:22:
In file included from $ROS2/install/rviz_default_plugins/include/rviz_default_plugins/tools/pose/pose_tool.hpp:43:
In file included from $ROS2/install/rviz_ogre_vendor/opt/rviz_ogre_vendor/include/OGRE/OgreVector3.h:31:
In file included from $ROS2/install/rviz_ogre_vendor/opt/rviz_ogre_vendor/include/OGRE/OgrePrerequisites.h:376:
In file included from $ROS2/install/rviz_ogre_vendor/opt/rviz_ogre_vendor/include/OGRE/OgreStdHeaders.h:28:
$XCODEPATH/usr/include/c++/v1/cmath:313:9: error: no member named 'signbit' in the global namespace
using ::signbit;
      ~~^

#### Expected behavior
Compiles and able to use the pose tool to start navigating

#### Actual behavior
Doesn't compile

#### Additional information
N/A

SteveMacenski, 2019-08-19:
Hi thanks for the ticket!

Looking at your traceback, the issue isn't in Navigation2, but in RVIZ. This ticket in rviz2 looks relevant: https://github.com/ros2/rviz/issues/440 can you indicate if this is what you're seeing? This definitely looks like an OSX problem which may need some c++ preprocessor conditions.

oneattosecond, 2019-08-19:
Thanks for the rapid response. yes, that looks quite similar - however rviz builds for me just fine, and it's only the plugin that fails on my setup.

SteveMacenski, 2019-08-19:
Unfortunately I don't have an OSX machine to debug, the error is coming from your ogre client in compiling the plugin. If you can debug we'd appreciate the PR

@intel-folks any of you have a Mac?

oneattosecond, 2019-08-19:
I'm trying to dig into this, but my Cmake-fu is lacking :-/

I ran Cmake manually with VERBOSE on, and there are a *ton* of -isystem includes. I removed them all and added them back one at a time, and it appears what breaks this for me is an -isystem <macOS SDK path>/usr/include

If I remove the -isystem <MacOS SDK path>/usr/include from command line manually, navigation_dialog.cpp compiles/links - but I am struggling to figure out why Cmake is adding this system include... not seeing anything obvious in CMakeLists.txt that would add this include.

¯\_(ツ)_/¯

SteveMacenski, 2019-08-20:
Well looking at your trace, its not in the cmakelists of this package, its in the `rviz_ogre_vendor` compilation that has those flags. Off hand I don't know too much about that system but I think that rviz ticket is another good place to follow up

oneattosecond, 2019-08-20:
thanks, I'm not seeing it in rviz_ogre_vendor either but probably just overlooking something. will follow up over in the other ticket.

oneattosecond, 2019-09-04:
After working past https://github.com/ros-planning/navigation2/issues/1049, I recompiled from scratch and not hitting this. Possible dupe of https://github.com/ros-planning/navigation2/issues/1049?

SteveMacenski, 2019-09-04:
Also might be just a messed up cache. @oneattosecond can you close this then?

closed 2019-09-04 by oneattosecond