ros-navigation/navigation2 #1044: bt_navigator isn't calling on_configure --> can't set nav2 goals via topic publish

oneattosecond, 2019-08-19:
## Bug report

**Required Info:**

- Operating System:
  - MacOS 10.14.x
- Installation type:
  - source
- Version or commit hash:
  - dashing-devel
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - bt_navigator

#### Steps to reproduce issue
Set /initialpose and /move_base_simple/goal via "ros2 topic pub -1 <bunch of stuff>

Nothing happens

#### Expected behavior
bt_navigator should pick up the initial pose and goal and start navigation process

#### Actual behavior
Nothing happens, bt_navigator doesn't appear to intercept the goal and start navigation process

#### Additional information
It appears bt_navigator is subscribing to the mentioned topics, but nobody is invoking on_configure(). Manually hacked around this by calling on_configure, and this results in bt_navigator picking up the goal and printing the added "Yes I got a goal" --> but the goal is lost somewhere in the call to async_send_goal()...

SteveMacenski, 2019-08-19:
Ah yes, thanks for tracking that down, I hadn't gotten to it yet, I'll close my ticket since yours has more context https://github.com/ros-planning/navigation2/issues/1021

orduno, 2019-08-19:
@oneattosecond In nav2, a `LifecycleManager` node is responsible for transitioning the multiple nav2 `Lifecycle` nodes (bt_navigator, dwb, nav_fn, etc.) through the [different states](https://github.com/ros2/demos/blob/master/lifecycle/README.rst). The `on_configure` method on the node is called when a `configure` transition is requested, usually through the manager.

Once all the nodes have successfully transitioned to the `active` state, the stack is ready to receive goals.

It's possible that the lifecycle manager is not transitioning the nodes or it's not getting launched at all. Could you provide the following info:

- How are you launching the nodes?
- What is the terminal output?

oneattosecond, 2019-08-19:
How are you launching the nodes?
- via TB3 launch script from latest ROS2 branch: https://github.com/ROBOTIS-GIT/turtlebot3/blob/ros2/turtlebot3_navigation2/launch/navigation2.launch.py

What is the terminal output?
- OK I see what's happening here, lifecycle_manager (actually multiple nodes) are throwing an assert because the bt_xml_filename is >128 bytes long. I don't really want to move my repo to another directory, but I'll try that as a temporary workaround and see what happens next.

[lifecycle_manager-1] libc++abi.dylib: terminating with uncaught exception of type std::runtime_error: Failed to parse parameters from file '/var/folders/0m/441s_ghd5mz21g_f0bb597hc0000gn/T/tmpwnxxqstg': Scalar value at line 48 is bigger than 128 bytes, at ~/ros2_osx_dashing/src/ros2/rcl/rcl_yaml_param_parser/src/parser.c:849

snippet of /var/folders/0m/441s_ghd5mz21g_f0bb597hc0000gn/T/tmpwnxxqstg:
bt_navigator:
  ros__parameters:
    bt_xml_filename: /really_long_path_here/install/nav2_bt_navigator/behavior_trees/navigate_w_replanning_and_recovery.xml

oneattosecond, 2019-08-20:
OK, worked around that specific assert by rebuilding code with a shorter directory path. Shall I create an issue on tcl_yaml_param_parser to increase the max file size? 128 bytes seems like a quite an arbitrary limit, and failing with a non-obvious error code also seems bad.

However the reason I filed this bug was on_configure not being called and not being able to set goals via publishing to topics. After above workaround, I'm still not seeing lifecycle_manager configure bt_navigator and not able to set goals via ros2 topic pub. Will attach the console output shortly.

oneattosecond, 2019-08-20:
[console_output_no_bt_config.txt](https://github.com/ros-planning/navigation2/files/3518130/console_output_no_bt_config.txt)

oneattosecond, 2019-08-20:
note that I'm unable to build nav2_rviz_plugins (https://github.com/ros-planning/navigation2/issues/1043) and I'm not certain if that impacts this issue or not.

oneattosecond, 2019-08-20:
ok, if I ignore the errors and just click "2D pose estimate" in rviz, and give a semi-accurate pose, then bt_navigator is configured and I can start navigation by entering a manual goal via ros2 topic pub. bt_navigator promptly crashes (will file separate issue for that) but it's a start at least.

However, setting initial pose via ros2 topic pub does NOT result in bt_navigator being configured. I just see the below on console:

[amcl-3] [INFO] [amcl]: initialPoseReceived
[amcl-3] [INFO] [amcl]: Setting pose (1566261106.585851): 0.100 0.000 0.000
[amcl-3] [INFO] [amcl]: Received a 96 X 83 map @ 0.050 m/pix

No output from lifecycle_node is printed, so I am led to believe bt_navigator is not configured and navigation doesn't work.

Please do confirm the UI interaction is not supposed to be necessary? Shouldn't I be able to set initial pose and goals purely from command line / by publishing to the appropriate topics?

mkhansen-intel, 2019-08-20:
UI interaction should not be necessary if you're using our nav2_bringup_launch.py file and our nav2_params.yaml file (both in the nav2_bringup package). I believe the turtlebot3_navigation2_launch.py is including those and using them by default. If you're failing to bring up Rviz, you can launch Rviz without requiring the config file by commenting out this line of the launch file: https://github.com/ROBOTIS-GIT/turtlebot3/blob/a7c13beb30ae959841ae96abafbee5fbe96a8dbe/turtlebot3_navigation2/launch/navigation2.launch.py#L68

At that point at least you will have rviz running and you can add the topics you want to view manually. 

If lifecycle_manager is correctly running you should see each node in sequence come up with an "Activating" message. Once they're all up and active, it should have a message that says "The system is active"

SteveMacenski, 2019-08-20:
@mkhansen-intel I can say that on my machine, the nav2 goals via topic has never worked using the existing rviz pose tool. May be related, may not be, but I'd be interested to hear if anyone can get it to work right now as the code stands.

mkhansen-intel, 2019-08-20:
I've been mostly using the dashing-devel branch (see the demo video I posted to Discourse) and the topic works on that branch. Last I had tried the master branch it was working there but it's possible of course that something changed. Did you try it with another rmw layer like opensplice? We can test again today.

oneattosecond, 2019-08-20:
I'm also on dashing-devel branch of nav2, and I'm not sure how to try another rmw layer, I'll go read some wikis today.

mlherd, 2019-08-21:
@oneattosecond
Publishing to the goal topic should work. I will test it again today.

Here is example of how to use a different rmw layer. (Example for opensplice)
First, you need to install opensplice;

- sudo apt install libopensplice69

Next, you need to set the RMW_IMPLEMENTATION variable to rmw_opensplice_cpp;

- export RMW_IMPLEMENTATION=rmw_opensplice_cpp

If you want to go back to FastRTPS, either don't set any environment variables or just do;

- export RMW_IMPLEMENTATION=rmw_fastrtps_cpp

SteveMacenski, 2019-08-21:
This should not be dependent on DDS vendor, if it is, then that's a ticket to open

mkhansen-intel, 2019-08-22:
We have seen different issues with different RMW implementations, some are more reliable than others it seems. In this case since he's running on MacOS, he may have different bugs than we see on Ubuntu.

SteveMacenski, 2019-08-22:
@mkhansen-intel I'm also seeing the topic-based pose tool as broken in Ubuntu

oneattosecond, 2019-08-22:
fixing title, prior was confusing

mlherd, 2019-08-24:
@SteveMacenski @oneattosecond 

Today, I tested sending goals via command line. I was able to send goal pose by using 2D Nav Goal in Rviz also via command line as well.

- Test system 1:
    - Ubuntu 18.0
    - Ros2 Master
    - Navigation2 Master
    - Turtlebot 3 (real robot)
    - FastRTPS

- Test system 2:
    - Ubuntu 18.04
    - Ros2 Dashing Binary
    - Navigation2 Dashing Binary
    - Turtlebot 3 (real robot)
    - FastRTPS
    
     - In addition to the 2D Nav Goal tool, also I tested the Navigation2 Goal tool and it works.

In terminal:
- ros2 topic pub /move_base_simple/goal geometry_msgs/PoseStamped "{header: {stamp: {sec: 0}, frame_id: 'map'}, pose: {position: {x: 0.2, y: 0.0, z: 0.0}, orientation: {w: 1.0}}}"

This should move the robot 0.2 meters forward.

I haven't tested the Navigation 2 stack on MacOS.

mkhansen-intel, 2019-08-29:
> @mkhansen-intel I'm also seeing the topic-based pose tool as broken in Ubuntu

Are you sure you're seeing this? Can you see the costmap in Rviz when this happens? Our system test which runs in CI is publishing to the /goal topic and it works fine. I suspect something else is wrong.

mkhansen-intel, 2019-08-29:
@oneattosecond - can you update us if you got this issue resolved? As we mentioned we're not testing on MacOS right now and we don't see this issue in Ubuntu.

SteveMacenski, 2019-08-29:
@mkhansen-intel I couldn't tell you, I'm having all types of launching problems that I can't get to a position to even test that

oneattosecond, 2019-08-29:
@mkhansen-intel thanks for following up, I just tried two below commands and it seems nav2 stack is bringing up bt_navigator, etc. I'm not able to reproduce this just now.

***

ros2 topic pub -1 /initialpose geometry_msgs/PoseWithCovarianceStamped '{ header: {stamp: {sec: 0, nanosec: 0}, frame_id: "map"}, pose:   { pose: {position: {x: 0.1, y: 0.0, z: 0.0}, orientation: {w: 0.1}}, } }'

[lifecycle_manager-1] [INFO] [lifecycle_manager]: Configuring and activating bt_navigator
[bt_navigator-8] [INFO] [bt_navigator]: Configuring

Cost maps load in rviz after setting initial pose via command line, so that's nice.

ros2 topic pub -1 /move_base_simple/goal geometry_msgs/PoseStamped '{header: {stamp: {sec: 0, nanosec: 0}, frame_id: "map"}, pose: {position: {x: 0.2, y: 0, z: 0}, orientation: {w: 1}}}'

nav2 still crashes due to https://github.com/ros-planning/navigation2/issues/1049

mkhansen-intel, 2019-08-30:
When you say you're not reproducing this, is that on Mac or Ubuntu?

oneattosecond, 2019-08-30:
@mkhansen-intel Mac, I have limited testing on Ubuntu.

mkhansen-intel, 2019-08-30:
OK, then can you close this issue?

closed 2019-08-30 by oneattosecond