ros-navigation/navigation2 #1047: [Bt Navigator] Generic service caller BTNode & plugin

SteveMacenski, 2019-08-19:
In order to allow users to call services from the behavior tree, they would have to create a new `BTNode`-derived class and load it in the factory. I'd like to avoid that.

I propose the following
- a service and an action calling BTNode with pluginlib interfaces to fill in the messages
  - the action calling would be a derivative of `bt_action_node.hpp`
  - the service node would be similar to my clear costmaps service derivative of the same
- plugin definitions that allow them to `createMessage()` `sendMessage()` so they can use custom types
- transition the existing `_action.hpp` files to use this plugin interface

An example of a new API for it `<BTServiceCaller name="plugin_name"/>` to have the `generic_service_caller.hpp` call this plugin's `createMessage()` `sendMessage()` `wasSuccessful()` etc.

SteveMacenski, 2019-08-19:
To be more clear I see 3 options to run time configurability for interacting with our components in the behavior tree:

1) BT has super generic node implementations that call services/topics/actions. They can do anything and enforce no type of organization, but also comes with the "bad" of being really nasty to read. There will be a set of plugins for the BT

2) BT has a finite set of node implementations for classes of things that interact with it (planner, controller, recovery, conditions, etc), each of these implementations has a standard set of messages we use to communicate between the BT and the servers they're calling. This enforces API on the actions/services/topic **types** and names are runtime configurable in the BT

3) Classes the BT interacts with (planner, controller, recovery, etc) are loaded as plugins in generic action servers which themselves expose unique topics which BT node implementations have essentially hardcoded. This comes with the benefit of pushing all the API elements to the pluginlib definitions and the BT is really not changing much between one and other. The drawback here is if you want multiple of a class running, the single BT node implementation needs to have an additional param field for the string in some map of plugins that hte classes the BT interacts with (controller, planner, recovery, etc) selects. Example: `<planner args="..." name="dwb">` and in the generic local planner action server that has loaded some `N` plugins into a `map<string, plugin>` would take the name and utilize the appropriate plugin `map[name]->computeControlEffort()`.

I think 3 is the most "clean and easily understood", 1 is the most general, and 2 would let us have organized chaos since there's virtually no enforcement of what's happening beyond message types

crdelsey, 2019-08-20:
I think a picture or some simple pseudo-code of the 3 options would help understanding.

My expectation of BT navigator was that it would provide nodes for specific actions or conditions, such as the existing FollowPath and ComputePath nodes. These would not be built into the BT navigator, but would instead be plugins we provide.  End users can then add any additional plugins they want beyond the set we provide.

The navigator part of BT navigator becomes just an BT tree definition and extra plugins provided to a generic ROS BT node in this model.

SteveMacenski, 2019-08-21:
I think we covered this in the Riot chat -- summary 

```

I think we’re on the same page then, you’re thinking maybe plugins on both sides of an action server, when I’m thinking mainly 1 plugin for a function and 1 Bt node implementation for each type of function (and a generic Bt node implementation with a plugin for the things that dont neatly fit in those boxes). Example: there's a PlanningActionBTNode tree implementation that can be called as <PlanningActionBTNode name="navfn/planning"/>, versus plugins would be less nice to read as <PluginBTNode plugin="plugin_name" name="navfn/planning"/> and then requires you to write plugins on either side of the actual actual server for each new algorithm

```

crdelsey, 2019-08-21:
> BTPlugin for things that don't "quite" fit

I think this last one maybe needs be thought out a bit more. This sounds a bit like you are proposing to make a "DoAnything" BTPlugin. That seems too generic to be useful, or alternatively too open for abuse, depending on how exactly it is implemented. I guess it depends on exactly what you had in mind for this.

SteveMacenski, 2019-08-22:
I'm proposing we do the actual organization, but also include a do anything plugin because we're not trying to stop people from doing new cool things, but it should be good if the core things weren't based on top of something that generic.

crdelsey, 2019-08-22:
I'm not seeing a straight-forward path to creating a "DoAnything" component, what with every communication mechanism in ROS being strongly typed. I feel like we'd need to create a node for every data type (eg. an "EmptyServiceNode", "BoolServiceNode", "PointServiceNode" etc)

I'd be interested to see how hard it is to write an actual BT plugin using the new library. If it's easy, maybe it's good enough to have someone create a custom BT plugin when they want to do something outside of our design.

I can try to port over to the new BT library and convert existing nodes to a BT plugin next week to see where we stand.

SteveMacenski, 2019-08-22:
That's fine, I dont think the "do anything" plugin is needed out the gate. It's a "nice to have" without a specific purpose in mind more than a "its needed by something in particular". Maybe someone builds some super cool thing that just needs to be injected into the tree that's where it might be useful.

mkhansen-intel, 2019-08-22:
@SteveMacenski - can you create a diagram for this and we can discuss in the Nav2 WG?

SteveMacenski, 2019-08-22:
In progress

SteveMacenski, 2019-08-22:
We'll go over this in the WG, but wanted to post for people to look over beforehand.

[image]

[image]

[image]

SteveMacenski, 2019-08-22:
Some design thinking from here too https://github.com/ros-planning/navigation2/issues/1033

SteveMacenski, 2019-08-22:
Update from weekly: looks like I'll be starting in this direction

mkhansen-intel, 2019-08-29:
@SteveMacenski - can you draw up a high level diagram for this that can be used as documentation. Your hand-drawn version was fine for a first draft discussion but I'd like to see something that can be used in our docs & tutorials. We'll probably need a plugin tutorial now too.

SteveMacenski, 2019-11-14:
With BT plugins and the multiple controllers/planners this is done

closed 2019-11-14 by SteveMacenski