ros-navigation/navigation2 #1049: bt_navigator crashes immediately after setting initial pose and goal (MacOS)

oneattosecond, 2019-08-20:
## Bug report

**Required Info:**

- Operating System:
  - MacOS 10.14.6
- Installation type:
  - source
- Version or commit hash:
  - latest dashing-devel
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - bt_navigator

#### Steps to reproduce issue
- build all the packages
- run navigation via TB3 launch script, point to correct map.xml
- use UI to set the initial pose (workaround for https://github.com/ros-planning/navigation2/issues/1044)
- set the goal pose via command line (first check that it's pretty close to the goal pose listed in logs after clicking above. Note this is a workaround for https://github.com/ros-planning/navigation2/issues/1043)

ros2 topic pub -1 /move_base_simple/goal geometry_msgs/PoseStamped '{header: {stamp: {sec: 0, nanosec: 0}, frame_id: "map"}, pose: {position: {x: 1.35, y: -0.1, z: -0.3}, orientation: {w: 0.0}}}'

#### Expected behavior
TB3 starts navigating to desired position, or tells me that it can't

#### Actual behavior
bt_navigator crashes immediately without navigating

looks like a memory stomper, sometimes someone is de-ref'ing a nearly-null pointer, sometimes malloc aborts as it notices memory corruption. Either way, bt_navigator is crashing immediately :(

#### Additional information
Console and .crash logs uploaded here
[bt_navigator__crash_logs_and_console_logs.zip](https://github.com/ros-planning/navigation2/files/3522009/bt_navigator__crash_logs_and_console_logs.zip)

SteveMacenski, 2019-08-20:
Your issue if your bad request, you cannot set a quaternion to (0,0,0,0). That's invalid

oneattosecond, 2019-08-20:
@SteveMacenski thanks again for the rapid reply...

ros2 topic pub -1 /move_base_simple/goal geometry_msgs/PoseStamped '{header: {stamp: {sec: 0, nanosec: 0}, frame_id: "map"}, pose: {position: {x: 1.35, y: -0.1, z: -0.3}, orientation: {w: 0.0}}}'

Sorry, which one above is the invalid quaternion? orientation is an array (32? of them?) of floats, so setting it to all zero should be fine, no?

Not sure what I'm doing wrong... it seems non-optimal to crash in an obtuse way if the user is doing something silly :-/

SteveMacenski, 2019-08-20:
Please look at the definition for the topic your publishing `geometry_msgs/PoseStamped` -- a guiding question would be "what does XYZW mean?" and then the direct answer is quaternions define as 0,0,0,0 are not valid quaternions.

oneattosecond, 2019-08-20:
Ok, I see what happened there. I copy/pasted from an earlier attempt to set /initialpose, and that message is a geometry_msgs/PoseStampeWithCovariance, so yes, me asking for 0,0,0,0 doesn't make a lot of sense. I'm still not sure why there's a memory stomper instead of an error, but perhaps that's of lesser importance.

Unfortunately I tried setting orientation to 0.1, 0.1, 0.1, 0.1 hoping for better results and now I get a consistent trap in std::bad_alloc. Is 0.1, 0.1, 0.1, 0.1 a bogus orientation as well?

ros2 topic pub -1 /move_base_simple/goal geometry_msgs/PoseStamped '{header: {stamp: {sec: 0, nanosec: 0}, frame_id: "map"}, pose: {position: {x: 0.0, y: -1.2, z: -1.184}, orientation: {x: 0.1, y: 0.1, z: 0.1, w: 0.1}}}'

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
abort() called
terminating with uncaught exception of type std::bad_alloc: std::bad_alloc

Thread 21 crashed with X86 Thread State (64-bit):
  rax: 0x0000000000000000  rbx: 0x000070000e261000  rcx: 0x000070000e25fc28  rdx: 0x0000000000000000
  rdi: 0x0000000000002007  rsi: 0x0000000000000006  rbp: 0x000070000e25fc60  rsp: 0x000070000e25fc28
   r8: 0x000070000e25faf0   r9: 0x000070000e25fcc0  r10: 0x0000000000000000  r11: 0x0000000000000206
  r12: 0x0000000000002007  r13: 0x0000003000000008  r14: 0x0000000000000006  r15: 0x000000000000002d
  rip: 0x00007fff6bbd52c6  rfl: 0x0000000000000206  cr2: 0x00007fffa2302188
  
Logical CPU:     0
Error Code:      0x02000148
Trap Number:     133

SteveMacenski, 2019-08-20:
Yes. Please look into quaternions. If you're looking for a default to test use (0,0,0,1). 

But in general quaternions are everywhere, I'd do some reading on it

oneattosecond, 2019-08-21:
thanks, will do the needed reading.

setting orientation to (0,0,0,1) results in a consistent NULL pointer de-ref and/or SIGSEGV unfortunately... I'll do some study and try to figure out a valid & useful orientation Quaternion that works & makes sense for my map...

ros2 topic pub -1 /move_base_simple/goal geometry_msgs/PoseStamped '{header: {stamp: {sec: 0, nanosec: 0}, frame_id: "map"}, pose: {position: {x: 0.0, y: -1.2, z: -1.184}, orientation: {x: 0, y: 0, z: 0, w: 1}}}

Crashed Thread:        21
Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY
Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [61596]

mkhansen-intel, 2019-08-21:
@oneattosecond - have you tested using Rviz by following our nav2_bringup README instructions? Does that work? We definitely don't see this problem but admit we're not testing on MacOS.

oneattosecond, 2019-08-21:
@mkhansen-intel I can't test with rviz due to https://github.com/ros-planning/navigation2/issues/1043, which I need to follow up on their suggestion today & see if newer rviz works

I'm just launching nav via TB3 launch scripts, and hitting a bunch of issues, I'll try updating to TB3's latest support package (released last night) and see if that improves anything...

oneattosecond, 2019-08-23:
FWIW, still blocked on testing with Rviz (#1043), and unfortunately updating to latest TB3 support package didn't help here on Mac. If I set an initial pose with UI, then try to set a goal pose via above test orientation - get the same crash as before.

oneattosecond, 2019-08-24:
Enabled ASAN for bt_navigator, pasting details here in case someone is able to see what leads up to the corrupted memory... apologies for the long post, adding here instead of as attachment.

Note this may / may not be the actual cause of the crash, but ASAN is aborting after receiving goal due to the below.

[bt_navigator-8] ==53199==ERROR: AddressSanitizer: requested allocation size 0x612000009dc0 (0x61200000adc0 after adjustments for alignment, red zones etc.) exceeds maximum supported size of 0x10000000000 (thread T21)
[bt_navigator-8]     0 0x10c4c5d22 in wrap__Znam (libclang_rt.asan_osx_dynamic.dylib:x86_64h+0x68d22)
[bt_navigator-8]     1 0x10a5fd38a in coroutine::resume(unsigned int) coroutine.h:335
[bt_navigator-8]     2 0x10a5fcfce in BT::CoroActionNode::executeTick() action_node.cpp:156
[bt_navigator-8]     3 0x10a691c5b in BT::FallbackNode::tick() fallback_node.cpp:35
[bt_navigator-8]     4 0x10a60ea42 in BT::TreeNode::executeTick() tree_node.cpp:38
[bt_navigator-8]     5 0x10b1dce7d in nav2_behavior_tree::RateController::tick() rate_controller_node.hpp:75
[bt_navigator-8]     6 0x10a60ea42 in BT::TreeNode::executeTick() tree_node.cpp:38
[bt_navigator-8]     7 0x10a68bdcb in BT::SequenceNode::tick() sequence_node.cpp:32
[bt_navigator-8]     8 0x10a60ea42 in BT::TreeNode::executeTick() tree_node.cpp:38
[bt_navigator-8]     9 0x10b3ccf2d in nav2_bt_navigator::RecoveryNode::tick() recovery_node.cpp:45
[bt_navigator-8]     10 0x10a60ea42 in BT::TreeNode::executeTick() tree_node.cpp:38
[bt_navigator-8]     11 0x10958254a in nav2_behavior_tree::BehaviorTreeEngine::run(std::__1::unique_ptr<BT::Tree, std::__1::default_delete<BT::Tree> >&, std::__1::function<void ()>, std::__1::function<bool ()>, std::__1::chrono::duration<long long, std::__1::ratio<1l, 1000l> >) behavior_tree_engine.cpp:83
[bt_navigator-8]     12 0x10b0da789 in nav2_bt_navigator::BtNavigator::navigateToPose() bt_navigator.cpp:194
[bt_navigator-8]     13 0x10b1644cb in decltype(*(std::__1::forward<nav2_bt_navigator::BtNavigator*&>(fp0)).*fp()) std::__1::__invoke<void (nav2_bt_navigator::BtNavigator::*&)(), nav2_bt_navigator::BtNavigator*&, void>(void (nav2_bt_navigator::BtNavigator::*&&&)(), nav2_bt_navigator::BtNavigator*&&&) type_traits:4280
[bt_navigator-8]     14 0x10b16437f in std::__1::__bind_return<void (nav2_bt_navigator::BtNavigator::*)(), std::__1::tuple<nav2_bt_navigator::BtNavigator*>, std::__1::tuple<>, __is_valid_bind_return<void (nav2_bt_navigator::BtNavigator::*)(), std::__1::tuple<nav2_bt_navigator::BtNavigator*>, std::__1::tuple<> >::value>::type std::__1::__apply_functor<void (nav2_bt_navigator::BtNavigator::*)(), std::__1::tuple<nav2_bt_navigator::BtNavigator*>, 0ul, std::__1::tuple<> >(void (nav2_bt_navigator::BtNavigator::*&)(), std::__1::tuple<nav2_bt_navigator::BtNavigator*>&, std::__1::__tuple_indices<0ul>, std::__1::tuple<>&&) functional:2216
[bt_navigator-8]     15 0x10b1642c6 in std::__1::__bind_return<void (nav2_bt_navigator::BtNavigator::*)(), std::__1::tuple<nav2_bt_navigator::BtNavigator*>, std::__1::tuple<>, __is_valid_bind_return<void (nav2_bt_navigator::BtNavigator::*)(), std::__1::tuple<nav2_bt_navigator::BtNavigator*>, std::__1::tuple<> >::value>::type std::__1::__bind<void (nav2_bt_navigator::BtNavigator::*)(), nav2_bt_navigator::BtNavigator*>::operator()<>() functional:2249
[bt_navigator-8]     16 0x10b16419c in decltype(std::__1::forward<std::__1::__bind<void (nav2_bt_navigator::BtNavigator::*)(), nav2_bt_navigator::BtNavigator*>&>(fp)()) std::__1::__invoke<std::__1::__bind<void (nav2_bt_navigator::BtNavigator::*)(), nav2_bt_navigator::BtNavigator*>&>(std::__1::__bind<void (nav2_bt_navigator::BtNavigator::*)(), nav2_bt_navigator::BtNavigator*>&&&) type_traits:4339
[bt_navigator-8]     17 0x10b16414c in void std::__1::__invoke_void_return_wrapper<void>::__call<std::__1::__bind<void (nav2_bt_navigator::BtNavigator::*)(), nav2_bt_navigator::BtNavigator*>&>(std::__1::__bind<void (nav2_bt_navigator::BtNavigator::*)(), nav2_bt_navigator::BtNavigator*>&&&) __functional_base:349
[bt_navigator-8]     18 0x10b160f60 in std::__1::__function::__func<std::__1::__bind<void (nav2_bt_navigator::BtNavigator::*)(), nav2_bt_navigator::BtNavigator*>, std::__1::allocator<std::__1::__bind<void (nav2_bt_navigator::BtNavigator::*)(), nav2_bt_navigator::BtNavigator*> >, void ()>::operator()() functional:1562
[bt_navigator-8]     19 0x109582214 in std::__1::function<void ()>::operator()() const functional:1913
[bt_navigator-8]     20 0x10b1a196c in decltype(std::__1::forward<std::__1::function<void ()> >(fp)()) std::__1::__invoke<std::__1::function<void ()> >(std::__1::function<void ()>&&) type_traits:4339
[bt_navigator-8]     21 0x10b1a18d4 in void std::__1::__thread_execute<std::__1::unique_ptr<std::__1::__thread_struct, std::__1::default_delete<std::__1::__thread_struct> >, std::__1::function<void ()> >(std::__1::tuple<std::__1::unique_ptr<std::__1::__thread_struct, std::__1::default_delete<std::__1::__thread_struct> >, std::__1::function<void ()> >&, std::__1::__tuple_indices<>) thread:342
[bt_navigator-8]     22 0x10b1a0668 in void* std::__1::__thread_proxy<std::__1::tuple<std::__1::unique_ptr<std::__1::__thread_struct, std::__1::default_delete<std::__1::__thread_struct> >, std::__1::function<void ()> > >(void*) thread:352
[bt_navigator-8]     23 0x7fff5c22b2ea in _pthread_body (libsystem_pthread.dylib:x86_64+0x32ea)
[... cut ...]
[bt_navigator-8]     13 0x1092dbf69 in std::__1::shared_ptr<nav2_bt_navigator::BtNavigator> std::__1::shared_ptr<nav2_bt_navigator::BtNavigator>::make_shared<>() memory:4327
[bt_navigator-8]     14 0x1092da7d3 in std::__1::enable_if<!(is_array<nav2_bt_navigator::BtNavigator>::value), std::__1::shared_ptr<nav2_bt_navigator::BtNavigator> >::type std::__1::make_shared<nav2_bt_navigator::BtNavigator>() memory:4706
[bt_navigator-8]     15 0x1092da513 in main main.cpp:23
[bt_navigator-8]     16 0x7fff5c0373d4 in start (libdyld.dylib:x86_64+0x163d4)
[bt_navigator-8] 
[bt_navigator-8] ==53199==ABORTING

oneattosecond, 2019-09-03:
memory corruption appears to be happening due to makecontext/swapcontext, which are deprecated on MacOS (with no direct replacement AFAIK). Some hackery with #defines & order of headers resolves this crash, but I'm not sure why

change 1: move include of ucontext.h to before STACK_LIMIT is defined in coroutine.h
change 2: include coroutine.h before action_node.h in action_node.cpp

SteveMacenski, 2019-09-03:
Go ahead and submit a PR and lets take a look. I don't see why that doesnt get merged today.

SteveMacenski, 2019-10-10:
@oneattosecond is this still something your compelled to resolve?

oneattosecond, 2019-10-10:
@SteveMacenski yes, without the header swap I described earlier, bt_navigator crashes on receipt of a goal (via topic or GUI)

SteveMacenski, 2019-10-10:
Can you submit that PR then? None of the rest of us have macs to do/test for you unfortunately.

SteveMacenski, 2019-11-12:
@oneattosecond update?

SteveMacenski, 2019-11-12:
V3 of BT library resolved

closed 2019-11-12 by SteveMacenski