ros-navigation/navigation2 #1056: world_model dies under OpenSplice with runtime symbol lookup

rotu, 2019-08-22:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Binaries
- Version or commit hash:
  - ROS2 Testing prerelease distribution
- DDS implementation:
  - OpenSplice
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
Try to launch Nav2 Stack with this launch file:
https://github.com/RoverRobotics/openrover-ros2/blob/dashing/openrover_demo/launch/nav2.launch.py

#### Expected behavior
Nav2 stack comes up.

#### Actual behavior

World model node crashes with this log:
```
[world_model-1] [INFO] [world_model]: Activating
[world_model-1] [INFO] [global_costmap.world_model]: Activating
[world_model-1] [INFO] [global_costmap.world_model]: Checking transform
[world_model-1] [INFO] [global_costmap.world_model]: Timed out waiting for transform from base_link to map to become available, tf error: . canTransform returned after 1.00017 timeout was 1.
[world_model-1] [INFO] [global_costmap.world_model]: Timed out waiting for transform from base_link to map to become available, tf error: . canTransform returned after 1.00946 timeout was 1.
[world_model-1] [INFO] [global_costmap.world_model]: Timed out waiting for transform from base_link to map to become available, tf error: . canTransform returned after 1.00923 timeout was 1.
[world_model-1] [INFO] [global_costmap.world_model]: Timed out waiting for transform from base_link to map to become available, tf error: . canTransform returned after 1.00885 timeout was 1.
[world_model-1] [INFO] [global_costmap.world_model]: Timed out waiting for transform from base_link to map to become available, tf error: . canTransform returned after 1.00966 timeout was 1.
[world_model-1] [INFO] [global_costmap.world_model]: Timed out waiting for transform from base_link to map to become available, tf error: . canTransform returned after 1.00922 timeout was 1.
[world_model-1] [INFO] [global_costmap.world_model]: Timed out waiting for transform from base_link to map to become available, tf error: . canTransform returned after 1.00944 timeout was 1.
[world_model-1] [INFO] [global_costmap.world_model]: Timed out waiting for transform from base_link to map to become available, tf error: . canTransform returned after 1.00878 timeout was 1.
[world_model-1] [INFO] [global_costmap.world_model]: Timed out waiting for transform from base_link to map to become available, tf error: . canTransform returned after 1.0097 timeout was 1.
[world_model-1] [INFO] [global_costmap.world_model]: Timed out waiting for transform from base_link to map to become available, tf error: . canTransform returned after 1.00004 timeout was 1.
[world_model-1] [INFO] [global_costmap.world_model]: start
[world_model-1] /home/dan/ros2_ws/install/lib/nav2_world_model/world_model: symbol lookup error: /home/dan/ros2_ws/install/lib/libnav2_msgs__rosidl_typesupport_opensplice_cpp.so: undefined symbol: _Z43__builtin_interfaces_msg_dds__Time___copyInP8c_base_sPKN18builtin_interfaces3msg4dds_5Time_EP34_builtin_interfaces_msg_dds__Time_
[ERROR] [world_model-1]: process has died [pid 14574, exit code 127, cmd '/home/dan/ros2_ws/install/lib/nav2_world_model/world_model __node:=world_model __params:=/home/dan/ros2_ws/install/share/openrover_demo/config/nav2.yaml'].
```

SteveMacenski, 2019-08-22:
maybe builtin interfaces needs to be added to be linked against @rotu does world model have that?

rotu, 2019-08-23:
https://github.com/ros-planning/navigation2/blob/dashing-devel/nav2_world_model/CMakeLists.txt
https://github.com/ros-planning/navigation2/blob/dashing-devel/nav2_msgs/CMakeLists.txt
https://github.com/ros2/rcl_interfaces/blob/dashing/builtin_interfaces/CMakeLists.txt

@SteveMacenski It should be pulling it in recursively through `nav2_msgs`. Moving the dependency "builtin_interfaces" in `nav2_msgs/CMakeLists.txt` from `rosidl_generate_interfaces( ... DEPENDENCIES builtin_interfaces)` to `ament_export_dependencies( ... builtin_interfaces)` did not resolve the issue (though it's possible I didn't build it right?).

It's likely that this has something to do with that `builtin_interfaces` is in installed from ROS2 binaries but `nav2_msgs` is in my overlay, installed from source.

SteveMacenski, 2019-08-23:
Yeah it was a crap shot of "maybe that'll do it", there's always some weirdness with people exporting dependencies correctly so it wouldn't have shocked me if just explicitly calling it fixed the issue. 

Anyhow, good report, we should fix this

crdelsey, 2019-08-23:
I was informed  that some ros2 libraries build differently, depending on whether libopensplice or rti Connext are available in the environment at build time. I suspect this problem is a result of that behavior.

If they are available at build time, the library builds with the ability to link to the DDS layer at runtime, otherwise the library builds with a direct link to the FastRTPS library

Is it possible that you have two packages that were built with different initial environments? One where libopensplice is available and another where it is not?

rotu, 2019-08-23:
I don't know what you mean by "different initial environments". ROS2 Dashing is being installed with `apt` and nav2 from source. This does not seem to happen when nav2 packages are installed with `apt`. I *think* this setup is supposed to work, but correct me if wrong.

SteveMacenski, 2019-08-23:
He means if in one case opensplice was installed before build or not, and/or if the RWM parameter was set to use it or not before builds

rotu, 2019-08-23:
Well, I don't know what the settings are in the build farm. In my local build environment, which is compiling nav2, OpenSplice is installed and `RMW_IMPLEMENTATION=rmw_opensplice_cpp` at the time of build.