ros-navigation/navigation2 #1060: Compilation Error: missing: GRAPHICSMAGICKCPP_INCLUDE_DIRS

lbegani, 2019-08-23:
This is probably due to new missing dependency libgraphicsmagick++1-dev. Found this info from the git log. This must be documented in instructions.
Error Log: 
$ colcon build --symlink-install
Starting >>> nav2_common
Starting >>> nav_2d_msgs
Finished <<< nav2_common [0.52s]                                                               
Starting >>> nav2_msgs
Starting >>> nav2_voxel_grid
Finished <<< nav2_voxel_grid [0.55s]                                                                                                
Finished <<< nav_2d_msgs [2.23s]                                                                  
Starting >>> dwb_msgs
Finished <<< nav2_msgs [3.21s]                                                                  
Starting >>> nav2_util
Starting >>> nav_2d_utils
Finished <<< nav_2d_utils [0.47s]                                                                                                 
Finished <<< dwb_msgs [2.58s]                                                                    
Finished <<< nav2_util [1.10s]
Starting >>> nav2_behavior_tree
Starting >>> nav2_lifecycle_manager
Starting >>> nav2_map_server
Starting >>> nav2_amcl
Starting >>> nav2_navfn_planner
Starting >>> nav2_dynamic_params
--- stderr: nav2_map_server                                                                                                                  
CMake Error at /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  GRAPHICSMAGICKCPP__LIBRARIES (missing: GRAPHICSMAGICKCPP_INCLUDE_DIRS)
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  cmake_modules/FindGRAPHICSMAGICKCPP.cmake:31 (find_package_handle_standard_args)
  CMakeLists.txt:15 (find_package)

---
Failed   <<< nav2_map_server	[ Exited with code 1 ]
Aborted  <<< nav2_lifecycle_manager                                                                                                  
Aborted  <<< nav2_navfn_planner                                                                                        
Aborted  <<< nav2_dynamic_params                                                                                        
Aborted  <<< nav2_amcl                                                                                               
Aborted  <<< nav2_behavior_tree                                            

Summary: 7 packages finished [22.8s]
  1 package failed: nav2_map_server
  5 packages aborted: nav2_amcl nav2_behavior_tree nav2_dynamic_params nav2_lifecycle_manager nav2_navfn_planner
  1 package had stderr output: nav2_map_server
  13 packages not processed

## Bug report

**Required Info:**

- Operating System:
  - 
- Installation type:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

SteveMacenski, 2019-08-23:
Did you rosdep install things? If you haven't used rosdep in the last week, that dependency is new and you'll need to run rosdep. 

Also, please fill in your OS, commit, branch info in the ticket format. That is necessary to start thinking about this problem

orduno, 2019-08-23:
You can find the command [here](https://index.ros.org/doc/ros2/Installation/Eloquent/Linux-Development-Setup/#install-dependencies-using-rosdep)

mlherd, 2019-08-23:
@lbegani 
Try
- rosdep install -y -r -q --from-paths src --ignore-src --rosdistro dashing -y
- colcon build --symlink-install

lbegani, 2019-09-04:
> @lbegani - A pull request to add the rosdep command to the BUILD.md file would be welcome if you have time!

@mkhansen-intel - Kindly review the PR to fix the build instructions - https://github.com/ros-planning/navigation2/pull/1095