ros-navigation/navigation2 #1068: [nav2_rviz_plugins] want to move a 'cancel button' in dialog box to panel region

hyunseok-yang, 2019-08-27:
## Bug report

It is not a bug.

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - binaries
- Version or commit hash:
  - dashing
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

I tried to make a new button as a 'navigation cancellation' on panel side.
Creating a Qt GUI button on a panel was easy but it's hard to implement action cancellation when I set the goal from external method not a inside of rviz2.

external method means......
I just call nav2 goal via ros2cli

ros2 action send_goal /NavigateToPose nav2_msgs/action/NavigateToPose "pose: {header: {frame_id: map}, pose: {position: {x: 6.0, y: 2.0, z: 0.0}, orientation:{x: 0.0, y: 0.0, z: 0, w: 1.0000000}}}"

Waiting for an action server to become available...
Sending goal:
     pose:
  header:
    stamp:
      sec: 0
      nanosec: 0
    frame_id: map
  pose:
    position:
      x: 6.0
      y: 2.0
      z: 0.0
    orientation:
      x: 0.0
      y: 0.0
      z: 0.0
      w: 1.0

Goal accepted with ID: cfa77d4f903747e5b2104315bc0a2840

Result:
    result:
  structure_needs_at_least_one_member: 0

```
void
Nav2Panel::onCancelButtonPressed()
{
  auto future_cancel = action_client_->async_cancel_goal(goal_handle_);
  if (rclcpp::spin_until_future_complete(client_node_, future_cancel) !=
      rclcpp::executor::FutureReturnCode::SUCCESS)
  {
    RCLCPP_ERROR(client_node_->get_logger(), "Failed to cancel goal");
    return;
  }
 }
```

I don't know how to get goal_handle without async_send_goal() function.

#### Expected behavior
It can easily cancel the navigation(path following?) in rviz panel wherever set the goal.

#### Actual behavior
navigation stop through cancel button in rviz panel

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

hyunseok-yang, 2019-08-28:
@crdelsey Hi~!, well...  I can say there is no specific "reason" for these behavior..

BUT, in my opinion, I think we need to open the chance to control an action like goal stop by other method especially in "Multi-Robot control environment".

For example, what if someone want to stop immediately the robot which was received the "move2goal" command by the other user?

User A ---> (move2goal command!!) ----> [Robot]  <----- (want to stop driving) <---- User B

hyunseok-yang, 2019-08-28:
@SteveMacensk I agree that "RVIZ isn't real production tool' and, of course, there is no obvious reason to implement a cancel button on nav2 rviz plugin. But I just wanted to put it on panel region in my own purpose. 
(Just think the level of tutorial ..... and I was digging if there is a method to cancel the goal handle.. that's all.)

And, it doesn't need to be able to cancel a goal in RVIZ by any multiple user. I didn't intend to implement that function on rviz plugin permanently.

My point was
 1. is there any API to cancel a goal by different process.
 2. should be considered that cancellation function for multi-robot control?

SteveMacenski, 2019-10-17:
https://github.com/ros-planning/navigation2/blob/master/nav2_rviz_plugins/src/nav2_panel.cpp#L259

Added

closed 2019-10-17 by SteveMacenski