ros-navigation/navigation2 #1078: [nav2_bringup] Error handling on bringup

SteveMacenski, 2019-08-28:
It took me a few minutes to debug something like:

```
[ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught exception when trying to load file of format [{}]:
```

Lovely little error. What ended up happening is I traced it back to 

```
    declare_world_cmd = launch.actions.DeclareLaunchArgument(
        'world',
        default_value=os.path.join(get_package_share_directory('turtlebot3_gazebo'),
                                   'worlds/turtlebot3_worlds/waffle.model'),
        description='Full path to world file to load')
```

And because I didnt have `turtlebot3_gazebo` in my path, it failed to find it. There should be some error thrown here that `turtlebot3_gazebo` is not found rather than just crashing

crdelsey, 2019-08-29:
That's funny. @orduno and I were just talk about that this afternoon.

nav2_bringup shouldn't have a dependency on turtlebot3 at all. These launch files with the turtlebot dependency shouldn't really be there. I'm just not sure where they should go. Maybe in tools since we don't really want to distribute them to customers. If someone is using nav2 with the turtlebot, they should be using the launch files supplied by Robotis.

On the more general problem of error handling in launch files, I agree it could be better. Some errors would need to be caught and reported within the launch tool because they aren't resolved until after the context of our launch file. This particular one - "can't find the package" - could be handled by us though I think

mkhansen-intel, 2019-08-29:
I mentioned in the PR that added this #987, that I don't think we should provide a default map file in this launch file, it should be a required parameter.

SteveMacenski, 2019-08-29:
I agree that it shouldn't have a dependency on TB3. I think having a `/test` directory in bringup with those turtlebot3 files would be fine and having the simulation launch file call those by default, but can be overrided by whatever. Renaming it `tb3_simulation` or the like would satisfy me as then its  working example for people to start with. 

I've been having issues today with setting run-time parameters like map name and having it given to the map-server. I ended up changing hard coded defaults in map server to get it running at all but that's an issue for another time.

mkhansen-intel, 2019-08-29:
I was specifically saying the "nav2_bringup_launch.py" shouldn't have a default. I agree that the "nav2_simulation_launch.py" can. The former is expected to be run on hardware, so the user is expected to provide a map. I agree we should rename the latter "nav2_tb3_simulation_launch.py" and move those files (map, world) into an examples folder or something.

There's a draft PR to refactor some of these launch files to enable multi-namespace launching (multi-robot). Depending on whether that's close to merging, we can do a PR for this one too.

mkhansen-intel, 2019-09-11:
linking to #1058 - I believe this issue will be resolved by that PR, need to verify when it's ready

orduno, 2019-09-11:
I completed all required tasks for #1058, the next step is to split the work into multiple PRs. The first will include refactoring of launch files.

mkhansen-intel, 2019-09-18:
I think this can be closed now, as #1118 has merged

closed 2019-09-18 by mkhansen-intel