ros-navigation/navigation2 #1079: System test failures occur when running 100s of times

mkhansen-intel, 2019-08-29:
## Bug report
This is not a single issue, but a meta-ticket for discussing the system test failures that are occurring.

For example, I ran the system test using the method in #1076 with three different RMW implementations and got these results:
FastRTPS - failed 4/100 times
OpenSplice - failed 17/100 times
CycloneDDS - failed 23/100 times

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Source (master branch as of 8/26/19)
- Version or commit hash:
  - 23628db71470d6416c25e8a5304f4f329350d805
- DDS implementation:
  - multiple (FastRTPS, CycloneDDS, OpenSplice)
- Client library (if applicable):
  - rclcpp (master branch as of 8/26/19) 
  - 17841d6b7cff96efef87400fee2dcd63a6f10101

#### Steps to reproduce issue
See PR #1076 

#### Expected behavior
Tests should pass 100/100 times

#### Actual behavior
FastRTPS - failed 4/100 times
OpenSplice - failed 17/100 times
CycloneDDS - failed 23/100 times

#### Additional information
I'll update the ticket with more info in the comments.

mkhansen-intel, 2019-08-29:
FastRTPS failure 1 - gazebo crashes - this is a failure we see randomly
```
10: [test_system_node-12] [INFO] [nav2_tester]: Waiting for a few seconds for all nodes to initialize
10: [gzserver-1] [ERROR] [rclcpp]: Couldn't add guard_condition to wait set: guard_conditions set is full, at /home/mkhansen/ros2_dev/ros2_ws/src/ros2/rcl/rcl/src/rcl/wait.c:453
10: [gzserver-1] terminate called after throwing an instance of 'std::runtime_error'
10: [gzserver-1]   what():  Couldn't fill wait set
10: [ERROR] [gzserver-1]: process has died [pid 14171, exit code -6, cmd 'gzserver -s libgazebo_ros_init.so --minimal_comms /home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/worlds/turtlebot3_ros2_demo.world'].

```

mkhansen-intel, 2019-08-29:
Failure 2 - this one is a dwb crash, no idea why, also occurred 2/100 times when running FastRTPS
```
10: [ERROR] [dwb_controller-7]: process has died [pid 6618, exit code -11, cmd '/home/mkhansen/ros2_dev/navigation2_ws/install/dwb_controller/lib/dwb_controller/dwb_controller --ros-args __params:=/home/mkhansen/ros2_dev/navigation2_ws/install/nav2_bringup/share/nav2_bringup/launch/nav2_params.yaml'].
```
We can probably add some exception handling to dwb to 'try catch' whatever's happening here and at least print out a message.

dirk-thomas, 2019-08-29:
I expect this thread to grow pretty quickly due to the large number of different error cases. Therefore I recommend we label each different case with a unique number, like `(N)`. I did this for your previous two cases.

dirk-thomas, 2019-08-29:
Also lets focus on FastRTS for now.

dirk-thomas, 2019-08-29:
For each failure case please include the full output of the test and not only a part.

dirk-thomas, 2019-08-29:
Failure case **(3)**:

`test_localization` timing out. Likely due to the following reported assertion:

> 10: [amcl-5] amcl: /home/dthomas/ros2/ws_nav/src/navigation2/nav2_util/src/pf/pf.c:379: pf_update_resample: Assertion `i < set_a->sample_count' failed.
> 10: [ERROR] [amcl-5]: process has died [pid 5065, exit code -6, cmd '/home/dthomas/ros2/ws_nav/install/nav2_amcl/lib/nav2_amcl/amcl --ros-args'].

<details>
<summary>Full output of the `test_localization` test</summary>

    Start 10: test_localization

10: Test command: /usr/bin/python3 "-u" "/home/dthomas/ros2/ws/install/ament_cmake_test/share/ament_cmake_test/cmake/run_test.py" "/home/dthomas/ros2/ws_nav/build/nav2_system_tests/test_results/nav2_system_tests/test_localization.xml" "--package-name" "nav2_system_tests" "--generate-result-on-success" "--env" "TEST_MAP=/home/dthomas/ros2/ws_nav/src/navigation2/nav2_system_tests/maps/map_circular.yaml" "TEST_EXECUTABLE=/home/dthomas/ros2/ws_nav/build/nav2_system_tests/src/localization/test_localization_node" "TEST_WORLD=/home/dthomas/ros2/ws_nav/src/navigation2/nav2_system_tests/worlds/turtlebot3_ros2_demo.world" "GAZEBO_MODEL_PATH=/home/dthomas/ros2/ws_nav/src/navigation2/nav2_system_tests/models" "--command" "/home/dthomas/ros2/ws_nav/src/navigation2/nav2_system_tests/src/localization/test_localization_launch.py"
10: Test timeout computed to be: 60
10: -- run_test.py: extra environment variables:
10:  - GAZEBO_MODEL_PATH=/home/dthomas/ros2/ws_nav/src/navigation2/nav2_system_tests/models
10:  - TEST_WORLD=/home/dthomas/ros2/ws_nav/src/navigation2/nav2_system_tests/worlds/turtlebot3_ros2_demo.world
10:  - TEST_MAP=/home/dthomas/ros2/ws_nav/src/navigation2/nav2_system_tests/maps/map_circular.yaml
10:  - TEST_EXECUTABLE=/home/dthomas/ros2/ws_nav/build/nav2_system_tests/src/localization/test_localization_node
10: -- run_test.py: invoking following command in '/home/dthomas/ros2/ws_nav/src/navigation2/nav2_system_tests':
10:  - /home/dthomas/ros2/ws_nav/src/navigation2/nav2_system_tests/src/localization/test_localization_launch.py
10: [INFO] [launch]: All log files can be found below /home/dthomas/.ros/log/2019-08-28-23-04-43-985902-drudge-5058
10: [INFO] [launch]: Default logging verbosity is set to INFO
10: [INFO] [gzserver-1]: process started with pid [5061]
10: [INFO] [static_transform_publisher-2]: process started with pid [5062]
10: [INFO] [static_transform_publisher-3]: process started with pid [5063]
10: [INFO] [map_server-4]: process started with pid [5064]
10: [INFO] [amcl-5]: process started with pid [5065]
10: [INFO] [lifecycle_manager-6]: process started with pid [5066]
10: [INFO] [test_localization_node-7]: process started with pid [5067]
10: [test_localization_node-7] Running main() from /home/dthomas/ros2/ws/install/gtest_vendor/src/gtest_vendor/src/gtest_main.cc
10: [test_localization_node-7] [==========] Running 1 test from 1 test case.
10: [test_localization_node-7] [----------] Global test environment set-up.
10: [test_localization_node-7] [----------] 1 test from TestAmclPose
10: [test_localization_node-7] [ RUN      ] TestAmclPose.SimpleAmclTest
10: [amcl-5] [INFO] [amcl]: Creating             
10: [amcl-5] [INFO] [amcl]: Configuring
10: [amcl-5] [INFO] [amcl]: initTransforms
10: [amcl-5] [INFO] [amcl]: initPubSub
10: [amcl-5] [INFO] [amcl]: Subscribed to map topic.
10: [amcl-5] [INFO] [amcl]: Activating
10: [amcl-5] [INFO] [amcl]: initialPoseReceived
10: [amcl-5] [INFO] [amcl]: Setting pose (1567058684.785032): -2.000 -0.500 0.000
10: [amcl-5] [INFO] [amcl]: initialPoseReceived
10: [amcl-5] [INFO] [amcl]: Setting pose (1567058684.790382): -2.000 -0.500 0.000
10: [amcl-5] [INFO] [amcl]: initialPoseReceived
10: [amcl-5] [INFO] [amcl]: Setting pose (1567058684.885893): -2.000 -0.500 0.000
10: [amcl-5] [INFO] [amcl]: initialPoseReceived
10: [amcl-5] [INFO] [amcl]: Setting pose (1567058684.986464): -2.000 -0.500 0.000
10: [amcl-5] [INFO] [amcl]: initialPoseReceived
10: [amcl-5] [INFO] [amcl]: Setting pose (1567058685.087255): -2.000 -0.500 0.000
10: [amcl-5] [INFO] [amcl]: initialPoseReceived
10: [amcl-5] [INFO] [amcl]: Setting pose (1567058685.188145): -2.000 -0.500 0.000
10: [amcl-5] [INFO] [amcl]: initialPoseReceived
10: [amcl-5] [INFO] [amcl]: Setting pose (1567058685.288855): -2.000 -0.500 0.000
10: [amcl-5] [INFO] [amcl]: initialPoseReceived
10: [amcl-5] [INFO] [amcl]: Setting pose (1567058685.390849): -2.000 -0.500 0.000
10: [amcl-5] [INFO] [amcl]: initialPoseReceived
10: [amcl-5] [INFO] [amcl]: Setting pose (1567058685.491553): -2.000 -0.500 0.000
10: [amcl-5] [INFO] [amcl]: initialPoseReceived
10: [amcl-5] [INFO] [amcl]: Setting pose (1567058685.592326): -2.000 -0.500 0.000
10: [amcl-5] [INFO] [amcl]: initialPoseReceived
10: [amcl-5] [INFO] [amcl]: Setting pose (1567058685.693184): -2.000 -0.500 0.000
10: [amcl-5] [INFO] [amcl]: initialPoseReceived
10: [amcl-5] [INFO] [amcl]: Setting pose (1567058685.793887): -2.000 -0.500 0.000
10: [amcl-5] [INFO] [amcl]: initialPoseReceived
10: [amcl-5] [INFO] [amcl]: Setting pose (1567058685.894549): -2.000 -0.500 0.000
10: [amcl-5] [INFO] [amcl]: initialPoseReceived
10: [amcl-5] [INFO] [amcl]: Setting pose (1567058685.995384): -2.000 -0.500 0.000
10: [amcl-5] [INFO] [amcl]: initialPoseReceived
[... 29 lines cut ...]
10: [amcl-5] [INFO] [amcl]: initialPoseReceived
10: [amcl-5] [INFO] [amcl]: Setting pose (1567058687.505624): -2.000 -0.500 0.000
10: [amcl-5] [INFO] [amcl]: initialPoseReceived
[... cut ...]
10: [amcl-5] amcl: /home/dthomas/ros2/ws_nav/src/navigation2/nav2_util/src/pf/pf.c:379: pf_update_resample: Assertion `i < set_a->sample_count' failed.
10: [ERROR] [amcl-5]: process has died [pid 5065, exit code -6, cmd '/home/dthomas/ros2/ws_nav/install/nav2_amcl/lib/nav2_amcl/amcl --ros-args'].
10: [amcl-5] [INFO] [amcl]: Settin
[Processing: nav2_system_tests]                   
[Processing: nav2_system_tests]                           
1/1 Test #10: test_localization ................***Timeout  60.10 sec

</details>

crdelsey, 2019-08-29:
I tested RTI Connext

RTI Connext - failed 45/100 times

dirk-thomas, 2019-08-29:
I experienced the failure **(1)** reported above and attached my full log to the comment: https://github.com/ros-planning/navigation2/issues/1079#issuecomment-526230684

dirk-thomas, 2019-08-29:
> Also lets focus on FastRTS for now.

@crdelsey Please keep this focused for now on FastRTPS. Otherwise this ticket will explode in length and become unmanagable.

mkhansen-intel, 2019-08-29:
Actually looking at the FastRTPS fails, this is the best it has ever looked. The 2 fails I saw yesterday I saw 2x each, and don't look like RMW/DDS issues at all. I believe 1.9.0 might have fixed some issues (at least I hope).

I will run another 100 runs this morning. If someone else can run another 100, that would be a good comparison.

crdelsey, 2019-08-29:
The last two times I ran FastRTPS (Tuesday and today) I got 8 and 4 failures out of 100, respectively.

dirk-thomas, 2019-08-29:
> The last two times I ran FastRTPS (Tuesday and today) I got 8 and 4 failures out of 100, respectively.

@crdelsey Please post specific failure cases with the necessary details in separate comments. These summary numbers aren't helpful in determining what causes them.

crdelsey, 2019-08-29:
Sorry. That data was requested in the thread above. I won't post those numbers again unless something is substantially different.

mkhansen-intel, 2019-08-29:
@dirk-thomas - I think the statistical numbers **are** relevant. The fact that Carl and I got similar numbers is good data to have, it means I probably didn't get 'lucky' and that overall FastRTPS has improved vs. Dashing where we were seeing ~85% pass rate. 

We do need the logs also, but please don't mark those comments as off-topic.

@crdelsey - can you triage the fails you saw? Do you still have the logs?

dirk-thomas, 2019-08-29:
> but please don't mark those comments as off-topic.

We can use a different category - maybe "resolved"?

The goal if this thread is to identify exact failures and resolve them. While the statistics are valuable they don't help with this goal. Therefore I would prefer to fold all comment not describing a specific failure - otherwise the length of this ticket will make it impossible to enumerate / iterate on the actual failures.

mkhansen-intel, 2019-08-29:
@dirk-thomas - BTW the failure (1) above is the same failure seen in the build farm:
http://build.ros2.org/view/Eci/job/Eci__nightly_navigation2_ubuntu_bionic_amd64/lastSuccessfulBuild/testReport/(root)/nav2_system_tests/test_bt_navigator_missing_result/

Thank you for debugging it, are you going to file a bug in rclcpp?

dirk-thomas, 2019-08-29:
The referenced PR should fix failure **(1)**.

dirk-thomas, 2019-08-29:
Failure case **(4)**:

`test_localization` timing out. Likely due to the following reported error:

> 10: [map_server-4] free(): invalid pointer
> 10: [ERROR] [map_server-4]: process has died [pid 29814, exit code -6, cmd '/home/dthomas/ros2/ws_nav/install/nav2_map_server/lib/nav2_map_server/map_server --ros-args __node:=map_server __params:=/tmp/launch_params_gczqwz34'].

<details>
<summary>Full output of the `test_localization` test</summary>

    Start 10: test_localization

10: Test command: /usr/bin/python3 "-u" "/home/dthomas/ros2/ws/install/ament_cmake_test/share/ament_cmake_test/cmake/run_test.py" "/home/dthomas/ros2/ws_nav/build/nav2_system_tests/test_results/nav2_system_tests/test_localization.xml" "--package-name" "nav2_system_tests" "--generate-result-on-success" "--env" "TEST_MAP=/home/dthomas/ros2/ws_nav/src/navigation2/nav2_system_tests/maps/map_circular.yaml" "TEST_EXECUTABLE=/home/dthomas/ros2/ws_nav/build/nav2_system_tests/src/localization/test_localization_node" "TEST_WORLD=/home/dthomas/ros2/ws_nav/src/navigation2/nav2_system_tests/worlds/turtlebot3_ros2_demo.world" "GAZEBO_MODEL_PATH=/home/dthomas/ros2/ws_nav/src/navigation2/nav2_system_tests/models" "--command" "/home/dthomas/ros2/ws_nav/src/navigation2/nav2_system_tests/src/localization/test_localization_launch.py"
10: Test timeout computed to be: 60
10: -- run_test.py: extra environment variables:
10:  - GAZEBO_MODEL_PATH=/home/dthomas/ros2/ws_nav/src/navigation2/nav2_system_tests/models
10:  - TEST_WORLD=/home/dthomas/ros2/ws_nav/src/navigation2/nav2_system_tests/worlds/turtlebot3_ros2_demo.world
10:  - TEST_EXECUTABLE=/home/dthomas/ros2/ws_nav/build/nav2_system_tests/src/localization/test_localization_node
10:  - TEST_MAP=/home/dthomas/ros2/ws_nav/src/navigation2/nav2_system_tests/maps/map_circular.yaml
10: -- run_test.py: invoking following command in '/home/dthomas/ros2/ws_nav/src/navigation2/nav2_system_tests':
10:  - /home/dthomas/ros2/ws_nav/src/navigation2/nav2_system_tests/src/localization/test_localization_launch.py
10: [INFO] [launch]: All log files can be found below /home/dthomas/.ros/log/2019-08-29-10-40-47-770303-drudge-29808
10: [INFO] [launch]: Default logging verbosity is set to INFO
10: [INFO] [gzserver-1]: process started with pid [29811]
10: [INFO] [static_transform_publisher-2]: process started with pid [29812]
10: [INFO] [static_transform_publisher-3]: process started with pid [29813]
10: [INFO] [map_server-4]: process started with pid [29814]
10: [INFO] [amcl-5]: process started with pid [29815]
10: [INFO] [lifecycle_manager-6]: process started with pid [29816]
10: [INFO] [test_localization_node-7]: process started with pid [29817]
10: [test_localization_node-7] Running main() from /home/dthomas/ros2/ws/install/gtest_vendor/src/gtest_vendor/src/gtest_main.cc
10: [test_localization_node-7] [==========] Running 1 test from 1 test case.
10: [test_localization_node-7] [----------] Global test environment set-up.
10: [test_localization_node-7] [----------] 1 test from TestAmclPose
10: [test_localization_node-7] [ RUN      ] TestAmclPose.SimpleAmclTest
10: [map_server-4] free(): invalid pointer      
10: [ERROR] [map_server-4]: process has died [pid 29814, exit code -6, cmd '/home/dthomas/ros2/ws_nav/install/nav2_map_server/lib/nav2_map_server/map_server --ros-args __node:=map_server __params:=/tmp/launch_params_gczqwz34'].
10: [ERROR] [amcl-5]: process has died [pid 29815, exit code -11, cmd '/home/dthomas/ros2/ws_nav/install/nav2_amcl/lib/nav2_amcl/amcl --ros-args'].
[Processing: nav2_system_tests]                   
[Processing: nav2_system_tests]                           
1/1 Test #10: test_localization ................***Timeout  60.04 sec

</details>

mkhansen-intel, 2019-08-29:
Running another 100x with FastRTPS, same build, same system, this time 12/100 failed. The first fail I see is this: Failure case (4)
```
10: [amcl-6] [WARN] [amcl]: Waiting for map....
10: [amcl-6] double free or corruption (fasttop)
10: [ERROR] [amcl-6]: process has died [pid 4325, exit code -6, cmd '/home/mkhansen/ros2_dev/navigation2_ws/install/nav2_amcl/lib/nav2_amcl/amcl --ros-args __node:=amcl __params:=/home/mkhansen/ros2_dev/navigation2_ws/install/nav2_bringup/share/nav2_bringup/launch/nav2_params.yaml'].
```
This may be a bug in amcl, not sure as I've not seen this one before.
<details>
<summary>Full output of the test</summary>
10: Test command: /usr/bin/python3 "-u" "/home/mkhansen/ros2_dev/ros2_ws/install/ament_cmake_test/share/ament_cmake_test/cmake/run_test.py" "/home/mkhansen/ros2_dev/navigation2_ws/build/nav2_system_tests/test_results/nav2_system_tests/test_bt_navigator.xml" "--package-name" "nav2_system_tests" "--generate-result-on-success" "--env" "TEST_DIR=/home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/src/system" "TEST_MAP=/home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/maps/map_circular.yaml" "TEST_WORLD=/home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/worlds/turtlebot3_ros2_demo.world" "GAZEBO_MODEL_PATH=/home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/models" "BT_NAVIGATOR_XML=navigate_w_replanning_and_recovery.xml" "ASTAR=True" "--command" "/home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/src/system/test_system_launch.py"

10: Test timeout computed to be: 120
10: -- run_test.py: extra environment variables:
10:  - ASTAR=True
10:  - TEST_WORLD=/home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/worlds/turtlebot3_ros2_demo.world
10:  - GAZEBO_MODEL_PATH=/home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/models
10:  - TEST_MAP=/home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/maps/map_circular.yaml
10:  - TEST_DIR=/home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/src/system
10:  - BT_NAVIGATOR_XML=navigate_w_replanning_and_recovery.xml
10: -- run_test.py: invoking following command in '/home/mkhansen/ros2_dev/navigation2_ws/build/nav2_system_tests/src/system':
10:  - /home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/src/system/test_system_launch.py
10: [INFO] [launch]: All log files can be found below /home/mkhansen/.ros/log/2019-08-29-10-45-14-626430-mkhansen-desk-4317
10: [INFO] [launch]: Default logging verbosity is set to INFO
10: [INFO] [gzserver-1]: process started with pid [4320]
10: [INFO] [static_transform_publisher-2]: process started with pid [4321]
10: [INFO] [static_transform_publisher-3]: process started with pid [4322]
10: [INFO] [map_server-4]: process started with pid [4323]
10: [INFO] [world_model-5]: process started with pid [4324]
10: [INFO] [amcl-6]: process started with pid [4325]
10: [INFO] [dwb_controller-7]: process started with pid [4326]
10: [INFO] [navfn_planner-8]: process started with pid [4327]
10: [INFO] [recoveries_node-9]: process started with pid [4328]
10: [INFO] [bt_navigator-10]: process started with pid [4341]
10: [INFO] [lifecycle_manager-11]: process started with pid [4350]
10: [INFO] [test_system_node-12]: process started with pid [4373]
10: [static_transform_publisher-2] [INFO] []: LOOPING due to no latching at the moment
10: [static_transform_publisher-3] [INFO] []: LOOPING due to no latching at the moment
10: [map_server-4] [INFO] [map_server]: Creating
10: [world_model-5] [INFO] [world_model]: Creating World Model
10: [amcl-6] [INFO] [amcl]: Creating
10: [dwb_controller-7] [INFO] [dwb_controller]: Creating
10: [navfn_planner-8] [INFO] [navfn_planner]: Creating
10: [world_model-5] [INFO] [global_costmap.global_costmap]: Creating Costmap
10: [dwb_controller-7] [INFO] [local_costmap.local_costmap]: Creating Costmap
10: [lifecycle_manager-11] [INFO] [lifecycle_manager]: Creating
10: [lifecycle_manager-11] [INFO] [lifecycle_manager]: [34m[1mStarting the system bringup...[0m[0m
10: [lifecycle_manager-11] [INFO] [lifecycle_manager]: [34m[1mCreating and initializing lifecycle service clients[0m[0m
10: [lifecycle_manager-11] [INFO] [lifecycle_manager]: [34m[1mConfiguring and activating map_server[0m[0m
10: [recoveries_node-9] [INFO] [recoveries]: Configuring Spin
10: [recoveries_node-9] [INFO] [recoveries]: Configuring BackUp
10: [bt_navigator-10] [INFO] [bt_navigator]: Creating
10: [map_server-4] [INFO] [map_server]: Configuring
10: [map_server-4] [INFO] [map_server]: OccGridLoader: Creating
10: [map_server-4] [INFO] [map_server]: OccGridLoader: Configuring
10: [map_server-4] [INFO] [map_server]: Activating
10: [map_server-4] [INFO] [map_server]: OccGridLoader: Activating
10: [lifecycle_manager-11] [INFO] [lifecycle_manager]: [34m[1mConfiguring and activating amcl[0m[0m
10: [amcl-6] [INFO] [amcl]: Configuring
10: [amcl-6] [INFO] [amcl]: initTransforms
10: [amcl-6] [INFO] [amcl]: initPubSub
10: [amcl-6] [INFO] [amcl]: Subscribed to map topic.
10: [amcl-6] [INFO] [amcl]: Activating
10: [lifecycle_manager-11] [INFO] [lifecycle_manager]: [34m[1mConfiguring and activating world_model[0m[0m
10: [world_model-5] [INFO] [world_model]: Configuring
10: [world_model-5] [INFO] [global_costmap.global_costmap]: Configuring
[... cut ...]
10: [world_model-5] [INFO] [global_costmap_rclcpp_node]: [signalFailure] Drop message: frame 'base_scan' at time 107.400 for reason(0)
10: [world_model-5] [INFO] [global_costmap_rclcpp_node]: [signalFailure] Drop message: frame 'base_scan' at time 107.600 for reason(0)
[HANDLER_OUTPUT] 
1/
1 Test #10: test_bt_navigator ................***Timeout 120.11 sec

0% tests passed, 1 tests failed out of 1

Total Test time (real) = 120.11 sec

The following tests FAILED:
	 10 - test_bt_navigator (Timeout)
[ERROR_MESSAGE] 
Errors while running CTest

</details>

crdelsey, 2019-08-29:
> [amcl-6] double free or corruption (fasttop)

I've seen that one for a while, but only as amcl was shutting down. I don't think I've ever seen it cause the shutdown.

mkhansen-intel, 2019-08-29:
> > [amcl-6] double free or corruption (fasttop)
> 
> I've seen that one for a while, but only as amcl was shutting down. I don't think I've ever seen it cause the shutdown.

It may only be a symptom, not the cause of the failure. Looking at the amcl code, I can see where it could try to free memory that's already been freed here:
https://github.com/ros-planning/navigation2/blob/97de88d91dd90141486f34a4b45258f03b2874e2/nav2_amcl/src/amcl_node.cpp#L286

We can either check for a null before calling map_free(map_) and pf_free(pf_) or check inside those functions before calling free() itself. The latter is probably safer. I'll file a new issue for this.

dirk-thomas, 2019-08-29:
Please use the latest state when testing. ros2/rclcpp#837 has been merged which resolves the failure **(1)**.

dirk-thomas, 2019-08-29:
Failure case **(5)**

`test_planner` timing out. Likely due to the following reported errors:

> 9: [navfn_planner-1] [ERROR] [getCurrentPose]: No Transform available Error looking up robot pose: "map" passed to lookupTransform argument target_frame does not exist. 
> 9: [test_planner_node-3] [ERROR] [PlannerTester]: Failed to get a plan within the allowed time

<details>
<summary>Full output of the `test_planner` test</summary>

    Start 9: test_planner

9: Test command: /usr/bin/python3 "-u" "/home/dthomas/ros2/ws/install/ament_cmake_test/share/ament_cmake_test/cmake/run_test.py" "/home/dthomas/ros2/ws_nav/build/nav2_system_tests/test_results/nav2_system_tests/test_planner.xml" "--package-name" "nav2_system_tests" "--generate-result-on-success" "--env" "TEST_LAUNCH_DIR=" "TEST_EXECUTABLE=/home/dthomas/ros2/ws_nav/build/nav2_system_tests/src/planning/test_planner_node" "TEST_MAP=/home/dthomas/ros2/ws_nav/src/navigation2/nav2_system_tests/maps/map.pgm" "--command" "/home/dthomas/ros2/ws_nav/src/navigation2/nav2_system_tests/src/planning/test_planner_launch.py"
9: Test timeout computed to be: 60
9: -- run_test.py: extra environment variables:   
9:  - TEST_LAUNCH_DIR=
9:  - TEST_MAP=/home/dthomas/ros2/ws_nav/src/navigation2/nav2_system_tests/maps/map.pgm
9:  - TEST_EXECUTABLE=/home/dthomas/ros2/ws_nav/build/nav2_system_tests/src/planning/test_planner_node
9: -- run_test.py: invoking following command in '/home/dthomas/ros2/ws_nav/build/nav2_system_tests/src/planning':
9:  - /home/dthomas/ros2/ws_nav/src/navigation2/nav2_system_tests/src/planning/test_planner_launch.py
[Processing: nav2_system_tests]                   
9: [INFO] [launch]: All log files can be found below /home/dthomas/.ros/log/2019-08-29-15-06-07-419372-drudge-20025
9: [INFO] [launch]: Default logging verbosity is set to INFO
9: [INFO] [navfn_planner-1]: process started with pid [20028]
9: [INFO] [lifecycle_manager-2]: process started with pid [20029]
9: [INFO] [test_planner_node-3]: process started with pid [20030]
9: [test_planner_node-3] Running main() from /home/dthomas/ros2/ws/install/gtest_vendor/src/gtest_vendor/src/gtest_main.cc
9: [test_planner_node-3] [==========] Running 2 tests from 1 test case.
9: [test_planner_node-3] [----------] Global test environment set-up.
9: [test_planner_node-3] [----------] 2 tests from PlannerTester
9: [test_planner_node-3] [ RUN      ] PlannerTester.testSimpleCostmaps
9: [navfn_planner-1] [ERROR] [getCurrentPose]: No Transform available Error looking up robot pose: "map" passed to lookupTransform argument target_frame does not exist. 
9: [navfn_planner-1] 
9: [test_planner_node-3] [ERROR] [PlannerTester]: Failed to get a plan within the allowed time
9: [test_planner_node-3] /home/dthomas/ros2/ws_nav/src/navigation2/nav2_system_tests/src/planning/test_planner_node.cpp:55: Failure
9: [test_planner_node-3] Expected equality of these values:
9: [test_planner_node-3]   true
9: [test_planner_node-3]   defaultPlannerTest(result)
9: [test_planner_node-3]     Which is: false
9: [test_planner_node-3] [ERROR] [PlannerTester]: Failed to get a plan within the allowed time
9: [test_planner_node-3] /home/dthomas/ros2/ws_nav/src/navigation2/nav2_system_tests/src/planning/test_planner_node.cpp:55: Failure
9: [test_planner_node-3] Expected equality of these values:
9: [test_planner_node-3]   true
9: [test_planner_node-3]   defaultPlannerTest(result)
9: [test_planner_node-3]     Which is: false
[Processing: nav2_system_tests]                           
9: [test_planner_node-3] [ERROR] [PlannerTester]: Failed to get a plan within the allowed time
9: [test_planner_node-3] /home/dthomas/ros2/ws_nav/src/navigation2/nav2_system_tests/src/planning/test_planner_node.cpp:55: Failure
9: [test_planner_node-3] Expected equality of these values:
9: [test_planner_node-3]   true
9: [test_planner_node-3]   defaultPlannerTest(result)
9: [test_planner_node-3]     Which is: false
9: [test_planner_node-3] [ERROR] [PlannerTester]: Failed to get a plan within the allowed time
9: [test_planner_node-3] /home/dthomas/ros2/ws_nav/src/navigation2/nav2_system_tests/src/planning/test_planner_node.cpp:55: Failure
9: [test_planner_node-3] Expected equality of these values: 
9: [test_planner_node-3]   true
9: [test_planner_node-3]   defaultPlannerTest(result)
9: [test_planner_node-3]     Which is: false
9: [test_planner_node-3] [ERROR] [PlannerTester]: Failed to get a plan within the allowed time
9: [test_planner_node-3] /home/dthomas/ros2/ws_nav/src/navigation2/nav2_system_tests/src/planning/test_planner_node.cpp:55: Failure
9: [test_planner_node-3] Expected equality of these values:
9: [test_planner_node-3]   true
9: [test_planner_node-3]   defaultPlannerTest(result)
9: [test_planner_node-3]     Which is: false
    Test #9: test_planner .....................***Timeout  60.04 sec

</details>

dirk-thomas, 2019-08-29:
This didn't actually fail the `test_bt_navigator` test for me but I wanted to post the error message anyway since it seems to indicate a problem:

> 11: [bt_navigator-10] [ERROR] [bt_navigator]: Navigation failed
> 11: [bt_navigator-10] terminate called after throwing an instance of 'std::runtime_error'
> 11: [bt_navigator-10]   what():  Asked to publish result for goal that does not exist
> 11: [ERROR] [bt_navigator-10]: process has died [pid 24442, exit code -6, cmd '/home/dthomas/ros2/ws_nav/install/nav2_bt_navigator/lib/nav2_bt_navigator/bt_navigator --ros-args __node:=bt_navigator __params:=/tmp/launch_params_1587neky __params:=/tmp/launch_params_u9c5u96v'].

dirk-thomas, 2019-08-29:
Failure case **(6)**:

`test_bt_navigator_with_dijkstra` timing out. Likely due to the following reported error:

> 12: [gzserver-1] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
> 12: [gzserver-1]   what():  failed to publish message: cannot publish data, at /home/dthomas/ros2/github/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/src/rmw_publish.cpp:52, at /home/dthomas/ros2/github/ros2/rcl/rcl/src/rcl/publisher.c:257
> 12: [ERROR] [gzserver-1]: process has died [pid 25828, exit code -6, cmd 'gzserver -s libgazebo_ros_init.so --minimal_comms /home/dthomas/ros2/ws_nav/src/navigation2/nav2_system_tests/worlds/turtlebot3_ros2_demo.world'].

<details>
<summary>Full output of the `test_bt_navigator_with_dijkstra` test</summary>

    Start 12: test_bt_navigator_with_dijkstra

12: Test command: /usr/bin/python3 "-u" "/home/dthomas/ros2/ws/install/ament_cmake_test/share/ament_cmake_test/cmake/run_test.py" "/home/dthomas/ros2/ws_nav/build/nav2_system_tests/test_results/nav2_system_tests/test_bt_navigator_with_dijkstra.xml" "--package-name" "nav2_system_tests" "--generate-result-on-success" "--env" "TEST_DIR=/home/dthomas/ros2/ws_nav/src/navigation2/nav2_system_tests/src/system" "TEST_MAP=/home/dthomas/ros2/ws_nav/src/navigation2/nav2_system_tests/maps/map_circular.yaml" "TEST_WORLD=/home/dthomas/ros2/ws_nav/src/navigation2/nav2_system_tests/worlds/turtlebot3_ros2_demo.world" "GAZEBO_MODEL_PATH=/home/dthomas/ros2/ws_nav/src/navigation2/nav2_system_tests/models" "BT_NAVIGATOR_XML=navigate_w_replanning_and_recovery.xml" "ASTAR=False" "--command" "/home/dthomas/ros2/ws_nav/src/navigation2/nav2_system_tests/src/system/test_system_launch.py"
12: Test timeout computed to be: 120
12: -- run_test.py: extra environment variables:
12:  - GAZEBO_MODEL_PATH=/home/dthomas/ros2/ws_nav/src/navigation2/nav2_system_tests/models
12:  - TEST_MAP=/home/dthomas/ros2/ws_nav/src/navigation2/nav2_system_tests/maps/map_circular.yaml
12:  - TEST_WORLD=/home/dthomas/ros2/ws_nav/src/navigation2/nav2_system_tests/worlds/turtlebot3_ros2_demo.world
12:  - TEST_DIR=/home/dthomas/ros2/ws_nav/src/navigation2/nav2_system_tests/src/system
12:  - ASTAR=False
12:  - BT_NAVIGATOR_XML=navigate_w_replanning_and_recovery.xml
12: -- run_test.py: invoking following command in '/home/dthomas/ros2/ws_nav/build/nav2_system_tests/src/system':
12:  - /home/dthomas/ros2/ws_nav/src/navigation2/nav2_system_tests/src/system/test_system_launch.py
12: [INFO] [launch]: All log files can be found below /home/dthomas/.ros/log/2019-08-29-15-34-40-634583-drudge-25825
12: [INFO] [launch]: Default logging verbosity is set to INFO
12: [INFO] [gzserver-1]: process started with pid [25828]
12: [INFO] [static_transform_publisher-2]: process started with pid [25829]
12: [INFO] [static_transform_publisher-3]: process started with pid [25830]
12: [INFO] [map_server-4]: process started with pid [25831]
12: [INFO] [world_model-5]: process started with pid [25832]
12: [INFO] [amcl-6]: process started with pid [25833]
12: [INFO] [dwb_controller-7]: process started with pid [25834]
12: [INFO] [navfn_planner-8]: process started with pid [25835]
12: [INFO] [recoveries_node-9]: process started with pid [25836]
12: [INFO] [bt_navigator-10]: process started with pid [25848]
12: [INFO] [lifecycle_manager-11]: process started with pid [25851]
12: [INFO] [test_system_node-12]: process started with pid [25855]
12: [amcl-6] [WARN] [amcl]: Waiting for map....   
12: [amcl-6] [WARN] [amcl]: Waiting for map....   
12: [amcl-6] [WARN] [amcl]: Waiting for map....   
12: [amcl-6] [WARN] [amcl]: Waiting for map....   
12: [world_model-5] [INFO] [world_model]: Creating World Model
12: [world_model-5] [INFO] [global_costmap.global_costmap]: Creating Costmap
12: [world_model-5] [INFO] [world_model]: Configuring
12: [world_model-5] [INFO] [global_costmap.global_costmap]: Configuring
12: [world_model-5] [INFO] [global_costmap.global_costmap]: Using plugin "static_layer"
12: [world_model-5] [INFO] [global_costmap.global_costmap]: Using plugin "obstacle_layer"
12: [world_model-5] [INFO] [global_costmap.global_costmap]: Subscribed to Topics: scan
12: [world_model-5] [INFO] [global_costmap.global_costmap]: StaticLayer: Resizing costmap to 384 X 384 at 0.050000 m/pix
12: [world_model-5] [INFO] [global_costmap.global_costmap]: Using plugin "inflation_layer"
12: [world_model-5] [INFO] [world_model]: Activating
12: [world_model-5] [INFO] [global_costmap.global_costmap]: Activating
12: [world_model-5] [INFO] [global_costmap.global_costmap]: Checking transform
12: [world_model-5] [INFO] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: . canTransform returned after 1.00828 timeout was 1.
12: [world_model-5] [INFO] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: . canTransform returned after 1.00093 timeout was 1.
12: [world_model-5] [INFO] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: . canTransform returned after 1.00837 timeout was 1.
12: [world_model-5] [INFO] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: . canTransform returned after 1.00119 timeout was 1.
[... cut ...]
12: [INFO] [navfn_planner-8]: process has finished cleanly [pid 25835]
12: -- run_test.py: return code 1                           
12: -- run_test.py: generate result file '/home/dthomas/ros2/ws_nav/build/nav2_system_tests/test_results/nav2_system_tests/test_bt_navigator_with_dijkstra.xml' with failed test
12: -- run_test.py: verify result file '/home/dthomas/ros2/ws_nav/build/nav2_system_tests/test_results/nav2_system_tests/test_bt_navigator_with_dijkstra.xml'
    Test #12: test_bt_navigator_with_dijkstra ...***Failed   94.45 sec

</details>

mkhansen-intel, 2019-08-30:
Ignore (5) and any fails within test_planner for now. PR #1083 is fixing that test. 

I'm **only** running the bt_navigator test for my runs of 100.

mkhansen-intel, 2019-08-31:
Fail (7) - this is new to me. Map server failing to load the map:
10: [map_server-4] [INFO] [map_server]: OccGridLoader: Configuring
10: [map_server-4] [ERROR] [map_server]: Failed to load image file /home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/maps/map_circular.yaml/map_circular.pgm for reason: Magick: Unable to open file (/home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/maps/map_circular.yaml/map_circular.pgm) reported by magick/blob.c:3002 (OpenBlob)
<details>
10: [map_server-4] [ERROR] []: Caught exception in callback for transition 10
10: [map_server-4] [ERROR] []: Original error: Failed to load map image file.
10: [map_server-4] [WARN] []: Error occurred while doing error handling.
10: [map_server-4] [FATAL] [map_server]: Lifecycle node entered error state
10: [lifecycle_manager-6] [ERROR] [lifecycle_manager]: Failed to change state for node: map_server
10: [lifecycle_manager-6] [ERROR] [lifecycle_manager]: Failed to bring up node: map_server, aborting bringup

10: Test command: /usr/bin/python3 "-u" "/home/mkhansen/ros2_dev/ros2_ws/install/ament_cmake_test/share/ament_cmake_test/cmake/run_test.py" "/home/mkhansen/ros2_dev/navigation2_ws/build/nav2_system_tests/test_results/nav2_system_tests/test_localization.xml" "--package-name" "nav2_system_tests" "--generate-result-on-success" "--env" "TEST_MAP=/home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/maps/map_circular.yaml" "TEST_EXECUTABLE=/home/mkhansen/ros2_dev/navigation2_ws/build/nav2_system_tests/src/localization/test_localization_node" "TEST_WORLD=/home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/worlds/turtlebot3_ros2_demo.world" "GAZEBO_MODEL_PATH=/home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/models" "--command" "/home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/src/localization/test_localization_launch.py"
10: Test timeout computed to be: 60
10: -- run_test.py: extra environment variables:            
10:  - TEST_WORLD=/home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/worlds/turtlebot3_ros2_demo.world
10:  - TEST_MAP=/home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/maps/map_circular.yaml
10:  - TEST_EXECUTABLE=/home/mkhansen/ros2_dev/navigation2_ws/build/nav2_system_tests/src/localization/test_localization_node
10:  - GAZEBO_MODEL_PATH=/home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/models
10: -- run_test.py: invoking following command in '/home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests':
10:  - /home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/src/localization/test_localization_launch.py
10: [INFO] [launch]: All log files can be found below /home/mkhansen/.ros/log/2019-08-30-16-55-19-205802-mkhansen-desk-29879
10: [INFO] [launch]: Default logging verbosity is set to INFO
10: [INFO] [gzserver-1]: process started with pid [29883]
10: [INFO] [static_transform_publisher-2]: process started with pid [29884]
10: [INFO] [static_transform_publisher-3]: process started with pid [29885]
10: [INFO] [map_server-4]: process started with pid [29886]
10: [INFO] [amcl-5]: process started with pid [29887]
10: [INFO] [lifecycle_manager-6]: process started with pid [29888]
10: [INFO] [test_localization_node-7]: process started with pid [29889]
10: [static_transform_publisher-3] [INFO] []: LOOPING due to no latching at the moment
10: [static_transform_publisher-2] [INFO] []: LOOPING due to no latching at the moment
10: [test_localization_node-7] Running main() from /home/mkhansen/ros2_dev/ros2_ws/install/gtest_vendor/src/gtest_vendor/src/gtest_main.cc
10: [test_localization_node-7] [==========] Running 1 test from 1 test case.
10: [test_localization_node-7] [----------] Global test environment set-up.
10: [test_localization_node-7] [----------] 1 test from TestAmclPose
10: [test_localization_node-7] [ RUN      ] TestAmclPose.SimpleAmclTest
10: [lifecycle_manager-6] [INFO] [lifecycle_manager]: Creating
10: [lifecycle_manager-6] [INFO] [lifecycle_manager]: Starting the system bringup...
10: [lifecycle_manager-6] [INFO] [lifecycle_manager]: Creating and initializing lifecycle service clients
10: [lifecycle_manager-6] [INFO] [lifecycle_manager]: Configuring and activating map_server
10: [amcl-5] [INFO] [amcl]: Creating
10: [map_server-4] [INFO] [map_server]: Creating
10: [map_server-4] [INFO] [map_server]: Configuring
10: [map_server-4] [INFO] [map_server]: OccGridLoader: Creating
10: [map_server-4] [INFO] [map_server]: OccGridLoader: Configuring
10: [map_server-4] [ERROR] [map_server]: Failed to load image file /home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/maps/map_circular.yaml/map_circular.pgm for reason: Magick: Unable to open file (/home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/maps/map_circular.yaml/map_circular.pgm) reported by magick/blob.c:3002 (OpenBlob)
10: [map_server-4] [ERROR] []: Caught exception in callback for transition 10
10: [map_server-4] [ERROR] []: Original error: Failed to load map image file.
10: [map_server-4] [WARN] []: Error occurred while doing error handling.
10: [map_server-4] [FATAL] [map_server]: Lifecycle node entered error state
10: [lifecycle_manager-6] [ERROR] [lifecycle_manager]: Failed to change state for node: map_server
[... cut ...]
10: [gzserver-1] [INFO] [turtlebot3_joint_state]: Going to publish joint [wheel_left_joint]
10: [gzserver-1] [INFO] [turtlebot3_joint_state]: Going to publish joint [wheel_right_joint]
[Processing: nav2_system_tests]                             
10: [static_transform_publisher-3] [INFO] []: LOOPING due to no latching at the moment
10: [static_transform_publisher-2] [INFO] []: LOOPING due to no latching at the moment
[Processing: nav2_system_tests]                             
    Test #10: test_localization ................***Timeout  60.07 sec

mkhansen-intel, 2019-09-03:
Fail case **(8)**
```
11: [lifecycle_manager-11] [ERROR] [lifecycle_manager]: Failed to change state for node: world_model
11: [lifecycle_manager-11] [ERROR] [lifecycle_manager]: Failed to bring up node: world_model, aborting bringup
11: [lifecycle_manager-11] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
11: [lifecycle_manager-11]   what():  Failed to create interrupt guard condition in Executor constructor: the given context is not valid, either rcl_init() was not called or rcl_shutdown() was called., at /home/mkhansen/ros2_dev/ros2_ws/src/ros2/rcl/rcl/src/rcl/guard_condition.c:69
```

<details>

11: Test command: /usr/bin/python3 "-u" "/home/mkhansen/ros2_dev/ros2_ws/install/ament_cmake_test/share/ament_cmake_test/cmake/run_test.py" "/home/mkhansen/ros2_dev/navigation2_ws/build/nav2_system_tests/test_results/nav2_system_tests/test_bt_navigator.xml" "--package-name" "nav2_system_tests" "--generate-result-on-success" "--env" "TEST_DIR=/home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/src/system" "TEST_MAP=/home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/maps/map_circular.yaml" "TEST_WORLD=/home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/worlds/turtlebot3_ros2_demo.world" "GAZEBO_MODEL_PATH=/home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/models" "BT_NAVIGATOR_XML=navigate_w_replanning_and_recovery.xml" "ASTAR=True" "--command" "/home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/src/system/test_system_launch.py"
11: Test timeout computed to be: 120
11: -- run_test.py: extra environment variables:  
11:  - TEST_WORLD=/home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/worlds/turtlebot3_ros2_demo.world
11:  - ASTAR=True
11:  - GAZEBO_MODEL_PATH=/home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/models
11:  - TEST_MAP=/home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/maps/map_circular.yaml
11:  - BT_NAVIGATOR_XML=navigate_w_replanning_and_recovery.xml
11:  - TEST_DIR=/home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/src/system
11: -- run_test.py: invoking following command in '/home/mkhansen/ros2_dev/navigation2_ws/build/nav2_system_tests/src/system':
11:  - /home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/src/system/test_system_launch.py
11: [INFO] [launch]: All log files can be found below /home/mkhansen/.ros/log/2019-09-03-11-09-37-629342-mkhansen-desk-29462
11: [INFO] [launch]: Default logging verbosity is set to INFO
11: [INFO] [gzserver-1]: process started with pid [29466]
11: [INFO] [static_transform_publisher-2]: process started with pid [29467]
11: [INFO] [static_transform_publisher-3]: process started with pid [29468]
11: [INFO] [map_server-4]: process started with pid [29469]
11: [INFO] [world_model-5]: process started with pid [29470]
11: [INFO] [gdb-6]: process started with pid [29471]
11: [INFO] [dwb_controller-7]: process started with pid [29472]
11: [INFO] [navfn_planner-8]: process started with pid [29473]
11: [INFO] [recoveries_node-9]: process started with pid [29474]
11: [INFO] [bt_navigator-10]: process started with pid [29491]
11: [INFO] [lifecycle_manager-11]: process started with pid [29512]
11: [INFO] [test_system_node-12]: process started with pid [29529]
11: [static_transform_publisher-2] [INFO] []: LOOPING due to no latching at the moment
11: [static_transform_publisher-3] [INFO] []: LOOPING due to no latching at the moment
11: [map_server-4] [INFO] [map_server]: Creating
11: [world_model-5] [INFO] [world_model]: Creating World Model
11: [dwb_controller-7] [INFO] [dwb_controller]: Creating
11: [dwb_controller-7] [INFO] [local_costmap.local_costmap]: Creating Costmap
11: [navfn_planner-8] [INFO] [navfn_planner]: Creating
11: [world_model-5] [INFO] [global_costmap.global_costmap]: Creating Costmap
11: [recoveries_node-9] [INFO] [recoveries]: Configuring Spin
11: [lifecycle_manager-11] [INFO] [lifecycle_manager]: Creating
11: [lifecycle_manager-11] [INFO] [lifecycle_manager]: Starting the system bringup...
11: [lifecycle_manager-11] [INFO] [lifecycle_manager]: Creating and initializing lifecycle service clients
11: [bt_navigator-10] [INFO] [bt_navigator]: Creating
11: [lifecycle_manager-11] [INFO] [lifecycle_manager]: Configuring and activating map_server
11: [recoveries_node-9] [INFO] [recoveries]: Configuring BackUp
11: [map_server-4] [INFO] [map_server]: Configuring
11: [map_server-4] [INFO] [map_server]: OccGridLoader: Creating
11: [map_server-4] [INFO] [map_server]: OccGridLoader: Configuring
11: [map_server-4] [INFO] [map_server]: Activating
11: [map_server-4] [INFO] [map_server]: OccGridLoader: Activating
11: [lifecycle_manager-11] [INFO] [lifecycle_manager]: Configuring and activating amcl
11: [gdb-6] GNU gdb (Ubuntu 8.1-0ubuntu3) 8.1.0.20180409-git
11: [gdb-6] Copyright (C) 2018 Free Software Foundation, Inc.
11: [gdb-6] License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
11: [gdb-6] This is free software: you are free to change and redistribute it.
11: [gdb-6] There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
11: [gdb-6] and "show warranty" for details.
11: [gdb-6] This GDB was configured as "x86_64-linux-gnu".
11: [gdb-6] Type "show configuration" for configuration details.
11: [gdb-6] For bug reporting instructions, please see:
[... cut ...]
11: [INFO] [gdb-6]: process has finished cleanly [pid 29471]
11: [gdb-6] (gdb) 
11: -- run_test.py: return code 1
11: -- run_test.py: generate result file '/home/mkhansen/ros2_dev/navigation2_ws/build/nav2_system_tests/test_results/nav2_system_tests/test_bt_navigator.xml' with failed test
11: -- run_test.py: verify result file '/home/mkhansen/ros2_dev/navigation2_ws/build/nav2_system_tests/test_results/nav2_system_tests/test_bt_navigator.xml'
    Test #11: test_bt_navigator ................***Failed   26.09 sec

0% tests passed, 1 tests failed out of 1

mkhansen-intel, 2019-09-03:
Fail case **(9)** - dwb_controller crashes. We see this about 1% of the time.
```
11: [test_system_node-12] [INFO] [nav2_tester]: Distance from goal is: 3.2015621187164243
11: [gdb-7] [INFO] [dwb_controller]: Received a goal, begin following path
11: [gdb-7] [New Thread 0x7fffbd542700 (LWP 5935)]
11: [gdb-7] 
11: [gdb-7] Thread 15 "dwb_controller" received signal SIGSEGV, Segmentation fault.
11: [gdb-7] [Switching to Thread 0x7fffbd542700 (LWP 5935)]
11: [gdb-7] 0x00007fffbe16afd3 in costmap_queue::CostmapQueue::enqueueCell(unsigned int, unsigned int, unsigned int, unsigned int, unsigned int) ()
11: [gdb-7]    from /home/mkhansen/ros2_dev/navigation2_ws/install/dwb_critics/lib/libdwb_critics.so
11: [test_system_node-12] [INFO] [nav2_tester]: Distance from goal is: 3.2015621187164243
```
I'll file a bug on this. 
<details>
11: Test command: /usr/bin/python3 "-u" "/home/mkhansen/ros2_dev/ros2_ws/install/ament_cmake_test/share/ament_cmake_test/cmake/run_test.py" "/home/mkhansen/ros2_dev/navigation2_ws/build/nav2_system_tests/test_results/nav2_system_tests/test_bt_navigator.xml" "--package-name" "nav2_system_tests" "--generate-result-on-success" "--env" "TEST_DIR=/home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/src/system" "TEST_MAP=/home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/maps/map_circular.yaml" "TEST_WORLD=/home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/worlds/turtlebot3_ros2_demo.world" "GAZEBO_MODEL_PATH=/home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/models" "BT_NAVIGATOR_XML=navigate_w_replanning_and_recovery.xml" "ASTAR=True" "--command" "/home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/src/system/test_system_launch.py"
11: Test timeout computed to be: 120
11: -- run_test.py: extra environment variables:
11:  - ASTAR=True
11:  - TEST_WORLD=/home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/worlds/turtlebot3_ros2_demo.world
11:  - BT_NAVIGATOR_XML=navigate_w_replanning_and_recovery.xml
11:  - GAZEBO_MODEL_PATH=/home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/models
11:  - TEST_MAP=/home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/maps/map_circular.yaml
11:  - TEST_DIR=/home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/src/system
11: -- run_test.py: invoking following command in '/home/mkhansen/ros2_dev/navigation2_ws/build/nav2_system_tests/src/system':
11:  - /home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_system_tests/src/system/test_system_launch.py
11: [INFO] [launch]: All log files can be found below /home/mkhansen/.ros/log/2019-09-03-13-00-36-057690-mkhansen-desk-5718
11: [INFO] [launch]: Default logging verbosity is set to INFO
11: [INFO] [gzserver-1]: process started with pid [5721]
11: [INFO] [static_transform_publisher-2]: process started with pid [5722]
11: [INFO] [static_transform_publisher-3]: process started with pid [5723]
11: [INFO] [map_server-4]: process started with pid [5724]
11: [INFO] [world_model-5]: process started with pid [5725]
11: [INFO] [gdb-6]: process started with pid [5726]
11: [INFO] [gdb-7]: process started with pid [5727]
11: [INFO] [navfn_planner-8]: process started with pid [5728]
11: [INFO] [recoveries_node-9]: process started with pid [5729]
11: [INFO] [bt_navigator-10]: process started with pid [5750]
11: [INFO] [lifecycle_manager-11]: process started with pid [5771]
11: [INFO] [test_system_node-12]: process started with pid [5781]
11: [static_transform_publisher-2] [INFO] []: LOOPING due to no latching at the moment
11: [static_transform_publisher-3] [INFO] []: LOOPING due to no latching at the moment
11: [map_server-4] [INFO] [map_server]: Creating
11: [world_model-5] [INFO] [world_model]: Creating World Model
11: [world_model-5] [INFO] [global_costmap.global_costmap]: Creating Costmap
11: [navfn_planner-8] [INFO] [navfn_planner]: Creating
11: [recoveries_node-9] [INFO] [recoveries]: Configuring Spin
11: [recoveries_node-9] [INFO] [recoveries]: Configuring BackUp
11: [bt_navigator-10] [INFO] [bt_navigator]: Creating
11: [lifecycle_manager-11] [INFO] [lifecycle_manager]: Creating
11: [lifecycle_manager-11] [INFO] [lifecycle_manager]: Starting the system bringup...
11: [lifecycle_manager-11] [INFO] [lifecycle_manager]: Creating and initializing lifecycle service clients
11: [lifecycle_manager-11] [INFO] [lifecycle_manager]: Configuring and activating map_server
11: [map_server-4] [INFO] [map_server]: Configuring
11: [map_server-4] [INFO] [map_server]: OccGridLoader: Creating
11: [map_server-4] [INFO] [map_server]: OccGridLoader: Configuring
11: [map_server-4] [INFO] [map_server]: Activating
11: [map_server-4] [INFO] [map_server]: OccGridLoader: Activating
11: [lifecycle_manager-11] [INFO] [lifecycle_manager]: Configuring and activating amcl
11: [gdb-7] GNU gdb (Ubuntu 8.1-0ubuntu3) 8.1.0.20180409-git
11: [gdb-7] Copyright (C) 2018 Free Software Foundation, Inc.
11: [gdb-7] License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
11: [gdb-7] This is free software: you are free to change and redistribute it.
11: [gdb-7] There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
11: [gdb-7] and "show warranty" for details.
11: [gdb-7] This GDB was configured as "x86_64-linux-gnu".
11: [gdb-7] Type "show configuration" for configuration details.
[... cut ...]
11: [INFO] [gdb-7]: process has finished cleanly [pid 5727]
11: [gdb-6] (gdb) 
11: [gdb-7] (gdb) (gdb) 
11: -- run_test.py: return code 1
11: -- run_test.py: generate result file '/home/mkhansen/ros2_dev/navigation2_ws/build/nav2_system_tests/test_results/nav2_system_tests/test_bt_navigator.xml' with failed test
11: -- run_test.py: verify result file '/home/mkhansen/ros2_dev/navigation2_ws/build/nav2_system_tests/test_results/nav2_system_tests/test_bt_navigator.xml'
1/1 Test #11: test_bt_navigator ................***Failed   88.51 sec

crdelsey, 2019-09-04:
After updating to the latest ROS2 and Nav2, I get fail case **(8)** 100% of the time. I believe the primary error in the log is
```
11: [world_model-5] >>> [rcutils|error_handling.c:107] rcutils_set_error_state()
11: [world_model-5] This error state is being overwritten:
11: [world_model-5] 
11: [world_model-5]   'publisher's context is invalid, at /home/crdelsey/src/ros_dev_workspace/ros2_ws/src/ros2/rcl/rcl/src/rcl/publisher.c:343'
11: [world_model-5] 
11: [world_model-5] with this new error message:
11: [world_model-5] 
11: [world_model-5]   'Could not publish transition, at /home/crdelsey/src/ros_dev_workspace/ros2_ws/src/ros2/rcl/rcl_lifecycle/src/rcl_lifecycle.c:336'
11: [world_model-5] 
11: [world_model-5] rcutils_reset_error() should be called after error handling to avoid this.
11: [world_model-5] <<<
11: [world_model-5] [ERROR] []: Failed to finish transition 3. Current state is now: active
```

I believe this problem ultimately triggers the lifecycle manager problem cited.

AAlon, 2019-09-16:
@crdelsey @mkhansen-intel is there any update w.r.t the above? have you had the chance to retry with the latest ROS2 and if so how frequently are you seeing which errors?

Also, if folks need to use the latest ROS2, does that imply using the latest navigation stack (rather than 0.2.4) as well?

Thanks!

SteveMacenski, 2019-09-16:
@AAlon yes if you're using master ROS2 you should use our master as well

dirk-thomas, 2019-09-16:
The referenced PR ros2/rcl#501 only resolves the "This error state is being overwritten" problem reported in **(8)**. The actual error "Failed to finish transition 3" will still be there.

AAlon, 2019-09-17:
Going by the comments made by @crdelsey and @mkhansen-intel two weeks ago I thought there were still numerous pending issues here...? just want to make sure it was closed on purpose and not by accident.

mkhansen-intel, 2019-09-17:
> @crdelsey @mkhansen-intel is there any update w.r.t the above? have you had the chance to retry with the latest ROS2 and if so how frequently are you seeing which errors?
> 
Actually, with the latest ROS2 master (aka Eloquent), I'm not seeing any of these issues anymore. I am seeing some other issues occasionally, and will file separate issues for those, but I believe those are in our stack.

This is now closed! :+1:

dirk-thomas, 2019-09-17:
Sorry, I used a wording which GitHub interpreted to close this issue. I will reopen since it seems there are still numerous issues pending.

mkhansen-intel, 2019-09-17:
I'm re-closing this, since I'm not seeing these issues anymore. I'm running more tests 100's of times and will file individual issues as I find them.

@dirk-thomas - THANKS for your help with this. The fixes you made do seem to help.

closed 2019-09-17 by mkhansen-intel