ros-navigation/navigation2 #1081: [bt_navigator] Node not recognized: clearEntirelyCostmapServiceRequest when run with stress

crdelsey, 2019-08-29:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - source
- Version or commit hash:
  - 2eec143
- DDS implementation:
  - FastRTPS
- Client library (if applicable):
  - NA

#### Steps to reproduce issue
run `stress -c 8` (`sudo apt install stress`)
In another terminal run the `test_bt_navigator` test.

#### Expected behavior
It passes

#### Actual behavior
Test fails 100% of the time with the following error in the log

```
10: [bt_navigator-10] [ERROR] []: Caught exception in callback for transition 10
10: [bt_navigator-10] [ERROR] []: Original error: Error at line 18: -> Node not recognized: clearEntirelyCostmapServiceRequest
10: [bt_navigator-10] [WARN] []: Error occurred while doing error handling.
10: [bt_navigator-10] [FATAL] [bt_navigator]: Lifecycle node entered error state
10: [lifecycle_manager-11] [ERROR] [lifecycle_manager]: Failed to change state for node: bt_navigator
10: [lifecycle_manager-11] [ERROR] [lifecycle_manager]: Failed to bring up node: bt_navigator, aborting bringup
```

#### Additional information

This continued to fail even when I reduced the `stress -c` value from 8 to 2. That value sets the number of threads that run in the background consuming CPU.

SteveMacenski, 2019-08-29:
It doesn't exist anymore, its been superseded https://github.com/ros-planning/navigation2/blob/master/nav2_behavior_tree/include/nav2_behavior_tree/clear_costmap_service.hpp and the behavior trees have been updated to reflect that.

Where it is getting that from?

crdelsey, 2019-08-29:
You're right. I must have had something in an inconsistent state. I did a clean rebuild and no longer can reproduce.

closed 2019-08-29 by crdelsey