ros-navigation/navigation2 #1082: All nodes need to be able to be launched separately from lifecycle manager and in different launch files

SteveMacenski, 2019-08-29:
This is getting very aggravating when I'm debugging one issue and I can't isolate just the stuff I care about

Lifecycle should not be so intrusive

crdelsey, 2019-08-29:
@bpwilcox and I were talking about making every node automatically transition itself to the active state unless someone explicitly sets a parameter telling it not to do so.

SteveMacenski, 2019-08-29:
That might be a good solution, separate from lifecycle manager? 

I dont like having this central authority bringing everything down when 1 thing fails, or having to launch stuff all at once in one terminal because of how the manager is handling bringup.

What needs to be possible, whether or not its supported is to be able to open a screen session (or tmux, or 20 terminals, whatever) and be able to individually launch all of my nodes and then have everything work. The packages are separate entities and a typical use case is not going to use everything in Navigation2 so being able to pick and chose is necessary. And this makes me not want to jump out a window when I'm trying to fix complex problems and I'm just getting logs dumped at me from rviz/gazebo/life cycle transitions/etc

mkhansen-intel, 2019-08-29:
@SteveMacenski - dude, calm down. You're overreacting.

If you want to add an 'autostart' mode to each individual node, you're welcome to do that. 

Other solutions:
 - launch each thing individually, including lifecycle manager, setting autostart = false, and use the RViz tool to transition the states via the "startup" button in the tool.
 - launch each node individually and manually control the states through the ros2 cli by doing service calls
 - launch each node individually and write your own python script to do whatever you want to do

mkhansen-intel, 2019-08-29:
Another option - you should also be able to launch each node individually and lifecycle_manager separately as well with 'autostart=true' and it should bring up all the nodes for you. The manor in which the nodes are launched is irrelevant. As long as their 'change_state' service is available for the lifecycle_manager to call. 

So, really, the issue you cited in the title is already possible in several ways. The only thing that's not possible is launching all the nodes and have them bring themselves to the active state. As I mentioned before, if you want to add that capability, then great. I don't see a need for it with the four other options I've mentioned, but I'm not opposed to it.

bpwilcox, 2019-08-29:
In addition to allowing nodes to transition to the active state by default (not requiring a manager or external lifecycle transition calls), we still have the issue of some nodes depending on activity from other nodes (for example dependency of costmap on amcl transforms) in order to transition into the active state. I had spoken with @crdelsey on this issue and suggested that we could make each node's lifecycle transitions completely independent, then whilst in the active state the node can check for any of its external dependencies and set a flag (e.g. `running` or `idle`) if all dependencies have passed and the node can process its work. Additionally, this dependency check could perhaps be run continuously and toggle the running state if for some reason the dependencies checks fail during the run.

closed 2019-08-29 by SteveMacenski

mkhansen-intel, 2019-08-30:
Just for completeness, there's another way to do this using the launch code itself to "EmitEvent" to change the states of the lifecycle nodes. 

Here's the example: https://github.com/ros2/launch_ros/blob/master/launch_ros/examples/lifecycle_pub_sub_launch.py