ros-navigation/navigation2 #1092: Document how to run 'updown' test in README or somewhere

mkhansen-intel, 2019-08-30:
## Feature request
Add documentation for how to run the updown test

SteveMacenski, 2020-01-14:
@mkhansen-intel @crdelsey do you intend in doing this? I don't really know anything about that test to document myself.

crdelsey, 2020-01-14:
There is a brief explanation here.

https://github.com/ros-planning/navigation2/blob/1ec7b205ddee240273ac6ba5d218d050e8ab4f4c/nav2_system_tests/src/updown/updownresults.py#L17-L26

I could move this to a readme. 

@mkhansen-intel You made a bunch of changes to this test, but I don't see them in the branch. DId you ever get those changes working? I don't recall.

mkhansen-intel, 2020-01-14:
I updated the test and got it working without requiring simulation in #1413. I'll add a README file to the folder with simple instructions for running it.

closed 2020-01-18 by SteveMacenski