ros-navigation/navigation2 #1100: Multiple startups and shutdowns of navfn kills the process

orduno, 2019-09-05:
I'm seeing on the global planner *tests* that trying to startup the navfn node after a prior shutdown kills the process:

```bash
[navfn_planner-1] [INFO] [navfn_planner]: Creating
[navfn_planner-1] [INFO] [navfn_planner]: Configuring
[navfn_planner-1] [INFO] [navfn_planner]: Activating
[navfn_planner-1] [INFO] [navfn_planner]: Deactivating
[navfn_planner-1] [INFO] [navfn_planner]: Cleaning up
[test_planner_node-2] [       OK ] PlannerTester.testSimpleCostmaps (2350 ms)
[test_planner_node-2] [ RUN      ] PlannerTester.testWithHundredRandomEndPoints
[test_planner_node-2] [INFO] [PlannerTester]: Loading map with file_path: /home/caorduno/master/overlay_ws/src/navigation2/nav2_system_tests/maps/map.pgm
[test_planner_node-2] [INFO] [PlannerTester]: Costmap: Setting static costmap
[navfn_planner-1] [INFO] [navfn_planner]: Configuring
[navfn_planner-1] [INFO] [navfn_planner]: Activating
[ERROR] [navfn_planner-1]: process has died [pid 3006, exit code -11, cmd '/home/caorduno/master/overlay_ws/install/nav2_navfn_planner/lib/nav2_navfn_planner/navfn_planner --ros-args'].
```

SteveMacenski, 2019-09-05:
@orduno if it helps, my changes to the tests here https://github.com/ros-planning/navigation2/pull/1089 remove that issue since i made them 2 separate tests

orduno, 2019-09-09:
OK, once you merge #1089 and change the planner tests we can close this issue.

closed 2019-09-10 by SteveMacenski