ros-navigation/navigation2 #1102: Issue template needs adjusting

rotu, 2019-09-09:
1. Ambiguous whether “installation type” or “version or commit hash” is of Nav2 or of ROS itself. Also should mention how to get it if installed via apt
2. Client library should be removed. It’s an implementation detail within nav2

```
**Required Info:**

- Operating System:
  - <!-- OS and version (e.g. Windows 10, Ubuntu 16.04...) -->
- Installation type:
  - <!-- binaries or from source  -->
- Version or commit hash:
  - <!-- Output of git rev-parse HEAD, release version, or repos file  -->
- DDS implementation:
  - <!-- rmw_implementation used (e.g. Fast-RTPS, RTI Connext, etc -->
- Client library (if applicable):
  - <!-- e.g. rclcpp, rclpy, or N/A -->
```

SteveMacenski, 2019-09-09:
I don’t think the version is ambiguous in the navigation project but more clarity doesnt hurt. Feel free to submit a PR to clarify. 

Agreed on client library point. I think I just copy pasted that from another template and I was thinking the client calling the navigation action server, but I think thats a pretty specialized type of problem that someone would mention in the description.

rotu, 2019-09-09:
> Feel free to submit a PR to clarify.

Will do. I thought this lived in some GitHub config page, but it turns out it's [in the repo](https://github.com/ros-planning/navigation2/blob/master/.github/ISSUE_TEMPLATE.md). Yay!

rotu, 2019-09-09:
> I don’t think the version is ambiguous in the navigation project

@SteveMacenski 
Okay. Um so I'm going to ask the stupid question then. Version of what? ROS2 or Nav2?

SteveMacenski, 2019-09-09:
Oh sorry, I guess it _was_ really ambiguous, that's the version of Navigation2.