ros-navigation/navigation2 #1103: Display Goal on Gazebo

mhpanah, 2019-09-09:
Add a visualization marker on gazebo to display goal location.

SteveMacenski, 2019-09-09:
If you're going to make a publisher with the goal pose in NavigateToPose, can you also update the rviz config file to look for that marker? Would be nice to be shown in both. You could use the marker message and build a gazebo plugin for visualizing that type, if it doesnt already exist.

mhpanah, 2019-09-10:
@SteveMacenski I was planning to put a visual link on the model.sdf file and then use setEntityState service request to place that visual link on the goal location.  That's how I currently have it on my local machine for RL related tasks but can look at the gazebo plugins to see what other options are out there.

SteveMacenski, 2019-09-10:
Yeah, that seems like a not-ideal work around. You could make a model plugin that's update looks for the marker topic and updates the position so that 1 topic is good for rviz and gazebo http://gazebosim.org/tutorials?tut=plugins_model

I'm curious what your rationale for putting it in gazebo is? Its not aiding the simulation and I think most the time gazebo is being run in headless mode so rviz is the place for "visualization" things like markers for goal poses, since its not a physical thing.

mhpanah, 2019-09-10:
For RL related tasks I'm not using rviz. During training, random goal will be generated.  To observe the training progress it would be helpful to visualize the goal on Gazebo when rendering the environment.

SteveMacenski, 2019-09-10:
Ok, I think this is just a technology mismatch then. Gazebo's front end isn't really meant for userspace data visualization, it seems like you should be using rviz or some other tool rather than introducing artifacts into the simulation for visualization purposes.

mhpanah, 2019-09-10:
Yes, but I think for this purpose, a visual link (say a cylinder with a height of 0.001) that has no physical properties should be harmless. Green circle below shows the goal location. What do you think?
[image]

SteveMacenski, 2019-09-11:
As a workaround, that seems fine. It seems to me the tool you actually want is rviz and running gazebo headlessly since the gzclient takes a ton of CPU and rviz is comparatively lightweight so you can train faster.

ruffsl, 2019-09-17:
Adding visual markers into the simulation also limits its use when training or testing visual models that include cameras that render the same simulation scene. I.e. is the visual DNN model really leaning to navigating the environment, or just happening to chase the big green circle in the rgb image?

SteveMacenski, 2019-09-17:
He's only using the laser scanner for this particular example

ruffsl, 2019-09-17:
> He's only using the laser scanner for this particular example

One could also think of where a "visual" marker polluting a the simulation environment not only side-affects the 2D image sensors, but also 1D cameras or ranging LIDARs. I'm not sure if disabling collisions for the SDF model would prevent the ray-tracer from intersecting the marker's surface mesh.

SteveMacenski, 2019-09-17:
@ruffsl preaching to the choir, that's why I had the discussion above. Visualizations are for rviz, real world things are for gazebo. But with that said I always support more open-source visible implementations of gazebo plugins. They're so powerful and there's almost nothing out there (and I'm also guilty of not open sourcing mine...)

mhpanah, 2019-09-17:
@ruffsl @SteveMacenski I agree and I'm planning to put it in rviz eventually.  But for now since I'm not using rviz and map, to visualize the goal location I decided to put it in Gazebo temporarily as a workaround.  Later, I'm planning to use 2d costmap as input to my network instead of raw laser data.  When that happens then it would make sense for me to use rviz for visualization instead.  Also, since the height of this marker is 1mm, it cannot be detect by the TB3 LiDAR.

SteveMacenski, 2019-09-21:
Just from an issue tracking perspective, is this a ticket that needs to be open because its an issue? Certainly whatever we do here we'll merge but do we need an active ticket open for it?

mhpanah, 2019-09-23:
We usually open an issue before submitting a PR. My plan is to submit a PR for this as a temp workaround but I'm okay to close it if this solution is not needed by anyone else.

closed 2019-09-23 by SteveMacenski

nocoinman, 2021-06-14:
> Adding visual markers into the simulation also limits its use when training or testing visual models that include cameras that render the same simulation scene. I.e. is the visual DNN model really [...]

Is there a way to make visual markers in gazebo appear in camera images?

SteveMacenski, 2021-06-14:
This is super off topic, please do not revive old tickets on completely unrelated topics.