ros-navigation/navigation2 #1105: [bt_navigator] error wth getCurrentPose() ??

hyunseok-yang, 2019-09-10:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - 0.2.4 source
- Version or commit hash:
  - dashing
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

```
```

Only produced first time when I set the goal on the rviz.
But the next goal does not show ANY errors.

#### Expected behavior
No error log from 'bt_navigator' when I set the goal on the map with '0.2.3' version

#### Actual behavior

[bt_navigator-8] [ERROR] [getCurrentPose] No Transform available Error looking up robot pose: "map" passed to lookupTransform argument target_frame does not exist. 

.......
.......
.......

[bt_navigator-8] [ERROR] [getCurrentPose] Extrapolation Error looking up robot pose: Lookup would require extrapolation into the past.  Requested time 24.56500 but the earliest data is at time 24.59300, when looking up transform from frame [base_link] to frame [map]
.......
.......

#### Additional information

Have you guys seen these kind of error with '0.2.4' version?

----
## Feature request

#### Feature description

#### Implementation considerations

mkhansen-intel, 2019-09-11:
So do you see this every time? Can you please add the "Steps to Reproduce"?

crdelsey, 2019-09-11:
Sounds like bt navigator is trying to get the pose before AMCL has published it's first transform.

@hyunseok-yang About how much time has passed between clicking `set initial pose` in rviz and then clicking `set goal pose`?

AAlon, 2019-09-14:
We're seeing this too (https://github.com/aws-robotics/aws-robomaker-sample-application-cloudwatch/pull/38). Not using rviz to set the pose, we have an action client node that sends `NavigateToPose` requests. Tried setting `set_initial_pose: True` for amcl, didn't help.

hyunseok-yang, 2019-09-16:
@AAlon  I also tried setting 'set_initial_pose', but it didn't work too.

hyunseok-yang, 2019-09-16:
Sorry for late response! (it was long holidays in my country ;) )

@mkhansen-intel 
I can see these error every time only at first goal setting.

The step to reproduce is very simple.
1. launch the launch.py for driver+gazebo(sim)
2. launch the launch.py for nav2(0.2.4)
3. set init pose from rviz

@crdelsey

hyunseok-yang, 2019-09-16:
Here is the log I tried again 

[lifecycle_manager-1] [INFO] [lifecycle_manager] Configuring and activating dwb_controller
[dwb_controller-5] [INFO] [dwb_controller] Configuring
[dwb_controller-5] [INFO] [local_costmap.local_costmap] Configuring
[world_model-4] [INFO] [global_costmap.global_costmap_rclcpp_node] [signalFailure] Drop message: frame 'base_scan' at time 52.765 for reason(0)
[dwb_controller-5] [INFO] [local_costmap.local_costmap] Using plugin "obstacle_layer"
[dwb_controller-5] [INFO] [local_costmap.local_costmap] Subscribed to Topics: scan
[dwb_controller-5] [INFO] [local_costmap.local_costmap] Using plugin "inflation_layer"
[rviz2-9] [INFO] [rviz2] Trying to create a map of size 3000 x 3000 using 1 swatches
[dwb_controller-5] [INFO] [dwb_controller] Setting transform_tolerance to 0.100000
[dwb_controller-5] [INFO] [dwb_controller] Using critic "ObstacleFootprint" (dwb_critics::ObstacleFootprintCritic)
[dwb_controller-5] [INFO] [dwb_controller] Critic plugin initialized
[dwb_controller-5] [INFO] [dwb_controller] Using critic "PathDist" (dwb_critics::PathDistCritic)
[dwb_controller-5] [INFO] [dwb_controller] Critic plugin initialized
[dwb_controller-5] [INFO] [dwb_controller] Using critic "GoalDist" (dwb_critics::GoalDistCritic)
[dwb_controller-5] [INFO] [dwb_controller] Critic plugin initialized
[dwb_controller-5] [INFO] [dwb_controller] Using critic "RotateToGoal" (dwb_critics::RotateToGoalCritic)
[dwb_controller-5] [INFO] [dwb_controller] Critic plugin initialized
[dwb_controller-5] [INFO] [dwb_controller] Using critic "Oscillation" (dwb_critics::OscillationCritic)
[dwb_controller-5] [INFO] [dwb_controller] Critic plugin initialized
[dwb_controller-5] [INFO] [dwb_controller] Controller frequency set to 10.0000Hz
[dwb_controller-5] [INFO] [dwb_controller] Activating
[dwb_controller-5] [INFO] [local_costmap.local_costmap] Activating
[dwb_controller-5] [INFO] [local_costmap.local_costmap] Checking transform
[dwb_controller-5] [INFO] [local_costmap.local_costmap] start
[rviz2-9] [INFO] [rviz2] Trying to create a map of size 120 x 120 using 1 swatches
[lifecycle_manager-1] [INFO] [lifecycle_manager] Configuring and activating navfn_planner
[navfn_planner-6] [INFO] [navfn_planner] Configuring
[navfn_planner-6] [INFO] [navfn_planner] Activating
[lifecycle_manager-1] [INFO] [lifecycle_manager] Configuring and activating bt_navigator
[bt_navigator-8] [INFO] [bt_navigator] Configuring
[bt_navigator-8] [INFO] [bt_navigator_client_node] Waiting for "ComputePathToPose" action server
[bt_navigator-8] [INFO] [bt_navigator_client_node] "ComputePathToPose" BtActionNode initialized
[bt_navigator-8] [INFO] [bt_navigator_client_node] Waiting for "FollowPath" action server
[bt_navigator-8] [INFO] [bt_navigator_client_node] "FollowPath" BtActionNode initialized
[bt_navigator-8] [INFO] [bt_navigator_client_node] Waiting for "/local_costmap/clear_entirely_local_costmap" service
[bt_navigator-8] [INFO] [bt_navigator_client_node] "ClearEntireCostmap" BtServiceNode initialized
[bt_navigator-8] [INFO] [bt_navigator_client_node] Waiting for "/global_costmap/clear_entirely_global_costmap" service
[bt_navigator-8] [INFO] [bt_navigator_client_node] "ClearEntireCostmap" BtServiceNode initialized
[bt_navigator-8] [INFO] [bt_navigator_client_node] Waiting for "Spin" action server
[bt_navigator-8] [INFO] [bt_navigator_client_node] "Spin" BtActionNode initialized
[bt_navigator-8] [INFO] [bt_navigator] Activating
[lifecycle_manager-1] [INFO] [lifecycle_manager] The system is active
[world_model-4] [INFO] [global_costmap.global_costmap_rclcpp_node] [signalFailure] Drop message: frame 'base_scan' at time 52.794 for reason(0)
[world_model-4] [INFO] [global_costmap.global_costmap_rclcpp_node] [signalFailure] Drop message: frame 'base_scan' at time 52.824 for reason(0)
[world_model-4] [INFO] [global_costmap.global_costmap_rclcpp_node] [signalFailure] Drop message: frame 'base_scan' at time 52.854 for reason(0)
[world_model-4] [INFO] [global_costmap.global_costmap_rclcpp_node] [signalFailure] Drop message: frame 'base_scan' at time 52.900 for reason(0)
[world_model-4] [INFO] [global_costmap.global_costmap_rclcpp_node] [signalFailure] Drop message: frame 'base_scan' at time 52.914 for reason(0)
[world_model-4] [INFO] [global_costmap.global_costmap_rclcpp_node] [signalFailure] Drop message: frame 'base_scan' at time 52.974 for reason(0)
[world_model-4] [INFO] [global_costmap.global_costmap_rclcpp_node] [signalFailure] Drop message: frame 'base_scan' at time 53.005 for reason(0)
[world_model-4] [INFO] [global_costmap.global_costmap_rclcpp_node] [signalFailure] Drop message: frame 'base_scan' at time 53.051 for reason(0)
[world_model-4] [INFO] [global_costmap.global_costmap_rclcpp_node] [signalFailure] Drop message: frame 'base_scan' at time 53.065 for reason(0)
[world_model-4] [INFO] [global_costmap.global_costmap_rclcpp_node] [signalFailure] Drop message: frame 'base_scan' at time 53.094 for reason(0)
[world_model-4] [INFO] [global_costmap.global_costmap_rclcpp_node] [signalFailure] Drop message: frame 'base_scan' at time 53.124 for reason(0)
[world_model-4] [INFO] [global_costmap.global_costmap_rclcpp_node] [signalFailure] Drop message: frame 'base_scan' at time 53.155 for reason(0)
[world_model-4] [INFO] [global_costmap.global_costmap_rclcpp_node] [signalFailure] Drop message: frame 'base_scan' at time 53.197 for reason(0)
[... cut ...]
[dwb_controller-5] [INFO] [dwb_controller] Preempting the goal. Passing the new path to the planner.
[dwb_controller-5] [INFO] [dwb_controller] Preempting the goal. Passing the new path to the planner.
[dwb_controller-5] [INFO] [dwb_controller] Preempting the goal. Passing the new path to the planner.
[dwb_controller-5] [INFO] [dwb_controller] Preempting the goal. Passing the new path to the planner.
[dwb_controller-5] [INFO] [dwb_controller] Preempting the goal. Passing the new path to the planner.

SteveMacenski, 2019-09-16:
You've described several issues here, the different TF tree portion is something not coming up correctly. Please verify all your transformations are available.

hyunseok-yang, 2019-09-16:
@crdelsey  
I'm not sure how much time passed exactly... but I guess it's under 3 sec...?
I think the time you asked is not long.

hyunseok-yang, 2019-09-16:
@SteveMacenski 
Hi, several issues? 
I just want to know the reason of this error message.
**[bt_navigator-8] [ERROR] [getCurrentPose] Connectivity Error looking up robot pose: Could not find a connection between 'map' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees.**

Is this error contains several issues~?

BTW, in terms of TF, I thought TF is coming up correctly. Do you see or guess where is the problem..?
Because the robot is moving to the goal as I desired without no errors...

mkhansen-intel, 2019-09-17:
If your robot is moving, the transform must be OK. However you can verify either by looking in Rviz at the "Tree" view within the TF plugin, or by using `ros2 run tf2_ros tf2_echo map base_link`
You should see something like this:
```
mkhansen@mkhansen-desk:~/ros2_dev/navigation2_ws$ ros2 run tf2_ros tf2_echo map base_link
At time 33.530000000
- Translation: [-0.373, 1.550, 0.001]
- Rotation: in Quaternion [-0.000, 0.000, 0.525, 0.851]
At time 34.460000000
- Translation: [-0.251, 1.716, 0.001]
- Rotation: in Quaternion [0.001, 0.001, 0.440, 0.898]
```
See more tf tools here:
http://wiki.ros.org/tf2/Tutorials/Introduction%20to%20tf2

If this transform is NOT being published, then either AMCL has crashed, the map topic is not being published (unlikely if you see it in Rviz), or the /scan topic isn't being published. You can do a `ros2 topic echo /scan` to see if that's the problem.

mkhansen-intel, 2019-09-17:
Also, in order to use `set_initial_pose`, it will default to 0, 0, 0 location:
https://github.com/ros-planning/navigation2/blob/c241ab6cfa3f1504849824f604a8d59fc0b6a4d2/nav2_amcl/src/amcl_node.cpp#L107

You may need to change those values via the 'initial_pose.x` (and y, z, etc.) parameters.

hyunseok-yang, 2019-09-18:
@mkhansen-intel  
hi! 
I can see the transform what you mentioned by using this command `ros2 run tf2_ros tf2_echo map base_link`.

At time 142.633000000
- Translation: [113.986, 114.018, 0.010]
- Rotation: in Quaternion [0.000, 0.000, 0.924, 0.382]
At time 143.641000000
- Translation: [113.986, 114.018, 0.010]
- Rotation: in Quaternion [0.000, 0.000, 0.924, 0.382]
At time 144.649000000
- Translation: [113.986, 114.018, 0.010]
- Rotation: in Quaternion [0.000, 0.000, 0.924, 0.382]
At time 145.610000000
- Translation: [113.986, 114.018, 0.010]
- Rotation: in Quaternion [0.000, 0.000, 0.924, 0.382]
At time 146.620000000
- Translation: [113.986, 114.018, 0.010]
- Rotation: in Quaternion [0.000, 0.000, 0.924, 0.382]

hyunseok-yang, 2019-09-18:
@mkhansen-intel 

Should I use `set_initial_pose` even though I set the 'estimated pose(initial pose)' by manually through rviz?

orduno, 2019-09-23:
I was looking into this as the issue is amplified when working with multiple robots.

Regarding the first error:

```bash
[bt_navigator-8] [ERROR] [getCurrentPose] No Transform available Error looking up robot pose: "map" passed to lookupTransform argument target_frame does not exist.
```

I see this all the time (single or multiple robots) even after giving the system at least a minute after `lifecycle_manager` reports `The system is active` . The error originates from `nav2_behavior_tree::GoalReachedCondition`.

Consider that on the `isGoalReached` check we only `spin_some`

https://github.com/ros-planning/navigation2/blob/b7c6761a2c97dd774f6d2e25d129f021613bc411/nav2_behavior_tree/include/nav2_behavior_tree/goal_reached_condition.hpp#L77

And we only check once:
https://github.com/ros-planning/navigation2/blob/b7c6761a2c97dd774f6d2e25d129f021613bc411/nav2_behavior_tree/include/nav2_behavior_tree/goal_reached_condition.hpp#L78

I replaced the code to try multiple times:

```cpp
bool got_pose = false;
do {
  rclcpp::spin_some(node_);
  got_pose = nav2_util::getCurrentPose(current_pose, *tf_)
} while (rclcpp::ok() && !got_pose)
```

Repeating the process with this change results in multiple error outputs until the pose is finally obtained. Also, `getCurrentPose` takes a timeout but in this case, since the node is not spinning there, it's useless.

We should consider:
1. Adding multiple attempts of getting the pose (as shown above).
2. Always spin the node used by the tf. I've noticed some problems with tf's when it's node is not constantly spinning (something to explore further). I'd suggest modifying `GoalReachedCondition::initialize()` to launch a separate thread that is constantly spinning the node (the node on the blackboard is meant to spin as needed, so we would have to create a second one)

orduno, 2019-09-23:
I've pushed #1157 with some possible solution.

@hyunseok-yang Would you mind trying #1175 and see if it fixes the issue?

hyunseok-yang, 2019-09-24:
@orduno 
Hi, I've tried to apply your commit for test but I guess based version is different.
Which means... I'm working on latest dashing-devel branch (=0.2.4) and latest ros2 dashing debian package.
At least, "tf2_ros::CreateTimerROS" is not recognizable with my installed version.

It looks... you're working on master version...

orduno, 2019-10-10:
@hyunseok-yang Would you mind pulling the latest dashing-devel and test again? We've backported this fix and will be included on 0.2.5.

hyunseok-yang, 2019-10-19:
Hi, all
Sorry for late comment.

I  cannot see the error what I mentioned above after update to 0.2.5 version but can see new warning message.

> [dwb_controller-5] [INFO] [dwb_controller] Reached the goal!
> [bt_navigator-8] Do not call canTransform or lookupTransform with a timeout unless you are using another thread for populating data. Without a dedicated thread it will always timeout.  If you have a seperate thread servicing tf messages, call setUsingDedicatedThread(true) on your Buffer instance.Do not call canTransform or lookupTransform with a timeout unless you are using another thread for populating data. Without a dedicated thread it will always timeout.  If you have a seperate thread servicing tf messages, call setUsingDedicatedThread(true) on your Buffer instance.

But I guess it is already investigated by another ticket.

closed 2019-10-19 by hyunseok-yang